raspiblitz/raspiblitz #1236: Update Bitcoin Core to v0.20.0

openoms, 2020-06-03:
Successfully tested with running the latest scripts:
[image]

Manual update script: https://github.com/openoms/bitcoin-tutorials/blob/master/raspiblitz.updates/README.md#v0200

rootzoll, 2020-06-03:
https://github.com/bitcoin/bitcoin/releases/tag/v0.20.0

openoms, 2020-06-22:
Did not get any problems, but the new lnd version release notes have
```
This marks the second minor release in the v0.10.0 series! This release allows lnd to be compatible with bitcoind 0.20, and fixes an important bug related to an on disk Static Channel Backups (SCB).
```
https://github.com/lightningnetwork/lnd/releases/

Best to update.
Related: https://github.com/rootzoll/raspiblitz/issues/1240

rootzoll, 2020-07-24:
The Bitcoin 0.20.0 looking good. Closing issue.

closed 2020-07-24 by rootzoll