raspiblitz/raspiblitz #1240: Update: LND to 0.10.1

rootzoll, 2020-06-03:
LND 0.10.1 is aalready in RC3 .. so it seems valid to release it with v1.6.

The questions is @frennkie @openoms : In v1.5 we still has 0.9.2 as default and 0.10.0 optional to not force people into database migration. For v1.6 do we want to go 0.10.1 by default?

openoms, 2020-06-03:
yes certainly! Have not found any breaking problems running lnd 0.10. From the point of view of the lnd database updating to v0.10.0 or v0.10.1 should not make any difference.

rootzoll, 2020-06-03:
TODO: Update the protobuffs build python libs

frennkie, 2020-06-03:
I agree with going with 0.10.1... But I would refrain from using a RC release of LND in a stable Raspiblitz release.. 😊

rootzoll, 2020-06-03:
OK updated the build script. So this is now on final testing.
TODO: Check the optional update script and menu (should not be offered in this version)

openoms, 2020-06-22:
lnd v0.10.2-beta.rc2 is out:
```
This marks the second minor release in the v0.10.0 series! This release allows lnd to be compatible with bitcoind 0.20, and fixes an important bug related to an on disk Static Channel Backups (SCB).
```

rootzoll, 2020-06-22:
- [ ] Wait for final release of LND 0.10.2 to include in final v1.6 RaspiBlitz

openoms, 2020-06-23:
Updating to LND 0.10.2-beta.rc2 with the reckless update menu function caused  my Odroid HC1 node to restart LND randomly after minutes repeatedly. Did not find anything obvious in the logs.
Downgrading to lnd 0.10.1 with: https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v0101-beta has solved the problem.

Now testing  LND 0.10.2-beta.rc2 on an RPi4.

rootzoll, 2020-06-29:
check https://github.com/lightningnetwork/lnd/releases when final release is available

openoms, 2020-07-06:
LND v0.10.2 and v0.10.3 are out
https://github.com/lightningnetwork/lnd/releases/tag/v0.10.2-beta
https://github.com/lightningnetwork/lnd/releases/tag/v0.10.3-beta

Testing v0.10.3. Looking good so far.

rootzoll, 2020-07-12:
@openoms for the basic RaspberryPi do you see any reason not to release v1.6 with lnd 0.10.3?

openoms, 2020-07-12:
Yes! the 0.10.3 and 0.10.2 binaries built with go1.14.4 are producing random freezes (see the lnd issue).
Our options:
* stick to lnd 0.10.1
* build lnd 0.10.3 from source with Go 1.13.3
* wait for repacked binaries or next version

rootzoll, 2020-07-12:
Ah sorry you said before on Odroid and RP4 .. just read Odroid.

OK so this is a showstopper then again. I say we wait until the end of the week and check if there is a repacked binary.

Checksum of the old corupted binary was in https://github.com/lightningnetwork/lnd/releases/download/v0.10.3-beta/manifest-v0.10.3-beta.txt
'''
bc1cb10223b7c0449be5d27c22656a9af4c133a88b881eed6d7e6ef2efa9b638  lnd-linux-armv7-v0.10.3-beta.tar.gz
'''

Waiting to see another checksum here as proof of uploaded new binary.

openoms, 2020-07-16:
Good news here: https://github.com/lightningnetwork/lnd/releases/tag/v0.10.4-beta
It wax packaged with Go 1.13.3.
Started testing the binary.
```
admin@raspberrypi:~ $ lncli -n testnet version
{
    "lncli": {
        "commit": "v0.10.4-beta",
        "commit_hash": "86114c575c2dff9dff1e1bb4df961c64aea9fc1c",
        "version": "0.10.4-beta",
        "app_major": 0,
        "app_minor": 10,
        "app_patch": 4,
        "app_pre_release": "beta",
        "build_tags": [
            "autopilotrpc",
            "signrpc",
            "walletrpc",
            "chainrpc",
            "invoicesrpc",
            "watchtowerrpc"
        ],
        "go_version": "go1.13.13"
    },
    "lnd": {
        "commit": "v0.10.4-beta",
        "commit_hash": "86114c575c2dff9dff1e1bb4df961c64aea9fc1c",
        "version": "0.10.4-beta",
        "app_major": 0,
        "app_minor": 10,
        "app_patch": 4,
        "app_pre_release": "beta",
        "build_tags": [
            "autopilotrpc",
            "signrpc",
            "walletrpc",
            "chainrpc",
            "invoicesrpc",
            "watchtowerrpc"
        ],
        "go_version": "go1.13.13"
    }
}
```

rootzoll, 2020-07-24:
LND 0.10.4 rinstalls and runs without problems. Closing issue.

closed 2020-07-24 by rootzoll