raspiblitz/raspiblitz #1247: IP2Tor - shop-list

frennkie, 2020-06-04:
Should a call to shop-list without an URL work?

```
admin@host:~ $ config.scripts/blitz.subscriptions.ip2tor.py
# manage ip2tor subscriptions for raspiblitz
# blitz.subscriptions.ip2tor.py create-ssh-dialog [servicename] [toraddress] [torport]
# blitz.subscriptions.ip2tor.py shop-list [shopurl]
# blitz.subscriptions.ip2tor.py shop-order [shopurl] [servicename] [hostid] [toraddress:port] [duration] [msatsFirst] [msatsNext] [?description]
# blitz.subscriptions.ip2tor.py subscriptions-list
# blitz.subscriptions.ip2tor.py subscriptions-renew [secondsBeforeSuspend]
# blitz.subscriptions.ip2tor.py subscription-cancel [id]
# blitz.subscriptions.ip2tor.py subscription-by-service [servicename]

admin@tpt3:~ $ config.scripts/blitz.subscriptions.ip2tor.py shop-list

None
error='incorrect parameters'
admin@host:~ $
```

rootzoll, 2020-06-04:
When the url is missing, then a "incorrect parameters" is correct. Because which shop should be called?

frennkie, 2020-06-04:
That's fine.. But then this is misleading..!

```
# blitz.subscriptions.ip2tor.py shop-list [shopurl]
```

Need to remove the brackets from shopurl..

rootzoll, 2020-06-04:
Oh ok ... I always thought that something in brackets is a Variable. If a variable is optional it would be [?shopurl]

frennkie, 2020-06-04:
I would say

```
blitz.subscriptions.ip2tor.py shop-list [shopurl]
```

-> shopurl is optional

```
blitz.subscriptions.ip2tor.py shop-list shopurl
```

-> shopurl is required... 

🤷

rootzoll, 2020-06-04:
We could be more strict on having "one standard" but is it that important?

frennkie, 2020-07-12:
Just saying: https://en.wikipedia.org/wiki/Usage_message

If something has `[]` then around it then it is **optional**.

@openoms Do you want to be the *tiebreaker* here and decide whether we fix this or leave it as is?! ;-D

openoms, 2020-07-12:
> Just saying: https://en.wikipedia.org/wiki/Usage_message
> 
> If something has `[]` then around it then it is **optional**.
> 
> @openoms Do you want to be the *tiebreaker* here and decide whether we fix this or leave it as is?! ;-D

According the  wiki article:  
<for required>  
[for optional]  
arguments.

I think it is useful to put some kind of bracket to indicate it is an argument. Otherwise it looks like it is a part of the command.

So in this case it would be:
```​blitz.subscriptions.ip2tor.py <shop-list> [shopurl]
```

frennkie, 2020-07-12:
I don't like the <>. 

People should be easily able to copy paste from the help output.

The only things that should get any kind of brackets is the description as this can be given of not.

openoms, 2020-07-12:
The the options/variables which need to be filled (and not copy-pasted) would need to be in some kind of bracket.
`< >` for required
`[ ]` for optional
If you are showing the options as different commands then they can be easily copied and don't need brackets.

PR #1333 (2020-07-12): Update blitz.subscriptions.ip2tor.py
Feel free to reject.. :-D

Resolve/ fix #1247
closed 2020-07-17 by frennkie