raspiblitz/raspiblitz #1258: BTCPay over HTTPS

rootzoll, 2020-06-12:
In former releases BTCPay Server was just available tru TOR. Now with #1256 it should also be available over HTTPS using the new nginx server proxy.

- [ ] Test merged PR
- [ ] add https://LAN_IP:23001 login to menu
- [ ] note change in FAQ

rootzoll, 2020-06-25:
The selfsigned HTTPS seems to work, but the http at port 23000 is not responding and also the TOR looks still wrong.

openoms, 2020-06-26:
The port 23000 is bound to localhost:
    echo "
### Global settings ###
network=mainnet

### Server settings ###
port=23000
bind=127.0.0.1
externalurl=https://$BTCPayDomain

### NBXplorer settings ###
BTC.explorer.url=http://127.0.0.1:24444/
BTC.lightning=type=lnd-rest;server=https://127.0.0.1:8080/;macaroonfilepath=/home/btcpay/admin.macaroon;certthumbprint=$FINGERPRINT
" | sudo -u btcpay tee -a /home/btcpay/.btcpayserver/Main/settings.config

rootzoll, 2020-06-26:
@openoms I merged the PR, but now even the HTTPS self-signed is not working anymore and on TOR I get this:
[image]

openoms, 2020-06-26:
that i weird. It is working for me.

torrc:
```
$ sudo cat /etc/tor/torrc
...
# Hidden Service for btcpay
HiddenServiceDir /mnt/hdd/tor/btcpay
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:23002
HiddenServicePort 443 127.0.0.1:23003
```
ports:
```
$ sudo netstat -tulpn | grep 230
tcp        0      0 127.0.0.1:23000         0.0.0.0:*               LISTEN      15727/BTCPayServer  
tcp        0      0 0.0.0.0:23001           0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 127.0.0.1:23002         0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 127.0.0.1:23003         0.0.0.0:*               LISTEN      20603/nginx: master 
tcp6       0      0 :::23001                :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 ::1:23002               :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 ::1:23003               :::*                    LISTEN      20603/nginx: master 
```

what does:
`sudo nginx -t`
show you?

openoms, 2020-06-26:
you will need to edit this file manually to make sure they forward form the port 23000 (not 23001 which was a mistake)

/etc/nginx/sites-available/btcpay_ssl.conf
/etc/nginx/sites-available/btcpay_tor.conf
/etc/nginx/sites-available/btcpay_tor_ssl.conf

rootzoll, 2020-06-26:
@openoms  BTW can we delete the btcpaysetdomain.sh ?

openoms, 2020-06-26:
> @openoms  BTW can we delete the btcpaysetdomain.sh ?

Yes, that will not useful any more with this setup.

frennkie, 2020-07-12:
@openoms for `LNbits` I had to make a change so that it works behind (IP2)Tor: https://github.com/lnbits/lnbits/issues/46

What do you think about this: https://stackoverflow.com/questions/15414810/whats-the-difference-of-host-and-http-host-in-nginx

I would say we should be able to change out snippet (https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/assets/nginx/snippets/ssl-proxy-params.conf#L4) , right?

closed 2020-07-25 by rootzoll