raspiblitz/raspiblitz #1259: Docker on raspiblitz

BTCTOME, 2020-06-14:
Hi. I try to get Docker on my Raspiblitz to run. I can install and start docker, docker-ce and docker-compose. But as soon as I start a container, the Raspiblitz completely stops working and starts again. I don't find the cause of it. Maybe someone can help me.

BTCTOME, 2020-06-21:
Nobody has an idea?

phenomeus, 2020-07-12:
I think its because _background.sh is checking the IP and is not seeing your eth0 or wlan0 but docker0 or virtual eth. 
but there you have no IP assigned and thus the scrip from raspiblitz is rebooting because it thinks its failing to get an IP (but its not!) 

that's why I created this PR: #1337

Raspi247, 2020-07-13:
I have Raspiblitz on a RaspberryPi4 with 4GB. Since the image has to be rewritten with every update, it would make sense to get Docker up and running. I still have enough resources on the Pi4, but I tried everything possible, as soon as a container starts, Raspiblitz ends its work and starts again.

Would be nice if there is a solution in the future.

phenomeus, 2020-07-13:
I'm on the same page, that's the reason why I modified the checking just to work against a too easy reseter. 
the main reason for that behavior is that it only should be a RaspiBlitz and the image is overwritten every update. 

i also wants to have another update routine and am working on it. right now I got it by build_sdcard.sh script. but you ran in a bunch of problems that you have to fix by hand right now.

rootzoll, 2020-07-16:
@phenomeus I was rejecting the #1337 PR because it was targeted to master branch instead to v1.6 branch and changing the base to v1.6 for that PR already had too many merge conflicts. But no worry - I made the changes to the local IP detection again on the v1.6 branch and with this commit: https://github.com/rootzoll/raspiblitz/commit/9c99093574ecb3106dbdec933e31c4388e8c44ea

At all: This fix will be part of the v1.6 RC3 please test if this than fixes your Docker restart problems.

Pinkman420, 2020-07-17:
working fine. many thanks for the fix <3

closed 2020-07-24 by rootzoll