raspiblitz/raspiblitz #1262: Update to 1.6 hangs on the DNS install

HansiRumker, 2020-06-15:
I update to 1.6 with new sd card.
The setup hangs on to install the dyndns setup, i does not startet.
Setup with a tor install works.

rootzoll, 2020-06-15:
Thanks for the report, will test the dyndns recovery on update again.

HansiRumker, 2020-06-16:
Thanks you, can I deactivate the Service before make the Setup, so that the Installation works ?

rootzoll, 2020-06-16:
This could be an option if you can get to terminal run command `menu` und under SERVICES try to deactivate the DynDNS.

HansiRumker, 2020-06-18:
Thank you, so it works.

rootzoll, 2020-06-18:
I can confirm, that when a dynDNS "dynDomain" is set, then this is the last log-output .. then its stands still:
'''
Provisioning AUTO NAT DISCOVERY - keep default
Provisioning DYNAMIC DNS - run config script
number of args(2)
making sure services are not running
switching the DynamicDNS ON
dynDomain(freedns.afraid.org)
dynUpdateUrl()
changing lnd.conf
DynamicDNS is now ON
# making sure services are not running
# keep old tls data as backup
# start to create new generate new TLSCert
'''

HansiRumker, 2020-06-18:
Now is the Problem, that the Mobil Conect not works, 
I have no Idee. Make a new TLSCert, but becone this output.
FAIL - LND Macaroons not created
Please check the following LND issue:
https://github.com/lightningnetwork/lnd/issues/890
You may want try again with starting ./70initLND.sh

rootzoll, 2020-06-18:
@HansiRumker in the v1.6 go MAINMENU > LNDCREDS > RESET ... is that working?

HansiRumker, 2020-06-18:
# all your macaroons get deleted and recreated
rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/macaroons.db': No such file or directory
## Resetting TLS
# tls cert gets deleted and recreated
# making sure services are not running
# keep old tls data as backup
# start to create new generate new TLSCert
# wait until generated
# (61/60) check for cert
error='failed to generate new LND cert'
# restarting LND ... wait 10 secs
# checking LND wallet ... (can take some time)
# OK LND wallet was already unlocked
# copy new macaroons to central app-data directory and ensure unix ownerships and permissions
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/admin.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoice.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/readonly.macaroon': No such file or directory

rootzoll, 2020-06-18:
@HansiRumker something seems strange with your LND can you post the LND part of:
MAINMENU > REPAIR > SOFTWARE

Juized, 2020-06-19:
same thing happened with me after I updated to v1.6 ( Got hung up on the Dynamic Dns Setup). I disabled the feature with old version SD card (v1.5) then continued the restore procedure with the newer version. And after having restored my old data onto v1.6 it was quite easy to re-enable the Dyn DNS option as before

> I can confirm, that when a dynDNS "dynDomain" is set, then this is the last log-output .. then its stands still:
>
> ```
> Provisioning AUTO NAT DISCOVERY - keep default
> Provisioning DYNAMIC DNS - run [...]

rootzoll, 2020-06-19:
@Juized I got it fixed yesterday - the fix will be part of v1.6RC2 to test.

closed 2020-07-24 by rootzoll