raspiblitz/raspiblitz #1264: enable use with non-local/trusted bitcoin core node

philbertw4, 2020-06-17:
Architecturally, how much of the Raspiblitz depends on the bitcoin full node running on the same machine? 

For example, if I have a full node already available (e.g. under my control/trust), it seems like I should be able to simply tell the raspblitz to use that node as it's full node and everything else would still work fine. Do you think there would be demand for offering such a feature?

@openoms gave an answer in telegram:
>there is no central setting for this. Would need to set every app and deal with the resulting delay and possible connection issues

So, while there is no central setting support right now, I was able to get lnd to still "seemingly" work fine by telling it to communicate with a different bitcoind node (while the pi was syncing). The problems really only seemed to arise when services rebooting depended on the bitcoind service.

So I am wondering whether it might be wise to consider adding a central setting for this?

rootzoll, 2020-06-18:
If think having the option to connect a bitcoind remotely will make thinks complicated. Its maybe not hard to make it work with LND (changing some configs, etc) ... but there are a lot of places in the RaspiBlitz service scripts that just assume there is a local bitcoin client to call and blockchain data to work with.

And I dont see a big demand for that. The goal for people running a RaspiBlitz is to also run a Bitcoin Fullnode at home (because the dont have any bitcoin core node they trust/run yet). If you are already running a bitcoin node I think its much easier to setup the rest yourself manual like following a tutorial like RaspiBolt.

philbertw4, 2020-06-18:
Thanks for your reply. I think your assessment is more than reasonable. If the quantity/quality/ease of services available to a Raspiblitz user far exceeds those available to a standard Raspibolt/manual setup, then people may be willing to go ahead and run both or terminate the other node in favor of the Blitz.

closed 2020-06-19 by rootzoll

openoms, 2020-06-19:
From an other angle (not to speed up the sync time which is acceptable already) I am all for separating services physically. Agree that it would be a disproportionately big amount work compared to the demand/benefit.

I am considering looking into building a separate LND box (where there is the hot wallet sitting maximally isolated from other services) which then could be connected to a Bitcoin Core node (RaspiBlitz or else) similar to how the JoininBox https://github.com/openoms/joininbox project works with JoinMarket.

It could be a flexible and highly compatible a Lightning extension on an SBC.

Would likely be able to migrate the installation of most Lightning only services from the RaspiBlitz, but still it would make sense to make it a separate project.

rootzoll, 2020-06-19:
I can see the point of isolating LND as much as possible and it could make sense to make this as a seperate project that then connects to the RaspiBlitz as a service node.

> Would likely be able to migrate the installation of most Lightning only services from the RaspiBlitz, but still it would make sense to make it a separate project.

If security is the focus of the project seperating the LND, then possible additional apps should be very selective and mostly kept to CLI to reduce the attack vector.

philbertw4, 2020-06-21:
Thanks for the additional thoughts from both of you on this. It is helpful to see and understand the reasoning.

>Would likely be able to migrate the installation of most Lightning only services from the RaspiBlitz...
Yes, what @openoms is saying here is essentially the reason why I think it's worth considering.