raspiblitz/raspiblitz #1270: Can't restore LND

inpharmaticist, 2020-06-19:
Hello,

I had my node fail so I formatted and resynced my hard drive and trying to restore the LND file. I went through the process of restoring and it looked like it worked.

https://pastebin.com/irWWbfme

But it doesn't restore.

[image]

openoms, 2020-06-20:
Until the new SDcard release is out  it seesm you neeed to update LND separately first.

Set up a new wallet  (can be disposed later) and update LND from the update menu.

when update RESET the LND wallet and present your backup file as prompted by the menu.

Hope it solves the problem.

inpharmaticist, 2020-06-20:
Author check failed. Should I try the Reckless v0.10.2?

[image]

inpharmaticist, 2020-06-20:
I was able to install v0.10.2 but it does the same thing and asks to setup a new wallet :(

bucko13, 2020-06-22:
I'm also having this issue. Just started fresh with 1.5 and it kept on looping back to the LND Setup screen after loading up the backup file. So I followed the above instruction to start fresh first and then try and RESET LND to restore after updating to 1.5.1, but I got into the same loop again.

rootzoll, 2020-06-22:
Can you look in to the lnd-rescue file (unzip it) and check that there is a `/lnd/data/chain/bitcioin/mainnet/wallet.db`?

bucko13, 2020-06-22:
Well, I guess that explains it. The backup tarball is empty. Is there a known bug for the exporting of backups? I at least have the seed to backup from, but it's a little concerning that I followed the instructions for backup and ended up with something empty.

rootzoll, 2020-06-22:
It worked on my test all the time - but I will re-check and add in a plausibility check for filesize when the lnd-rescuefile gets created.  What was your filesize on the empty backup file?

inpharmaticist, 2020-06-22:
Shit. Mine is 45 bytes, also looks empty. But I can't find the backup seed.

rootzoll, 2020-06-22:
@inpharmaticist not having the backup seed is harsh ... do you have the old sd card? Or did you overwrite with new image?

bucko13, 2020-06-22:
Yep, also 45 bytes for me.

inpharmaticist, 2020-06-22:
> @inpharmaticist not having the backup seed is a had loss here ... do you have the old sd card? Or did you overwrite with new image?

I know im embarassed, usually better about that. I do have another SD that I think was used for this node but I thought that info was kept on the external HD? Which I overwrote after making the backup.

openoms, 2020-06-23:
Thwre is possibly an lnd-backup*.tar.gz on the sdcard.
You will only be able to read it on a Linux desktop (even when booting an Ubuntu Live USB)
 or boot the RPi with it and use the command line.
If you have the wallet.db you are in luck.

If YOU DID NOT RUN THE NODE since that last backup you can attempt to restore from the whole .tar.gz. If not sure just need to copy the wallet.db manually and use the SCB to get the offchain funds back.

inpharmaticist, 2020-06-23:
Where is the wallet.db file located?

openoms, 2020-06-23:
inside the compressed LND directory in the
lnd-backup*.tar.gz.

inpharmaticist, 2020-06-24:
The backup was empty, unless you mean there's one on the ssd?

rootzoll, 2020-06-24:
OK I added a check for file size when lnd rescue file gets created. Will be part of v1.6RC2.

inpharmaticist, 2020-06-24:
Hypothetically speaking, if I had the 24 word seed, and my peers force closed the channels, I could recover the funds on my side of the channels right?

rootzoll, 2020-06-24:
@inpharmaticist I think with the new channels there was a feature update that this now should work if you have honest channel partners - but I have never tested

inpharmaticist, 2020-06-24:
Thanks @openoms and @rootzoll. I lost 3.1 million sats, but I figure better it happens to me than someone else. I look forward to the next update. Cheers.

closed 2020-06-24 by inpharmaticist