raspiblitz/raspiblitz #1273: Enable multiwallet support for Bitcoin Core when activating the wallet

openoms, 2020-06-20:
Specter calls uses multiple additional bitcoin core wallets.

Jounmarket does not have a wallet specified currently do uses the default `wallet.dat`.

The problem is when Specter is running (with active wallets) the `wallet.dat` is not specified, but there are multiple entries in:  
`bitcoin-cli listwallets`

To specify a new wallet need to use multiwallet in Bitcoin Core: https://bitcoincore.org/en/releases/0.15.0/
Needs editing the bitcoin.conf an include at least one `wallet=` entry and restart.

Solution: 
will need to add  `wallet=wallet.dat` in the bitcoin.conf when the wallet feature is activated or the latest when joinmarket is installed.

Need to specify the `rpc_wallet_file = wallet.dat` in the `joinmarket.cfg` as well.

Related issues:
https://github.com/openoms/joininbox/issues/3
https://github.com/rootzoll/raspiblitz/issues/1162

rootzoll, 2020-06-29:
@openoms trying to get a RC2 ready this week ... should I wait for a PR on this?

openoms, 2020-06-29:
Will have a go at it on Wednesday, should be solved for v1.6.
Should be ok to test the many new features in the RC2 knowing that this is pending.

JoinMarket has updated too (should not affect the install process):
https://github.com/JoinMarket-Org/joinmarket-clientserver/blob/master/docs/release-notes/release-notes-0.6.3.1.md

openoms, 2020-06-30:
Tested that wallet.dat is always present and JM uses it by default.
Old JM configs will continue to work after recovery and no rescan is required.

rootzoll, 2020-06-30:
OK fix is merged :)

@openoms maybe last point: `bonus.cryptoadvance-specter.sh` is activating the btc wallet still by manually chaning the bitcoin conf .. should we change this also to use the `network.wallet.sh` script?

openoms, 2020-06-30:
> @openoms maybe last point: `bonus.cryptoadvance-specter.sh` is activating the btc wallet still by manually changing the bitcoin conf .. should we change this also to use the `network.wallet.sh` script?

can do for consistency, but Specter is not affected. It calls it's own wallets via RPC and does not rely on any specific file in the bitcoin directory.

rootzoll, 2020-06-30:
> can do for consistency, but Specter is not affected.

ah ok, then we can leave it as it is

openoms, 2020-06-30:
> > can do for consistency, but Specter is not affected.
> 
> ah ok, then we can leave it as it is

committed this directly https://github.com/rootzoll/raspiblitz/commit/5cdeb2ffa49856e253e6226f7e03958027399627  
 
TheOutcome is the same, but spares a bitcoind restart if the wallet is already active.

closed 2020-07-24 by rootzoll