raspiblitz/raspiblitz #1275: bitcoinErrorFull='error code: -28 error message: Verifying blocks...'

uninflatable, 2020-06-20:
Hi,
I setup my RaspiBlitz with version 1.4 and it worked fine. When I tried to install the "BTC RPC Explorer" the bitcoind service did not start any longer after rebooting the system. I got the advice to update to version 1.5 what I did - but the service still does not start.
Also I tried the repair option "Delete Blockchain & Re-Download" but this did not help too.
Please find the latest log file: https://termbin.com/sgf4
I looking forward for a hint how to resolve my issue.
Best regards
Martin

rootzoll, 2020-06-22:
There is some mention of bitcoind with Tor having problems connecting.
try command 'menu' > SERVICES > and deactivate TOR

uninflatable, 2020-06-27:
I deactivated TOR; after the reboot the terminal still shows the following message:

Blockchain Info ─────────────┐
                 │  The bitcoind service is not running.   │  
                 │  Login for more details     

please find attached the latest log-file: https://termbin.com/gf92

rootzoll, 2020-06-27:
Can you check on the `/home/bitcoin/.bitcoin/bitcoind.pid` file? Try to delete it and restart `sudo rm /home/bitcoin/.bitcoin/bitcoind.pid`

uninflatable, 2020-06-27:
I checked with `ls -a` if bitcoind.pid exists - this was/is not the case:

`admin@raspberrypi:/home/bitcoin/.bitcoin $ ls -la
total 608
drwxrwxrwx  5 bitcoin bitcoin   4096 Jun 27 21:50 .
drwxrwxrwx 13 root    root      4096 Jun 27 21:20 ..
-rw-------  1 bitcoin bitcoin     37 Jun 19 05:06 banlist.dat
-rwxr-xr-x  1 bitcoin bitcoin    506 Jun 27 10:44 bitcoin.conf
drwxr-xr-x  3 bitcoin bitcoin 122880 Jun 19 05:02 blocks
drwxr-xr-x  2 bitcoin bitcoin  69632 Jun 27 21:50 chainstate
-rw-------  1 bitcoin bitcoin 138935 Jun 27 21:50 debug.log
-rw-------  1 bitcoin bitcoin 247985 Jun 27 21:50 fee_estimates.dat
drwx------  3 bitcoin bitcoin   4096 Jun 19 05:07 indexes
-rw-------  1 bitcoin bitcoin      0 Jun  9 16:44 .lock
-rw-------  1 bitcoin bitcoin     17 Jun 27 21:50 mempool.dat
-rw-------  1 bitcoin bitcoin    820 Jun 27 21:46 onion_private_key
-rw-------  1 bitcoin bitcoin   5828 Jun 27 21:50 peers.dat
`

rootzoll, 2020-06-27:
try to turn off the index with `./config.scripts/network.txindex.sh off` and then `restart`

uninflatable, 2020-06-27:
this was helpful :-)
After I turned off the index and restarted the device, I could "Unlock the LNC Wallet" with password C.
The node is now syncing:
`Node is Syncing (7) ───────────┐
                 │  Blockchain Progress :  91.08 %         │  
                 │  Lightning Progress  :  99.68 %         │  
                 │  Please wait - this can take some time  `
When the node will run as expected, I will start the TOR service, restart again and send you a feedback

rootzoll, 2020-06-28:
So it looks like your blockchain data has maybe a small corruption that not allows you to build a valid txindex. If you need a txindex you might need to redownload the blockchain again - but if you dont need txindex=1 then you should be good.

I build a automatic detection for this into the new scripts for v1.6 - hope that this wil clear that up faster for other people running into that problem.

uninflatable, 2020-06-28:
thank you for your help - my node is working now as expected. TOR is also running and the addresses displayed in green on the status screen.

closed 2020-06-28 by uninflatable