raspiblitz/raspiblitz #1278: LND Update fails

AntonKanns, 2020-06-22:
Hello, 
when i try to Update LND 10.0.0 I get the following error message:
PGP Author Check failed

[image]

[image]

Can someone help me please?
Btw it is possible to install the Experimental LND update but the Optional LND Update leads to this error.

Greetings
Anton

openoms, 2020-06-23:
Can use this script instead: https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v0101-beta

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll