raspiblitz/raspiblitz #1282: Feature Request - OpenSSH update for Fido/U2F support

Silesia82, 2020-06-22:
Is it possible to update openSSH from 7.9 to at least 8.2
Why: The ssh 8.2 update supports fido/U2F so it is possible to use a physical device to log into the ssh session. It would be able to use a yubikey or a open source solokey instead of a password. This adds more security because there is no possibility to steal the password by keyloggers. 
https://solokeys.com
https://www.w3.org/2019/03/pressrelease-webauthn-rec.html.en

frennkie, 2020-06-22:
What is the significance on Fido/U2F here..? 

I already now use a YubiKey where my SSH private key is located that I use to ssh into the RaspiBlitz.

Silesia82, 2020-06-22:
Are you using a workaround?  Yubico OTP provides online validation with their Server. Maybe thats why it works for you with yubikey. I have searched a lot and found no information about fido/u2f with solokeys. The only thing i found was, that its supported in openssh since 8.2.
If there is another solution please link to it. 
Its my first week with solokeys and first month with raspiblitz :)

frennkie, 2020-06-22:
Have a look here: https://developers.yubico.com/PGP/SSH_authentication/

I would assume that we will not upgrade OpenSSH away from the default Debian/Rasbian Buster supported version. If 'upstream' (Debian) moves to 8.2 then this will be included obviously.

Silesia82, 2020-06-22:
Ok that makes sense. 
If it gets released it would be nice to add some user frindly support for this feature maybe.
Thanks for the link.

Silesia82, 2020-06-22:
> 
> Have a look here: https://developers.yubico.com/PGP/SSH_authentication/ I would assume that we will not upgrade OpenSSH away from the default Debian/Rasbian Buster supported version. If 'upstream' (Debian) moves to 8.2 then this will be included obviously.
> […](#)

yubikey is closed source and the solution may not work, also this is too complicated to set up for me, i dont want to mess up my node. It should work like this:

Generating key pair: ssh-keygen -t ecdsa-sk -f ~/.ssh/id_ecdsa_sk
enter password or no password and press button
Push key to server: ssh-copy-id -i ~/.ssh/id_ecdsa_sk.pub user@IP address

openoms, 2020-06-23:
Just to add here I am using an otherwise unused Ledger/Trezor to log in through ssh via the https://github.com/romanz/trezor-agent/blob/master/doc/README-SSH.md

Not too difficult to set up, but I wouldn't automate this because it is better to understand what is happening:
https://github.com/openoms/joininbox/blob/master/FAQ.md#log-in-through-ssh-using-a-hardware-wallet

Silesia82, 2020-06-24:
Ok, thanks so far. So we will have to wait till maybe 2021 when debian moves to 8.2 for a native support.
Maybe it is possible to add fido/u2f to the RTL webbrowser.
But i will close this "issue" (feature request) thank you all.

closed 2020-06-24 by Silesia82