raspiblitz/raspiblitz #1285: Installing v1.6.0rc1: RTL 502 Bad Gateway

adevoss, 2020-06-24:
After installing v1.6.0rc1 the web interface response is: 502 Bad Gateway
(https://localip:3001)

sudo netstat -tulpn|grep 3001:
tcp        0      0 0.0.0.0:3001            0.0.0.0:*               LISTEN      14821/nginx: master
tcp6       0      0 :::3001                 :::*                    LISTEN      14821/nginx: master

openoms, 2020-06-26:
RTL works ok for me.

Use the `https://RASPIBLITZ_LAN_IP:3001` instead of `localhost`.

test nginx with:
`sudo nginx -t`

adevoss, 2020-06-26:
By localip I meant I used the private IP address not the one from my ISP.

`sudo nginx -t`
`nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful`

error_rtl.log:
`2020/06/26 11:06:20 [error] 14822#14822: *22 connect() failed (111: Connection refused) while connecting to upstream, client: <PRIVATE_IP>, server: _, request: "GET /favicon.ico HTTP/1.1", upstream: "http://127.0.0.1:3000/favicon.ico", host: "<RASPIBLITZ_LAN_IP>:3001"`

In that log there is a upstream 127.0.0.1:3000
sudo netstat -tulpn|grep 3000
Nothing (Yes it returns BTCPayServer but that is port 23000)

openoms, 2020-06-26:
I have all ports active (together with the recently fixed BTCPay ports)
```
$ sudo netstat -tulpn|grep 300
tcp        0      0 127.0.0.1:23000         0.0.0.0:*               LISTEN      15727/BTCPayServer  
tcp        0      0 0.0.0.0:3001            0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 0.0.0.0:23001           0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 127.0.0.1:3002          0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 127.0.0.1:23002         0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 127.0.0.1:3003          0.0.0.0:*               LISTEN      20603/nginx: master 
tcp        0      0 127.0.0.1:23003         0.0.0.0:*               LISTEN      20603/nginx: master 
tcp6       0      0 :::3000                 :::*                    LISTEN      3126/node           
tcp6       0      0 :::3001                 :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 :::23001                :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 ::1:3002                :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 ::1:23002               :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 ::1:3003                :::*                    LISTEN      20603/nginx: master 
tcp6       0      0 ::1:23003               :::*                    LISTEN      20603/nginx: master 
```

is RTL running?
`sudo systemctl status RTL`
can benefit from a restart too.

It is likely that 1.6RC2 will sort this all out.

adevoss, 2020-06-26:
> is RTL running?
No.

Jun 26 14:00:36 -xxx- systemd[1]: Started RTL daemon.
Jun 26 14:00:37 -xxx- node[1444]: internal/modules/cjs/loader.js:960
Jun 26 14:00:37 -xxx- node[1444]:   throw err;
Jun 26 14:00:37 -xxx- node[1444]:   ^
Jun 26 14:00:37 -xxx- node[1444]: Error: Cannot find module 'express'
Jun 26 14:00:37 -xxx- node[1444]: Require stack:
Jun 26 14:00:37 -xxx- node[1444]: - /home/rtl/RTL/app.js
Jun 26 14:00:37 -xxx- node[1444]: - /home/rtl/RTL/rtl.js
Jun 26 14:00:37 -xxx- node[1444]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:957:15)
Jun 26 14:00:37 -xxx- node[1444]:     at Function.Module._load (internal/modules/cjs/loader.js:840:27)
Jun 26 14:00:37 -xxx- node[1444]:     at Module.require (internal/modules/cjs/loader.js:1019:19)
Jun 26 14:00:37 -xxx- node[1444]:     at require (internal/modules/cjs/helpers.js:77:18)
Jun 26 14:00:37 -xxx- node[1444]:     at Object.<anonymous> (/home/rtl/RTL/app.js:2:17)
Jun 26 14:00:37 -xxx- node[1444]:     at Module._compile (internal/modules/cjs/loader.js:1133:30)
Jun 26 14:00:37 -xxx- node[1444]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1153:10)
Jun 26 14:00:37 -xxx- node[1444]:     at Module.load (internal/modules/cjs/loader.js:977:32)
Jun 26 14:00:37 -xxx- node[1444]:     at Function.Module._load (internal/modules/cjs/loader.js:877:14)
Jun 26 14:00:37 -xxx- node[1444]:     at Module.require (internal/modules/cjs/loader.js:1019:19) {
Jun 26 14:00:37 -xxx- node[1444]:   code: 'MODULE_NOT_FOUND',
Jun 26 14:00:37 -xxx- node[1444]:   requireStack: [ '/home/rtl/RTL/app.js', '/home/rtl/RTL/rtl.js' ]
Jun 26 14:00:37 -xxx- node[1444]: }
Jun 26 14:00:37 -xxx- systemd[1]: RTL.service: Main process exited, code=exited, status=1/FAILURE
Jun 26 14:00:37 -xxx- systemd[1]: RTL.service: Failed with result 'exit-code'.

openoms, 2020-06-27:
reinstall RTL then

rootzoll, 2020-06-27:
@adevoss  also BEFORE yot deinstall RTL make a update on the v1.6 code with MAINMENU > UPDATE > PATCH > PATCHMENU > PATCH and then reinstall it

adevoss, 2020-06-27:
Patched - Reboot - Deinstall RTL - Reboot - Install RTL - Reboot
Same result

*** Run: npm install ***
npm WARN tarball tarball data for @angular/compiler@8.2.14 (sha512-ABZO4E7eeFA1QyJ2trDezxeQM5ZFa1dXw1Mpl/+1vuXDKNjJgNyWYwKp/NwRkLmrsuV0yv4UDCDe4kJOGbPKnw==) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for material-design-icons@3.0.1 (sha1-mnHEh0chjrylHlGmbaaCA4zct78=) seems to be corrupted. Trying one more time.

lots of lines like this:
npm WARN tar ENOENT: no such file or directory, open '/home/rtl/RTL/node_modules/.staging/@angular/material-d651ff7c/grid-list/package.json'
npm WARN tar ENOENT: no such file or directory, open '/home/rtl/RTL/node_modules/.staging/@angular/material-d651ff7c/icon/package.json'

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rtl/.npm/_logs/2020-06-27T17_16_31_824Z-debug.log
OK - RTL install looks good

/var/log/syslog:
....
Jun 27 19:20:19 xxx node[17157]: internal/modules/cjs/loader.js:960
Jun 27 19:20:19 xxx node[17157]:   throw err;
Jun 27 19:20:19 xxx node[17157]:   ^
Jun 27 19:20:19 xxx node[17157]: Error: Cannot find module 'express'
Jun 27 19:20:19 xxx node[17157]: Require stack:
Jun 27 19:20:19 xxx node[17157]: - /home/rtl/RTL/app.js
Jun 27 19:20:19 xxx node[17157]: - /home/rtl/RTL/rtl.js
Jun 27 19:20:19 xxx node[17157]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:957:15)
Jun 27 19:20:19 xxx node[17157]:     at Function.Module._load (internal/modules/cjs/loader.js:840:27)
Jun 27 19:20:19 xxx node[17157]:     at Module.require (internal/modules/cjs/loader.js:1019:19)
Jun 27 19:20:19 xxx node[17157]:     at require (internal/modules/cjs/helpers.js:77:18)
Jun 27 19:20:19 xxx node[17157]:     at Object.<anonymous> (/home/rtl/RTL/app.js:2:17)
Jun 27 19:20:19 xxx node[17157]:     at Module._compile (internal/modules/cjs/loader.js:1133:30)
Jun 27 19:20:19 xxx node[17157]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1153:10)
Jun 27 19:20:19 xxx node[17157]:     at Module.load (internal/modules/cjs/loader.js:977:32)
Jun 27 19:20:19 xxx node[17157]:     at Function.Module._load (internal/modules/cjs/loader.js:877:14)
Jun 27 19:20:19 xxx node[17157]:     at Module.require (internal/modules/cjs/loader.js:1019:19) {
Jun 27 19:20:19 xxx node[17157]:   code: 'MODULE_NOT_FOUND',
Jun 27 19:20:19 xxx node[17157]:   requireStack: [ '/home/rtl/RTL/app.js', '/home/rtl/RTL/rtl.js' ]
Jun 27 19:20:19 xxx node[17157]: }
Jun 27 19:20:19 xxx systemd[1]: RTL.service: Main process exited, code=exited, status=1/FAILURE
Jun 27 19:20:19 xxx systemd[1]: RTL.service: Failed with result 'exit-code'.
....

rootzoll, 2020-06-27:
hmm @saubyk can you see why we get a `Cannot find module 'express'` error here?

adevoss, 2020-06-27:
On a x86_64 VM RTL is installed OK

saubyk, 2020-06-28:
Tagging @shahanafarooqui

ShahanaFarooqui, 2020-06-28:
Usually these ‘tarballs data’ errors are thrown when npm has been corrupted for some reason. Please try following steps:

Go to your root RTL directory.
Delete ‘node_modules’ folder.
Delete package-lock.json (Optional step).
Run ‘npm cache verify’.
Run ‘npm install --only=prod’.

If above steps do not fix the issue then you might have to reinstall node and npm.

adevoss, 2020-06-28:
> Run ‘npm install --only=prod’.

(as rtl user)
Takes a long time...

openoms, 2020-06-28:
Signes of a dying SDcard there.

adevoss, 2020-06-29:
> Signes of a dying SDcard there.

Not using SD card. How can I check USB stick? With `fsck` or are there other things I can do?
Hardware is Pi4B/2G.

adevoss, 2020-06-29:
There is a web interface now. Can login. Looks fine (not browsed everything).
Completed in 5 hours with a few deprecated/dependencies warnings.

Also did `npm audit fix` because it is suggested at completion of install:

Cannot download "https://github.com/sass/node-sass/releases/download/v4.14.1/linux-arm-72_binding.node":
HTTP error 404 Not Found
...
> core-js@3.6.4 postinstall /home/rtl/RTL/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"
...
gyp verb `which` failed Error: not found: python2
...
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                                ^
gyp ERR! stack SyntaxError: invalid syntax
...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-sass@4.14.1 postinstall: `node scripts/build.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-sass@4.14.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

openoms, 2020-06-29:
> > Signes of a dying SDcard there.
> 
> Not using SD card. How can I check USB stick? With `fsck` or are there other things I can do?
> Hardware is Pi4B/2GB.

slowing down happens before becoming read only.
fsck is one but read-write/speed tests on desktop would show it better. Depends on your platform, plenty of tools are available.

See this related thread: https://github.com/rootzoll/raspiblitz/issues/924

adevoss, 2020-06-29:
@openoms Thank for the hint.
Using USB stick to boot maybe not such a good idea. USB stick is about a week old.
Other topic but just a thought. Will discuss this in #430 .

ShahanaFarooqui, 2020-06-29:
@adevoss we do not suggest to run audit fix at user’s end. We keep auditing/upgrading the dependencies with every release. But, fixing them at user’s end can result in unexpected behavior or broken UI.

adevoss, 2020-06-29:
@ShahanaFarooqui Will do these steps https://github.com/rootzoll/raspiblitz/issues/1285#issuecomment-650761583 again without audit fix. To be on v1.6.0rc1.

rootzoll, 2020-06-29:
@openoms I tested the RTL install multiple times - looks good for the v1.6 release. Anything still to do?

openoms, 2020-06-29:
I don't think RTL work well.

saubyk, 2020-06-29:
@openoms what's the issue?

openoms, 2020-06-29:
> @openoms what's the issue?

sorry, misleading words. Edited.
All good with RTL!

hermanlim, 2020-07-02:
> @openoms Thank for the hint.
> 
> Using USB stick to boot maybe not such a good idea. USB stick is about a week old.
> 
> Other topic but just a thought. Will discuss this in #430 .

@adevoss  on a side note,  you can boot Pi direct from SSD (USB) now if you are willing to tinker. 

I posted the link on another thread somewhere, will find it later. 

I believe the future versions of RB will use this function.

adevoss, 2020-07-06:
> I believe the future versions of RB will use this function.

Looking forward to that.

@hermanlim I boot from USB thumb drive now but had to change some RB scripts to do that. See my PR.

openoms, 2021-03-17:
closing this since solved

closed 2021-03-17 by openoms