raspiblitz/raspiblitz #1288: [Feature request] open the Electrum Wallet GUI from the RaspiBlitz with ssh -X

openoms, 2020-06-25:
Similar to how the JoinMarketQT GUI can be opened from the RaspiBlitz: https://github.com/openoms/bitcoin-tutorials/tree/master/joinmarket#joinmarketqt-gui-on-the-desktop

It is possible to run the Electrum GUI from the blitz too.
First successful try was:
installing Electrum 4.0.0b0 with this script: 
```
# Download
wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/electrs/electrum_wallet.sh 
# Run
bash electrum_wallet.sh
```
Opened the wallet to my Linux desktop with:

`ssh -X admin@192.168.3.161 /home/admin/.local/bin/electrum --oneserver --server localhost:50001:t`

Since activated Lightning in Electrum  the further tries give:
```

nyxnor, 2021-05-17:
Ping to remember :)

openoms, 2021-05-18:
There is this https://xpra.org/ to improve the quite sluggish performance of ssh -x. Needs to be installed on both the server and the desktop. Mentioned here: https://github.com/openoms/joininbox/issues/51

rootzoll, 2021-05-18:
@fusion44 with flutter for the LCD screen we would remove X11 and this would not work anymore, correct?

openoms, 2021-05-19:
Similar feature is allowing to use the JoinMarket QT GUI via ssh -x. Some people use it so will need to research alternatives if the x-server cannot coexist with flutter.

fusion44, 2021-05-19:
We might have options as there are actually two versions of Flutter available for the PI. 

**A: Flutter-PI**
This one uses the DRI directly without X and thus is much leaner. The only dependency this one (besides Flutter itself) is the Linux Kernel and libinput from Wayland (input processing). I think this can actually run if the remote X server doesn't block the DRI but it seems [possible](https://linuxconfig.org/how-to-run-x-applications-without-a-desktop-or-a-wm).

**B: Official Google Flutter distribution**
This one uses X and might be more "universal" but comes with all dependencies to make it work with X.

Option A would be awesome as we can keep the Blitz leaner this way. If we cannot make it work, we can always use option B. From an application development perspective it doesn't change much whether we go with A or B.