raspiblitz/raspiblitz #1298: [Testing] LND wallet recovery cannot be exited

openoms, 2020-06-28:
Looping through:
[image]
[image]
[image]
[image]

Solution: make `Cancel` functional

openoms, 2020-06-28:
Also fails (on Odroid C4 - Ubuntu - pytton 3.8):
[image]
[image]
[image]
[image]  
Pasted the 24words
[image]
[image]
[image]

rootzoll, 2020-06-29:
OK should be fixed.

closed 2020-07-24 by rootzoll