raspiblitz/raspiblitz #1302: SCB: Local backup and timestamping

dennisreimann, 2020-06-30:
Just a question if these things are wanted, because I'd like to extend the background checks with the following:

* A `localBackupTarget` option (in addition to `scpBackupTarget`) to save the SCB file to a USB drive/stick
* Timestamp the backups, so that they do not get overwritten (e.g. `channel-20200630-0615000.backup`)

rootzoll, 2020-06-30:
> A localBackupTarget option (in addition to scpBackupTarget) to save the SCB file to a USB drive/stick

Good idea :)

> Timestamp the backups, so that they do not get overwritten (e.g. channel-20200630-061500.backup)

Would add more resellience and because a new file is not created that often should work for all backup targets. Its to be kept in mind that this also then has to work on the technical side, when people upload the channel.backup file again to the RaspiBlitz.

dennisreimann, 2020-06-30:
Alright, I‘ll prepare a PR. Thanks for the pointer regarding the restore process. 👍

rootzoll, 2020-07-04:
merged #1305 and done in #1309

PR #1305 (2020-07-03): SCB: Local backup and timestamping
Addresses and closes #1302. Currently the restore part is missing, I'll keep it as a draft until that is implemented.
closed 2020-07-04 by rootzoll