raspiblitz/raspiblitz #1306: transaction lost

zopedope, 2020-06-30:
Hi,
I connected and sent a transaction from fullynoded to my raspiblitz. After upgrading the raspiblitz to 1.5 it doesn't show up in my fullynoded anymore and I can't find my bitcoin.

Can you help explain what I did wrong?
Can I get/recover that transaction out of the "wallet.dat" file?
I don't seem to have permissions to "scp" the wallet.dat file... would that even help?

Your help is very appreciated.

openoms, 2020-07-01:
Please specify what you mean by:
>sent a transaction from fullynoded to my raspiblitz

If you used the menu to fund it was sent to the LND wallet.
For a GUI for your LND wallet try RTL.

For FullyNoded related questions: https://t.me/FullyNoded

zopedope, 2020-07-01:
thanks for the reply.
So I had fullynoded connected to my raspblitz 1.4 (cant seem to connect them anymore) and used fullynoded to create an address to send to. I assume that was to core not to lnd...
Was the core wallet enabled before 1.5 and now disabled (or wiped)? 

I definitely dont understand how this all works so any help in understanding this is appreciated.

openoms, 2020-07-01:
I don't use Fully Noded myself so can't be much help. Best to ask the devs in their Telegram channel.

When you created the wallet you must have had a seed, so you can recover the wallet from that.

To switch the bitcoind wallet back on the blitz and connect use the 
connect Fully noded option
in the `MOBILE` menu.

openoms, 2020-07-09:
Did you have this sorted?

zopedope, 2020-10-29:
yes. thanks.

closed 2020-10-29 by zopedope