raspiblitz/raspiblitz #1311: [1.5.1] The bitcoind service is not running.

ferologics, 2020-07-02:
I tried to connect the Fully Noded app to my node but something wrong happened and now I'm stuck verifying the blocks.

```
*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

15 RESTARTS DETECTED - bitcoind might be in a error loop
1593707430 ERROR: Error found in Logs

More Error Detail:
error code: -28 error message: Loading block index...

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now
```

When I `sudo tail -f /mnt/hdd/bitcoin/debug.log` the following just keeps happening on and on.
```
2020-07-02T16:24:49Z Bitcoin Core version v0.19.1 (release build)
2020-07-02T16:24:49Z InitParameterInteraction: parameter interaction: -bind set -> setting -listen=1
2020-07-02T16:24:49Z InitParameterInteraction: parameter interaction: -proxy set -> setting -upnp=0
2020-07-02T16:24:49Z InitParameterInteraction: parameter interaction: -proxy set -> setting -discover=0
2020-07-02T16:24:49Z Assuming ancestors of block 00000000000000000005f8920febd3925f8272a6a71237563d78c2edfdd09ddf have valid signatures.
2020-07-02T16:24:49Z Setting nMinimumChainWork=000000000000000000000000000000000000000008ea3cf107ae0dec57f03fe8
2020-07-02T16:24:49Z Using the 'standard' SHA256 implementation
2020-07-02T16:24:49Z Default data directory /home/bitcoin/.bitcoin
2020-07-02T16:24:49Z Using data directory /mnt/hdd/bitcoin
2020-07-02T16:24:49Z Config file: /home/bitcoin/.bitcoin/bitcoin.conf
2020-07-02T16:24:49Z Using at most 40 automatic connections (1024 file descriptors available)
2020-07-02T16:24:49Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2020-07-02T16:24:49Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2020-07-02T16:24:49Z Using 4 threads for script verification
2020-07-02T16:24:49Z HTTP: creating work queue of depth 16
2020-07-02T16:24:49Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2020-07-02T16:24:49Z HTTP: starting 4 worker threads
2020-07-02T16:24:49Z Using wallet directory /mnt/hdd/bitcoin
2020-07-02T16:24:49Z init message: Verifying wallet(s)...
2020-07-02T16:24:49Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2020-07-02T16:24:49Z Using wallet /mnt/hdd/bitcoin
2020-07-02T16:24:49Z BerkeleyEnvironment::Open: LogDir=/mnt/hdd/bitcoin/database ErrorFile=/mnt/hdd/bitcoin/db.log
2020-07-02T16:24:49Z scheduler thread start
2020-07-02T16:24:49Z init message: Loading banlist...
2020-07-02T16:24:49Z Cache configuration:
2020-07-02T16:24:49Z * Using 2.0 MiB for block index database
2020-07-02T16:24:49Z * Using 15.8 MiB for transaction index database
2020-07-02T16:24:49Z * Using 8.0 MiB for chain state database
2020-07-02T16:24:49Z * Using 102.2 MiB for in-memory UTXO set (plus up to 61.0 MiB of unused mempool space)
2020-07-02T16:24:49Z init message: Loading block index...
2020-07-02T16:24:49Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2020-07-02T16:24:49Z Opened LevelDB successfully
2020-07-02T16:24:49Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2020-07-02T16:25:16Z LoadBlockIndexDB: last block file = 2136
2020-07-02T16:25:16Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=59, size=71227421, heights=637297...637355, time=2020-07-02...2020-07-02)
2020-07-02T16:25:16Z Checking all blk files are present...
2020-07-02T16:25:17Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2020-07-02T16:25:17Z Opened LevelDB successfully
2020-07-02T16:25:17Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: eadc35e843c5834c
2020-07-02T16:25:17Z Loaded best chain: hashBestChain=0000000000000000000416c64df2b418bbd26fc83c8a9b5e3b0b54e77ed6ee36 height=637355 date=2020-07-02T15:35:09Z progress=0.999979
2020-07-02T16:25:17Z init message: Rewinding blocks...
2020-07-02T16:25:18Z init message: Verifying blocks...
2020-07-02T16:25:18Z Verifying last 6 blocks at level 3
2020-07-02T16:25:18Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2020-07-02T16:25:38Z No coin database inconsistencies in last 6 blocks (15965 transactions)
2020-07-02T16:25:38Z  block index           48962ms
2020-07-02T16:25:38Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2020-07-02T16:25:38Z Opened LevelDB successfully
2020-07-02T16:25:38Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2020-07-02T16:25:38Z txindex thread start
2020-07-02T16:25:38Z Syncing txindex with block chain from height 22
2020-07-02T16:25:38Z init message: Loading wallet...
2020-07-02T16:25:38Z BerkeleyEnvironment::Open: LogDir=/mnt/hdd/bitcoin/database ErrorFile=/mnt/hdd/bitcoin/db.log
2020-07-02T16:25:39Z [default wallet] Wallet File Version = 169900
2020-07-02T16:25:39Z [default wallet] Keys: 2001 plaintext, 0 encrypted, 2001 w/ metadata, 2001 total. Unknown wallet records: 0
2020-07-02T16:25:39Z [default wallet] Wallet completed loading in             294ms
2020-07-02T16:25:39Z [default wallet] setKeyPool.size() = 2000
2020-07-02T16:25:39Z [default wallet] mapWallet.size() = 0
2020-07-02T16:25:39Z [default wallet] mapAddressBook.size() = 0
2020-07-02T16:25:39Z block tree size = 637383
2020-07-02T16:25:39Z nBestHeight = 637355
2020-07-02T16:25:39Z torcontrol thread start
2020-07-02T16:25:39Z Bound to 127.0.0.1:8333
2020-07-02T16:25:39Z init message: Loading P2P addresses...
2020-07-02T16:25:39Z ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex(pprev=0x5e27cf8, nHeight=73, merkle=41b48c64cba68c21e0b7b37f589408823f112bb7cbccef4aece29df25347ffb4, hashBlock=000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d) at FlatFilePos(nFile=0, nPos=16371)
2020-07-02T16:25:39Z *** ThreadSync: Failed to read block 000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d from disk
2020-07-02T16:25:39Z Error: Error: A fatal internal error occurred, see debug.log for details
2020-07-02T16:25:39Z txindex thread exit
2020-07-02T16:25:39Z Leaving InitialBlockDownload (latching to false)
2020-07-02T16:25:39Z tor: Got service ID wn2zsec5cx6ktdcl, advertising service wn2zsec5cx6ktdcl.onion:8333
2020-07-02T16:25:39Z AddLocal(wn2zsec5cx6ktdcl.onion:8333,4)
2020-07-02T16:25:39Z Loaded 6172 addresses from peers.dat  151ms
2020-07-02T16:25:39Z init message: Starting network threads...
2020-07-02T16:25:39Z DNS seeding disabled
2020-07-02T16:25:39Z addcon thread start
2020-07-02T16:25:39Z init message: Done loading
2020-07-02T16:25:39Z net thread start
2020-07-02T16:25:39Z tor: Thread interrupt
2020-07-02T16:25:39Z Shutdown: In progress...
2020-07-02T16:25:39Z RemoveLocal(wn2zsec5cx6ktdcl.onion:8333)
2020-07-02T16:25:39Z torcontrol thread exit
2020-07-02T16:25:39Z msghand thread start
2020-07-02T16:25:39Z msghand thread exit
2020-07-02T16:25:39Z opencon thread start
2020-07-02T16:25:39Z Socks5() connect to fno4aakpl6sg6y47.onion:8333 failed: InterruptibleRecv() timeout or other failure
2020-07-02T16:25:39Z opencon thread exit
2020-07-02T16:25:39Z addcon thread exit
2020-07-02T16:25:39Z net thread exit
2020-07-02T16:25:39Z scheduler thread interrupt
2020-07-02T16:25:39Z [default wallet] Releasing wallet
2020-07-02T16:25:39Z Shutdown: done
```

I tried rebooting the device but it didn't help.

Find more logs here:
`/home/admin/XXdebugLogs.sh` 👉 https://termbin.com/do5d
`/mnt/hdd/bitcoin/debug.log ` 👉  https://termbin.com/waa3
`/var/log/syslog` 👉 https://termbin.com/4xnc
And the configuration `/home/bitcoin/.bitcoin/bitcoin.conf` 👉
```
# bitcoind configuration

# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=0

# Connection settings
rpcuser=raspibolt
rpcpassword=perplexing
main.rpcport=8332
rpcallowip=127.0.0.1
main.rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=128
maxorphantx=10
maxmempool=64
maxconnections=40
maxuploadtarget=5000

datadir=/mnt/hdd/bitcoin
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
main.addnode=fno4aakpl6sg6y47.onion
main.addnode=toguvy5upyuctudx.onion
main.addnode=ndndword5lpb7eex.onion
main.addnode=6m2iqgnqjxh7ulyk.onion
main.addnode=5tuxetn7tar3q5kp.onion
main.addnode=juo4oneckybinerq.onion
test.rpcport=18332
test.rpcbind=127.0.0.1:18332
```

hermanlim, 2020-07-02:
Are you in a position to disable TOR? It appears to be failing consequent to it.

ferologics, 2020-07-02:
Not ideal but I’ll try that, thanks!

I was under the impression that the failing part is the 

```
2020-07-02T16:25:39Z ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex(pprev=0x5e27cf8, nHeight=73, merkle=41b48c64cba68c21e0b7b37f589408823f112bb7cbccef4aece29df25347ffb4, hashBlock=000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d) at FlatFilePos(nFile=0, nPos=16371)
2020-07-02T16:25:39Z *** ThreadSync: Failed to read block 000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d from disk
2020-07-02T16:25:39Z Error: Error: A fatal internal error occurred, see debug.log for details
2020-07-02T16:25:39Z txindex thread exit
```

Might’ve been wrong tho.

ferologics, 2020-07-03:
Okay, this did change the error message from hashes not matching to `Deserialize or I/O error - ReadCompactSize(): size too large: iostream error`

```
2020-07-03T08:31:02Z Verifying last 6 blocks at level 3
2020-07-03T08:31:02Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2020-07-03T08:35:57Z No coin database inconsistencies in last 6 blocks (15965 transactions)
2020-07-03T08:35:57Z  block index          325753ms
2020-07-03T08:35:57Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2020-07-03T08:35:57Z Opened LevelDB successfully
2020-07-03T08:35:57Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2020-07-03T08:35:57Z txindex thread start
2020-07-03T08:35:57Z Syncing txindex with block chain from height 667
2020-07-03T08:35:57Z init message: Loading wallet...
2020-07-03T08:35:57Z BerkeleyEnvironment::Open: LogDir=/mnt/hdd/bitcoin/database ErrorFile=/mnt/hdd/bitcoin/db.log
2020-07-03T08:35:57Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=0, nPos=154303)
2020-07-03T08:35:57Z *** ThreadSync: Failed to read block 000000002c366a1d494cfe69187059aa125ba410b9779ac6dcda38d7e6711fbe from disk
2020-07-03T08:35:57Z Error: Error: A fatal internal error occurred, see debug.log for details
2020-07-03T08:35:57Z txindex thread exit
```

Looks related to #1169. I did download the blockchain via torrent but it was working fine until I tried to connect the FullyNoded app. I'll try the fixes mentioned in the other issue.

ferologics, 2020-07-03:
FYI `powerWARN=0` after I ran the `REPAIR > SOFTWARE` flow via main menu.

ferologics, 2020-07-03:
I went ahead with the `REPAIR > RESET-HDD` option and re-flashed with 1.5.1. Now I need to setup again and then probably sync?, hopefully I won't have to re-download the blockchain. Not using Tor this time around.

rootzoll, 2020-07-03:
If you had "txindex" on (was seeing `txindex thread exit`in logs) it can be that there is some corruption in your blockchain. So to solve this you would need to trun off txindex again and restart (redownload blockchain if you really need txindex).

 How did you download the blockchain in the first place?

ferologics, 2020-07-03:
Small status update: I'm back up and running. 
Without Tor enabled at the moment ofc.

What I did 👇 
1. `REPAIR > RESET-HDD`
2. `re-flashed with 1.5.1`
3. `Recovered my LND wallet`
4. `Synced up with mainnet`

Re: 
> How did you download the blockchain in the first place?

@rootzoll I downloaded the blockchain via torrent. I had no issues during the download or sync, though there was a power outage that happened around ~98% sync progress. I just started the device and it continued the sync like nothing happened.

Also, I was already up and running when I tried to connect the `FullNode` app. I'm afraid that the same thing will happen again if I try to connect the app again.

It's uncertain what role Tor played in the above issue. It might not be related at all. I will try to enable it again and come back with the result.

As for `txindex` I probably don't need it. If the same issue happens again I could turn just it off.

ferologics, 2020-07-03:
I am running just fine with Tor enabled as well. The blockchain was in tact and no disk corruption occurred. Looks like the reinstall removed some erroneous state on the Pi.

ferologics, 2020-07-03:
I managed to reproduce the issue by attempting to connect my `FullyNoded` iOS app to my node. The same error occurred just as I stated at the start of the issue. 

1. `ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex`
2. The `FullyNoded` app is stuck on `Error verifying blocks...`

[image][image]

I will now attempt to disable the `txindex` option.

ferologics, 2020-07-03:
@rootzoll I disabled the `txindex` and now the `bitcoind` service is running just fine.
Also the `FullyNoded` app is working. 🎉 Good catch! thx 🙏 

I don't need to use the `txindex` at the moment. What about the `size too large: iostream error`?
Do you have any ideas what would be causing this issue? Not enough RAM?

rootzoll, 2020-07-04:
> Do you have any ideas what would be causing this issue?

Could be that the blocks in the torrent had a small data curruption. Working for normal use but failing when txindexing.

closed 2020-07-04 by rootzoll