raspiblitz/raspiblitz #1314: SSH Display changed. Lots of Qs and Xs

inpharmaticist, 2020-07-04:
I spun up a new node recently. The only things I've done with it are install electrs, stopping electrs, opening two channels, and adding two lines to the bitcoin.conf file (deprecatedrpc=size and peerbloomfilters=1).  This morning I noticed this.

[image]

rootzoll, 2020-07-04:
@openoms do you have the details on how to change the charset on windows?

frennkie, 2020-07-04:
@inpharmaticist can you test this: https://github.com/rootzoll/raspiblitz/issues/1200#issuecomment-636169777

openoms, 2020-07-04:
It is the dialog program playing funny with Putty.
Would recommend using openessh frombthe windows store with PowerShell: https://github.com/rootzoll/raspiblitz/issues/1099

inpharmaticist, 2020-07-04:
> It is the dialog program playing funny with Putty.
> Would recommend using openssh from the windows store with PowerShell: #1099

I wasn't using putty. I was using the terminal in linux mint

closed 2020-07-04 by rootzoll

openoms, 2020-07-04:
check for any `.dialog` files in the `/home/admin/` dir. Can stay from an unclear shutdown. Should be back to teh normal (default) settings if those are deleted.

inpharmaticist, 2020-07-05:
Not seeing any.

[image]

openoms, 2020-07-05:
> Not seeing any.

Use `ls -la` to show the hidden files.

inpharmaticist, 2020-07-05:
Still no.

https://pastebin.com/caVVqbnE

openoms, 2020-07-06:
Don't know where the problem is. Try another terminal app or device. Also flashing the sdcard should solve it if the problem is on the blitz filesystem.

b-t-e, 2021-08-21:
Had the same issue with PuTTY since installing raspiblitz (1.7.0) a few weeks ago.
Fixed it today with these settings under Window | Translation:
- Remote character set: **UTF-8**
- Handling of line drawing characters: **Use Unicode line drawing code points**
- Enable VT100 line drawing even in UTF-8 mode: **ON**

[image]