raspiblitz/raspiblitz #1315: Integrate DynDNS/NoIP Services that also support ACME

frennkie, 2020-07-05:
There are at least two (free) services that offer DynDNS like features which seem to support ACME:

* https://freedns.afraid.org/faq/#17
* https://www.duckdns.org/domains

rootzoll, 2020-07-05:
Make a "HTTPS-Domain"-Subscription out of this - you dont choose a shop here - but you have DuckDNS and FreeDNS here as free options. Let the user enter the needed credentials and then trigger the ACME script accordently and exchange the cert nginx.

frennkie, 2020-07-12:
"Bad" news: https://github.com/acmesh-official/acme.sh/issues/2305#issuecomment-529165181

> I ask because the dns_freedns plugin will not work with a "public" domain shared by FreeDNS... it will only work with a domain that you own.

This - kind of - defeats the idea to require almost "nothing" from our users. I will focus on Duckdns.

frennkie, 2020-08-06:
@rootzoll please be aware that the DuckDNS API does not work (with `acme.sh`) when more than one name is to be included in the certificate.

E.g. `raspiblitz.duckdns.org` will work, but `myraspiblitz.duckdns.org` + `raspiblitz.duckdns.org` will not.

The cause for this is that `acme.sh` tries to validate using different `TXT records`, but DuckDNS doesn't support multiple different `TXT records` to be set. 

For our current setup this is just something to keep in mind (and maybe point out to users). But my idea was to always set the primary subdomain and a wildcard: `raspiblitz.duckdns.org` + `*.raspiblitz.duckdns.org`. Unfortunately this won't work like that.

Also bare in mind that the certificate for `*.raspiblitz.duckdns.org` will not considered valid if you point your browser to `raspiblitz.duckdns.org`.

frennkie, 2020-08-08:
https://github.com/acmesh-official/acme.sh/wiki/dnsapi

rootzoll, 2020-09-16:
@frennkie ok I added dynu.com to for the v1.6.1 release - works fine on my tests with ACME and LetsEncrypt. Marking for final testing.

rootzoll, 2020-10-08:
Tested it a few more times and works well. Will be part of v1.6.1 release. Cosing issue.

closed 2020-10-08 by rootzoll