raspiblitz/raspiblitz #1317: Electrs not running

inpharmaticist, 2020-07-05:
I tried adding the Electrs from the services menu screen. After a while, I noticed the LCD said it wasn't running and to run `sudo journalctl -u electrs`. From the output, it looks like it just keeps starting and stopping electrs. Here are the first 5 instances of that.

https://pastebin.com/6Er2AbjJ

Selecting Electrum Rust Server from the ssh menu has the following output
>/home/admin/config.scripts/bonus.electrs.sh: line 111: /home/admin/XXdebugInfo.sh: No such file or directory

openoms, 2020-07-06:
```
ERROR - server failed: Error: getblock RPC error: {"code":-1,"message":"Block not found on disk"}
```
these errors suggest that there is disk.corruption.
Best to change the disk of you have an other one.
Will probably need to reset the blockchain from the REPAIR menu and redownload/copy it over again.

inpharmaticist, 2020-07-06:
Im using a brand new SSD so that would be surprising. But I did manually copy over the blockchain from my laptop node. I wonder if I did something wrong there.

inpharmaticist, 2020-08-07:
Resynced the blockchain. Electrs is now indexed.

closed 2020-08-07 by inpharmaticist