raspiblitz/raspiblitz #1318: ok..... tried the new 1.6rc1 (crap thought it was rc2) FAILED

DrBlackross, 2020-07-05:
replaced the 1tb barracuda, with a 500gb sandisk thinking that might have been the issue for a few days. but i still end with.

Generating new Wallet ....
/home/admin/70initLND.sh: line 200: /home/admin/python3-env-lnd/bin/activate: No such file or directory
Traceback (most recent call last):
  File "/home/admin/config.scripts/lnd.initwallet.py", line 239, in <module>
    main()
  File "/home/admin/config.scripts/lnd.initwallet.py", line 221, in main
    stub = lnrpc.WalletUnlockerStub(channel)
AttributeError: module 'lndlibs.rpc_pb2_grpc' has no attribute 'WalletUnlockerStub'
FAIL!! -> MISSING seedwords data - but also no err data ?!?
CHECK output data above - PRESS ENTER to restart 70initLND.sh

have rpi4
case https://smile.amazon.com/dp/B07GRZRW7J/ref=cm_sw_em_r_mt_dp_U_K5Z-Eb432NR7G
10a power supply (from old miner)
and the rpi4 adapter https://smile.amazon.com/dp/B073CGQD9C/ref=cm_sw_em_r_mt_dp_U_Z5Z-EbZHWKPR2

DOH!!!

you guys just posted RC2 LMAO!!! ok give me a sec, lets see if can break this release.

(then i'll continue this post maybe)

rootzoll, 2020-07-05:
Please report if you still have that problem with the RC2 - thanks.

DrBlackross, 2020-07-05:
yea biggest sign of when it fails (usually) is that 9gb limit it hits, right now its syncing sooo we'll see lol

so far so good

DrBlackross, 2020-07-07:
ok been running for 2 days let it sit after setting up then it did its final reboot....

hasn't gone above the 9gb mark, and after logging in i get this.

*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
error code: -28 error message: Loading block index...

-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

Use CTRL+c to EXIT to Terminal
bitcoin error: error code: -28
error message:
Loading block index...

--------------------------------
current space on drives
--------------------------------

devtmpfs        459M     0  459M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   47M  417M  11% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
tmpfs            32M     0   32M   0% /var/cache/raspiblitz
/dev/mmcblk0p1  253M   52M  202M  21% /boot
/dev/sda1       458G  9.8G  425G   3% /mnt/hdd
tmpfs            93M     0   93M   0% /run/user/1000
tmpfs            93M     0   93M   0% /run/user/1001

DrBlackross, 2020-07-07:
btw this was with a new 500gb hd (not the seagate 1tb barracuda) clean rebuild on a rpi4 20a power supply for that case.

DrBlackross, 2020-07-07:
yea have retried resetting the blockchain and still hits 9gb and stops :(

DrBlackross, 2020-07-07:
ok output from debug

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.0rc1
chainnetwork: bitcoin / main
 02:29:29 up 22:43,  2 users,  load average: 1.09, 1.26, 1.22

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Tue 2020-07-07 02:29:15 BST; 13s ago
  Process: 21167 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 21169 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 21170 (code=exited, status=1/FAILURE)

Jul 07 02:29:15 NuggetNode systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 414.
Jul 07 02:29:15 NuggetNode systemd[1]: Stopped Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2020-07-06 21:51:27 BST, end at Tue 2020-07-07 02:29:29 BST. --
Jul 07 02:26:01 NuggetNode systemd[1]: Stopped Bitcoin daemon.
Jul 07 02:26:18 NuggetNode systemd[1]: Starting Bitcoin daemon...
Jul 07 02:26:18 NuggetNode systemd[1]: Started Bitcoin daemon.
Jul 07 02:28:45 NuggetNode systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE
Jul 07 02:28:45 NuggetNode systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Jul 07 02:29:15 NuggetNode systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.
Jul 07 02:29:15 NuggetNode systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 414.
Jul 07 02:29:15 NuggetNode systemd[1]: Stopped Bitcoin daemon.
1594085359 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-07-07T01:28:42Z Rolling forward 000000000000091ed8f7ecbe913c26fb6d80f1fa2f25e0c2d920db19c1cef218 (143941)
2020-07-07T01:28:42Z Rolling forward 00000000000002cdc4d752dedbe23dc4aa2969de78449c4ccb25bdf9c3041233 (143942)
2020-07-07T01:28:42Z Rolling forward 00000000000009668c9a6215dbf5ab58b808e1c740db51058dc0c90749d7544b (143943)
2020-07-07T01:28:42Z Rolling forward 000000000000012c7408410d72f8dc6bf29856f73102e87f75ac319f75399ae3 (143944)
2020-07-07T01:28:42Z Rolling forward 0000000000000544bb9fb4f3822ca9fa40d260ce13f8c1c76f4f25bc5c28cc8e (143945)
2020-07-07T01:28:42Z Rolling forward 00000000000003332e4088e99a7db1e5e0b75bf5b1b09a9efd799a5920b3785f (143946)
2020-07-07T01:28:42Z Rolling forward 00000000000002819bcb0b1c1e1bcfcbb3c0b8e31e0859e027daf806a1fa7cea (143947)
2020-07-07T01:28:42Z Rolling forward 00000000000005712f342c791ce9694b88f85def6089b2562576848bc1505ac2 (143948)
2020-07-07T01:28:42Z Rolling forward 0000000000000581357da7db47f24d119f8fe2520da8a0561d624572cd149f6d (143949)
2020-07-07T01:28:42Z Rolling forward 00000000000002bc3ab04e8a83a1497dc219bf08c2c3b6571c9752aa174309a5 (143950)
2020-07-07T01:28:42Z Rolling forward 00000000000008c4e359903109f823e477ac88c008c12225806de2c84f502500 (143951)
2020-07-07T01:28:42Z Rolling forward 0000000000000363fb10aad8f70d1d680e4e09be1e8fafc02d0a78455337a745 (143952)
2020-07-07T01:28:42Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=4, nPos=30815544)
2020-07-07T01:28:42Z ERROR: ReplayBlock(): ReadBlockFromDisk failed at 143952, hash=0000000000000363fb10aad8f70d1d680e4e09be1e8fafc02d0a78455337a745
2020-07-07T01:28:43Z : Unable to replay blocks. You will need to rebuild the database using -reindex-chainstate..
Please restart with -reindex or -reindex-chainstate to recover.
2020-07-07T01:28:43Z Aborted block database rebuild. Exiting.
2020-07-07T01:28:43Z Shutdown: In progress...
2020-07-07T01:28:43Z scheduler thread exit
2020-07-07T01:28:43Z Shutdown: done

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-07-06 03:46:57 BST; 22h ago
  Process: 1205 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 1207 (lnd)
    Tasks: 14 (limit: 2200)
   Memory: 8.9M
   CGroup: /system.slice/lnd.service
           └─1207 /usr/local/bin/lnd --externalip=24.126.176.78:9735

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST LND ERROR LOGS ***
[... 52 lines cut ...]
*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.12'
tempCelsius='49.9'
uptime=81819
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

openoms, 2020-07-07:
You have a disk error  = hardware fault.
It can be either a power supply issue. You write 20 A , well 3 A is the required minmum so if you use 2 that is not enough.
Would suggest to format the disk (to any file system eg, NTFS) on your desktop and start with a clean SD as completely new.
Make sure to have a min 3A power supply.

This adapter: https://smile.amazon.com/dp/B073CGQD9C/ref=cm_sw_em_r_mt_dp_U_Z5Z-EbZHWKPR2 is for the RPi3 so I presume you ahve the X825 version for the RPi4.

DrBlackross, 2020-07-07:
i have 20a supply from an old miner works fine, this with 2 separate hdds, 1 was 1tb ssd, this is a 500gb hdd. ive reinstalled this enough over 4 months it just gets to 8gb or 9.8gb and stops syncing, on 2 totally different hard disks. 

crazy

install with the tor option this time thinking maybe if i used onion i wouldn't get the issues thinking maybe comcast was causing headaches. 

but i'll strip it down to usb hdd connection rpi4 (i dont use 3's) a different power supply and sdcard (basic) and see if it works.

got 3 of these 
 https://www.amazon.com/dp/B00L88M8TE/ref=cm_sw_em_r_mt_dp_U_ZGpbFbFVZYQFJ
(ill use one)

openoms, 2020-07-08:
> got 3 of these 
>  https://www.amazon.com/dp/B00L88M8TE/ref=cm_sw_em_r_mt_dp_U_ZGpbFbFVZYQFJ
> (ill use one)
> 

These have microUSB and the RPi4 has USB C.
A stable PSU is a must. Disks are much more sensitive than miners.

Be aware of the UASP issues with disk adapters.

DrBlackross, 2020-07-08:
It's a quad core rpi with microusb

openoms, 2020-07-08:
Do you mean that you are not using a Raspberry Pi 4?
The prebuilt image is not compatible with other Singke Board Computers.

DrBlackross, 2020-07-08:
thought i was, either way i can run over to microcenter and pick one up today, but i will strip everything down to basics and figure out if its the case power supply or something...

(this has been something to do at home during covid, sooo its not a problem to beat the crap out of something)

DrBlackross, 2020-07-08:
you said it could be a power issue... 
[image]

gonna try the usb 3 power on the side of the adapter to power the rpi (which is rpi 3 b+ (after i put my reading glasses on)), see if that give it a enough power.

i mean its a good case, i use them for NAS and media servers (and 1 minecraft server for the kids), but there's a barrel jack on the back for 5v input (microusb is flimsy). soooo i will use the usb 3.0 blue port under the hdmi to power the rpi instead of pulling 5v from the header inside, LOL.

and let it sit 

OH, do u think raspilitz will work on a olimex lime2? has sata built in, i do have one of those laying around.

openoms, 2020-07-08:
Never ever run the power through the GPIO and microUSB the same time. The disk powered through the USB bridge.
Here is the user manual: https://geekworm.com/products/x820-v3-0-usb-3-0-2-5-inch-sata-hdd-ssd-storage-expansion-board?_pos=1&_sid=28ecd4af5&_ss=r

I don't know the olimex board, but the manual sdcard build should work on anything, but likely needs manual debug steps.
See the details here: https://github.com/rootzoll/raspiblitz/tree/master/alternative.platforms

Closing this issue since it is about various hardware problems not a problem with the raspiblitz image.

closed 2020-07-08 by openoms

DrBlackross, 2020-07-08:
LOL, figured that out with this... this has a constant demand. 
but, heading to micro anyway today. picking up a case, 15w supply and a NEW rpi4.

added some local nodes (addnodes) with fiber to the bitcoin.conf for sync'ing, it should be past the 8gb mark by the time i get back. also, opened up the bandwidth limiter on the router for that ip (wife is done with conference calls today).

[image]

will see what smokes....
shouldn't need any ridiculous amount of ram, its just shuffling data... if it does ill grab the 4gb later)

DrBlackross, 2020-07-08:
no no it was a rpi 3 b+ not a 4... but i went and picked up one today. came back and it crashed, swapping over to the rpi4 and the 15w...

log below.

2020-07-08T21:22:12Z Bitcoin Core version v0.20.0 (release build)
2020-07-08T21:22:12Z Assuming ancestors of block 0000000000000000000f2adce67e49b0b6bdeb9de8b7c3d7e93b21e7fc1e819d have valid signatures.
2020-07-08T21:22:12Z Setting nMinimumChainWork=00000000000000000000000000000000000000000e1ab5ec9348e9f4b8eb8154
2020-07-08T21:22:12Z Using the 'standard' SHA256 implementation
2020-07-08T21:22:12Z Default data directory /home/bitcoin/.bitcoin
2020-07-08T21:22:12Z Using data directory /mnt/hdd/bitcoin
2020-07-08T21:22:12Z Config file: /home/bitcoin/.bitcoin/bitcoin.conf
2020-07-08T21:22:12Z Config file arg: addnode="136.55.36.157:8333"
2020-07-08T21:22:12Z Config file arg: addnode="136.55.169.166:8333"
2020-07-08T21:22:12Z Config file arg: daemon="1"
2020-07-08T21:22:12Z Config file arg: datadir="/mnt/hdd/bitcoin"
2020-07-08T21:22:12Z Config file arg: dbcache="512"
2020-07-08T21:22:12Z Config file arg: disablewallet="1"
2020-07-08T21:22:12Z Config file arg: maxconnections="40"
2020-07-08T21:22:12Z Config file arg: maxmempool="64"
2020-07-08T21:22:12Z Config file arg: maxorphantx="10"
2020-07-08T21:22:12Z Config file arg: maxuploadtarget="5000"
2020-07-08T21:22:12Z Config file arg: rpcallowip="127.0.0.1"
2020-07-08T21:22:12Z Config file arg: rpcpassword=****
2020-07-08T21:22:12Z Config file arg: rpcuser=****
2020-07-08T21:22:12Z Config file arg: server="1"
2020-07-08T21:22:12Z Config file arg: testnet="0"
2020-07-08T21:22:12Z Config file arg: txindex="0"
2020-07-08T21:22:12Z Config file arg: upnp="1"
2020-07-08T21:22:12Z Config file arg: zmqpubrawblock="tcp://127.0.0.1:28332"
2020-07-08T21:22:12Z Config file arg: zmqpubrawtx="tcp://127.0.0.1:28333"
2020-07-08T21:22:12Z Config file arg: [main] rpcbind=****
2020-07-08T21:22:12Z Config file arg: [main] rpcport="8332"
2020-07-08T21:22:12Z Config file arg: [test] rpcbind=****
2020-07-08T21:22:12Z Config file arg: [test] rpcport="18332"
2020-07-08T21:22:12Z Command-line arg: conf="/home/bitcoin/.bitcoin/bitcoin.conf"
2020-07-08T21:22:12Z Command-line arg: daemon=""
2020-07-08T21:22:12Z Command-line arg: pid="/home/bitcoin/.bitcoin/bitcoind.pid"
2020-07-08T21:22:12Z Using at most 40 automatic connections (1024 file descriptors available)
2020-07-08T21:22:13Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2020-07-08T21:22:13Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2020-07-08T21:22:13Z Script verification uses 3 additional threads
2020-07-08T21:22:13Z Wallet disabled!
2020-07-08T21:22:13Z scheduler thread start
2020-07-08T21:22:13Z HTTP: creating work queue of depth 16
2020-07-08T21:22:13Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2020-07-08T21:22:13Z HTTP: starting 4 worker threads
2020-07-08T21:22:13Z init message: Loading banlist...
2020-07-08T21:22:13Z Using /16 prefix for IP bucketing
2020-07-08T21:22:13Z Cache configuration:
2020-07-08T21:22:13Z * Using 2.0 MiB for block index database
2020-07-08T21:22:13Z * Using 8.0 MiB for chain state database
2020-07-08T21:22:13Z * Using 502.0 MiB for in-memory UTXO set (plus up to 61.0 MiB of unused mempool space)
2020-07-08T21:22:13Z init message: Loading block index...
2020-07-08T21:22:13Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2020-07-08T21:22:16Z Opened LevelDB successfully
2020-07-08T21:22:16Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2020-07-08T21:22:38Z LoadBlockIndexDB: last block file = 29
2020-07-08T21:22:38Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=687, size=78299133, heights=207579...208794, time=2012-11-12...2012-11-20)
2020-07-08T21:22:38Z Checking all blk files are present...
2020-07-08T21:22:38Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2020-07-08T21:22:38Z Opened LevelDB successfully
2020-07-08T21:22:38Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: 6f8f840f6fcc681f
2020-07-08T21:22:39Z Loaded best chain: hashBestChain=00000000000001d522c15928acce22a477c0afe9dfb79c5a99d5e869546153a7 height=208516 date=2012-11-18T20:37:29Z progress=0.016528
2020-07-08T21:22:39Z init message: Rewinding blocks...
2020-07-08T21:22:43Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) started
2020-07-08T21:22:43Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) completed (0.00s)
2020-07-08T21:22:43Z init message: Verifying blocks...
2020-07-08T21:22:43Z Verifying last 6 blocks at level 3
2020-07-08T21:22:43Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2020-07-08T21:22:48Z No coin database inconsistencies in last 6 blocks (3698 transactions)
[... 14 lines cut ...]
2020-07-08T21:22:49Z net thread start
2020-07-08T21:22:49Z dnsseed thread start
2020-07-08T21:22:49Z init message: Done loading
2020-07-08T21:22:49Z opencon thread start
2020-07-08T21:22:49Z addcon thread start
2020-07-08T21:22:49Z tor: Thread interrupt
2020-07-08T21:22:49Z Shutdown: In progress...
2020-07-08T21:22:49Z torcontrol thread exit
2020-07-08T21:22:49Z dnsseed thread exit
2020-07-08T21:22:49Z opencon thread exit
2020-07-08T21:22:49Z msghand thread start
2020-07-08T21:22:49Z msghand thread exit
[... cut ...]
2020-07-08T21:22:50Z UPNP_DeletePortMapping() returned: 0
2020-07-08T21:22:50Z upnp thread exit
2020-07-08T21:22:51Z scheduler thread exit
2020-07-08T21:22:51Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) started
2020-07-08T21:22:51Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) completed (0.00s)
2020-07-08T21:22:51Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) started
2020-07-08T21:22:51Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) completed (0.00s)
2020-07-08T21:22:52Z Shutdown: done

DrBlackross, 2020-07-08:
the olimex was just a general question

DrBlackross, 2020-07-08:
ok looks legit... 

[image]

new RPI4
new case for rpi4
new usb 3 hdd case
new 1tb drive (can't walk in to frys or microcenter without buying something else)
new 16gb micro sd with raspiblitz 1.6RC2 (balena etched)
new "official" raspberry 15w usb c power supply

OK..... no hdmi though i prefer headless

 lol
gonna go stick it in the lab/office and see if it runs (if it doesn't dont sweat it, i'll repurpose it)
thanks for the help though

DrBlackross, 2020-07-08:
11gb on space and still moving
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.0G  2.8G  3.9G  42% /
devtmpfs        937M     0  937M   0% /dev
tmpfs           970M     0  970M   0% /dev/shm
tmpfs           970M  8.5M  962M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           970M     0  970M   0% /sys/fs/cgroup
tmpfs            32M     0   32M   0% /var/cache/raspiblitz
/dev/mmcblk0p1  253M   52M  202M  21% /boot
/dev/sda1       916G   11G  859G   2% /mnt/hdd
tmpfs           194M     0  194M   0% /run/user/1000
tmpfs           194M     0  194M   0% /run/user/1001

i think its good!

DrBlackross, 2020-07-09:
you can close this, would definitely recommend rpi4 and usb 3 hdd adapter, this thing flies

DrBlackross, 2020-09-26:
so far i think its working.....

https://1ml.com/node/0282b3650afa7a0853231c5b2aed5da16a969730117541c9896fb7b9ce7d9ba306