raspiblitz/raspiblitz #1319: v1.6RC2 reportings

rootzoll, 2020-07-05:


adevoss, 2020-07-07:
Downloaded build_sdcard.sh v1.6

No download for lnd version v0.10.2-beta.rc4.
Version v0.10.2-beta downloads the requested files.

Jestopher-BTC, 2020-07-10:
The prompt after installing Thunderhub instructs the user to use port 3011, but it's actually on port 3010. 3011 returns a bad gateway error.

openoms, 2020-07-10:
> The prompt after installing Thunderhub instructs the user to use port 3011, but it's actually on port 3010. 3011 returns a bad gateway error.

It shoudl show a HTTPS link. Did you try 3011 with HTTPS? 3010 is the cleartext (HTTP) port.

openoms, 2020-07-10:
Trying Tesnet:

- [ ] wallet=wallet.dat needs to be specific: main.wallet=wallet.dat
- [ ] LND autounlock leads to errors if the testnet wallet has a different password (needs to be disabled before switching to testnet)

Jestopher-BTC, 2020-07-10:
> > The prompt after installing Thunderhub instructs the user to use port 3011, but it's actually on port 3010. 3011 returns a bad gateway error.
> 
> It should show a HTTPS link. Did you try 3011 with HTTPS? 3010 is the cleartext (HTTP) port.

Ah, yes. That works as HTTPS. It gives me a bunch of warnings about an invalid certificate though.

openoms, 2020-07-10:
> Ah, yes. That works as HTTPS. It gives me a bunch of warnings about an invalid certificate though.

the Thub info screen warns about needing to accept the self signed certificate, also shows you the fingerprint to compare. Welcome to the world of encryption (unapproved by a third a party).

frennkie, 2020-07-12:
LNbits over TOR

`/etc/tor/torrc`

```
# Hidden Service for lnbits
HiddenServiceDir /mnt/hdd/tor/lnbits
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:5000
```

This should be
```
# Hidden Service for lnbits
HiddenServiceDir /mnt/hdd/tor/lnbits
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:5002
HiddenServicePort 443 127.0.0.1:5003
```

Right?

openoms, 2020-07-12:
>This should be
```
# Hidden Service for lnbits
HiddenServiceDir /mnt/hdd/tor/lnbits
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:5002
HiddenServicePort 443 127.0.0.1:5003
```
@frennkie, yes that is the effect of:
`/home/admin/config.scripts/internet.hiddenservice.sh lnbits 80 5002 443 5003`
in the install script.

frennkie, 2020-07-12:
I guess I was messing around with this.. 😉

openoms, 2020-07-13:
Getting errors with macaroons on testnet.
```
$ lncli getinfoER    Cryptoadvance Specter                  │ │  
[lncli] rpc error: code = Unknown desc = verification failed: signature mismatch after caveat verification 
```
```
raspberrypi systemd[1]: Started LND Lightning Daemon.
1594667206 ERROR: [lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/testnet/admin.macaroon: no such file or directory
```
Permission or path issues I presume.

[ ] fix macaroons for testnet

frennkie, 2020-07-13:
Would you not run it like this on a testnet node: `lncli -n testnet getinfo`

openoms, 2020-07-13:
> Would you not run it like this on a testnet node: `lncli -n testnet getinfo`

Thanks, I was missing that. The main problem with the same errors I've been having with Balance of Satoshis, but reactivating Testnet did the trick. All works now.

adevoss, 2020-07-14:
When re-installing rtl (#1285) I get this error

```
12686 silly lifecycle rtl@0.8.1-beta~postinstall: Args: [ '-c', 'ngcc' ]
12687 silly lifecycle rtl@0.8.1-beta~postinstall: Returned: code: 126  signal: null
12688 info lifecycle rtl@0.8.1-beta~postinstall: Failed to exec postinstall script
```
There is a web interface

frennkie, 2020-07-20:
```
  File "/home/admin/config.scripts/internet.sshtunnel.py", line 191
    subprocess.call(sudo /home/admin/config.scripts/lnd.tlscert.sh refresh", shell=True)
```

frennkie, 2020-07-20:
Problems with Bitcoin Index

[image]

```
*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
error code: -28 error message: Loading block index...

-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

Use CTRL+c to EXIT to Terminal`
```

What's going on here? 

I have used `<Do Nothing>` and then `CTRL+c` twice now (separate systems) and everything (including the index appears to be fine.

frennkie, 2020-07-20:
Issue on `LNbits` install

1) tag is still outdated
2) Permission on temp tor dir

```
# Hidden Service for lnbits
HiddenServiceDir /mnt/hdd/tor/lnbits
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:5002
/home/admin/config.scripts/internet.hiddenservice.sh: line 80: /mnt/hdd/temp/tmp: Permission denied
HiddenServicePort 443 127.0.0.1:5003
```

adevoss, 2020-07-21:
After installing (on amd64 vm) I get:
`* The LND service is not running.`

/var/log/syslog:
```
sudo: /home/admin/config.scripts/blitz.subscriptions.ip2tor.py: command not found
...
dial tcp 127.0.0.1:9050: connect: connection refused
```
After reboot:
```
Use CTRL+c to EXIT to Terminal
bitcoin error: error code: -28
error message:
Loading block index...
```
Cursor at prompt

bitsam, 2020-07-30:
> Problems with Bitcoin Index
>
> [image]
>
> ```
> *****************************************
> * The bitcoind service is not running.
> *****************************************
> If you just started some [...]

This happened to me also. I decided to re-sync the blockchain. It is getting stuck under 1% with invalid blocks.

2020-07-30T09:18:26Z UpdateTip: new best=00000000000061e0951bb1802107743208cb709939fed622caffb1466e844786 height=123589 version=0x00000001 log2_work=62.415148 tx=497776 date='2011-05-13T06:10:52Z' progress=0.000903 cache=31.7MiB(244328txo)
2020-07-30T09:18:26Z UpdateTip: new best=00000000000058bb4ada68fa240c90e4348b06f35e9a2de0e695491b76c90d94 height=123590 version=0x00000001 log2_work=62.415306 tx=497798 date='2011-05-13T06:12:06Z' progress=0.000903 cache=31.7MiB(244359txo)
2020-07-30T09:18:26Z ERROR: ConnectBlock: Consensus::CheckTxInputs: 971a8b8bae55595809d0dd82a0d49ee76df1e1177975456889e8d19d3de54a3e, bad-txns-inputs-missingorspent, CheckTxInputs: inputs missing/spent
2020-07-30T09:18:26Z InvalidChainFound: invalid block=0000000000004bd05f245ceaf4a4d9d0e7b094cba9759ea9ca8ee1682c0495d0  height=123591  log2_work=62.415465  date=2011-05-13T06:17:35Z
2020-07-30T09:18:26Z InvalidChainFound:  current best=00000000000058bb4ada68fa240c90e4348b06f35e9a2de0e695491b76c90d94  height=123590  log2_work=62.415306  date=2011-05-13T06:12:06Z
2020-07-30T09:18:26Z ERROR: ConnectTip: ConnectBlock 0000000000004bd05f245ceaf4a4d9d0e7b094cba9759ea9ca8ee1682c0495d0 failed, bad-txns-inputs-missingorspent, CheckTxInputs: inputs missing/spent
2020-07-30T09:18:26Z InvalidChainFound: invalid block=0000000000003e1fa8c174e703f86119c9dde981de8831659fc435d5772c8132  height=123622  log2_work=62.420373  date=2011-05-13T08:49:18Z
2020-07-30T09:18:26Z InvalidChainFound:  current best=00000000000058bb4ada68fa240c90e4348b06f35e9a2de0e695491b76c90d94  height=123590  log2_work=62.415306  date=2011-05-13T06:12:06Z
2020-07-30T09:18:39Z New outbound peer connected: version: 70015, blocks=641448, peer=175 (full-relay)

rootzoll, 2020-07-31:
@bitsam there were some cases where "Probelems with Index" got detected if there were none. I tried to get rid of those. Will give it a try fpr the v1.6 relase ... if there is still a problem (not sure why you get invalid blocks on resync) please report on a seperate issue to we can put this into a bug fixing update like 1.6.1.

frennkie, 2020-08-02:
Testing on RC3 + XXsync I'm getting an error for LNbits:

```
Successfully installed python-dotenv-0.14.0
# updating databases
Loading .env environment variables…
https://127.0.0.1:8080
Traceback (most recent call last):
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/bin/flask", line 8, in <module>
    sys.exit(main())
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/cli.py", line 967, in main
    cli.main(args=sys.argv[1:], prog_name="python -m flask" if as_module else None)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/cli.py", line 586, in main
    return super(FlaskGroup, self).main(*args, **kwargs)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/cli.py", line 426, in decorator
    return __ctx.invoke(f, *args, **kwargs)
  File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/lnbits/lnbits/lnbits/__init__.py", line 78, in migrate_databases
    core_migrations.migrate()
  File "/home/lnbits/lnbits/lnbits/core/migrations.py", line 74, in migrate
    with open_db() as db:
  File "/home/lnbits/lnbits/lnbits/db.py", line 38, in open_db
    return Database(db_path=db_path)
  File "/home/lnbits/lnbits/lnbits/db.py", line 10, in __init__
    self.connection = sqlite3.connect(db_path)
sqlite3.OperationalError: unable to open database file
```

frennkie, 2020-08-02:
# testing password .. please wait
/home/admin/config.scripts/lnd.autounlock.sh: line 51: [: -gt: unary operator expected
# switching the Auto-Unlock ON
# storing password for root in /root/lnd.autounlock.pwd

frennkie, 2020-08-03:
`lndmanage` - although it should should have been installed it's not working:

```
admin@raspberrypi:~ $ manage
-bash: cd: /home/admin/lndmanage: No such file or directory
-bash: venv/bin/activate: No such file or directory
NOTICE: Needs at least one active channel to run without error.
to exit (venv) enter ---> deactivate
-bash: lndmanage: command not found
```

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll