raspiblitz/raspiblitz #1332: ip2tor - no /none port

frennkie, 2020-07-12:
I just ran into the situation that the host I was requesting a bridge from ran out of ports. 

I think we need to address this on both the server (for now I just added 10k additional ports) and client side.

Here is the message on the RaspiBlitz side:

[image]

```

OK your service is ready & your subscription is active.

You payed 50 sats for the first 24 hours.
Next AUTOMATED PAYMENTS will be 48 sats each.

Your service 'CUSTOM' should now publicly be reachable under:
18.184.174.142:None

Please test now if the service is performing as promised.
If not - dont forget to cancel the subscription under:
MAIN MENU > Manage Subscriptions > My Subscriptions
```

frennkie, 2020-07-12:
Trying to access this (to cancel) also fails after that:

```
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.py", line 164, in <module>
    mySubscriptions()
  File "/home/admin/config.scripts/blitz.subscriptions.py", line 112, in mySubscriptions
    publicaddress="{0}:{1}".format(selectedSub['ip'],selectedSub['port']),
KeyError: 'port'
Starting the main menu ..
```

rootzoll, 2020-07-16:
Will add a check if that value is a number and if not raising an Error.

closed 2020-07-24 by rootzoll