raspiblitz/raspiblitz #1336: Service Integration Guidelines

tiero, 2020-07-12:
Hi there, thanks for this amazing project.

I'm looking for how an external contributor could integrate an external service and how to properly propose and submit the work.
Would be very nice to have a detailed walkthrough on how to do that and especially what are the limitations/boundaries to follow.

I just searched for past issues/discussion (eg. BTCPay Integration issue) and from what I got from there, I have some questions:

1)  I'm right saying that the way to do it is to write a custom bash script?
2) The application/service must be compiled from source or it is acceptable to pull down already pre-built binaries for ARM?
3) what is the policy for outbound connections? For Blockchain data sourcing, is acceptable to fetch from APIs (eg. from block explorers) or the project imposes to lookup from the services already present? 
4) For inbound connections to expose the service, what's the policy to follow? (eg. Reverse Proxy on `host/<service_name>` ) A new entry should be added to the already existent `Nginx` webserver ([I got it from looking at the BTCPat Server installer script](https://gist.github.com/normandmickey/3f10fc077d15345fb469034e3697d0d0)) or the service should expose his own web server and How to avoid namespace collisions with others?

frennkie, 2020-07-14:
I think it really make sense to collect some guidelines. 

From the top of my head I can at least partially answer some of the raised questions

1. Yes. Have a look at the `bonus.*` files in `home.admin/config.scripts` (in the repo).
2. Currently we do not officially use docker. While I'm not entirely sure I believe that the fact that we still regard the Raspberry Pi 3 (1GB) as a supported platform plays a role here.
3. In general outbound connections must be opt-in. A user needs to beware of this and activate it herself.
4. If the service to be integrated has a web interface then we are currently switching to make sure that this is running on one unique port on `localhost`. Remote access should default to be made via Tor (Browser). Nginx as a reverse proxy helps to solve certain issues (e.g. with certificates).
5. I'm not aware of a sandbox. I develop in parts in a Windows IDE, a Linux VM (am64) - but always do integration and testing on the physical device.

tiero, 2020-07-14:
About point 2, it's ok to download pre-compiled binaries then? @frennkie

frennkie, 2020-07-14:
2) I would say that we strongly prefer to not use binaries that are pre built unless there is a strong way to gain trust that it's not tampered with. The best example here is that both `bitcoind` and lnd` are being downloaded as binaries (and then checksums are checked). We also trust stuff installed with `apt-get install`. But there are also some existing examples where the apps are compiled (even if this take 20+ minutes).

tiero, 2020-07-14:
Thanks for the swift response. It's much more clear now. I'll try to do my best