raspiblitz/raspiblitz #1341: ACME script over TOR

rootzoll, 2020-07-15:
At the moment the ACME script communicates over clearnet with the letsencrypt & dns API. This should also work over Tor in the future.

frennkie, 2020-07-19:
The `acme.sh` is a pure shell script. And it's using `curl` maybe it will work by setting the environment (if not with socks5h then maybe with the HTTPProxy of Tor on 8118).

rootzoll, 2020-07-20:
I just tried a curl default config file in the users home called `.curlrc` with the content `socks5-hostname localhost:9050` and that seemed to work :)

It looks like this `.curlrc` file needs to be also in `/root/` directory so that if curl is called with sudo it also takes those defaults.

@frennkie or @openoms  can you make a quick test and confirm - I think then we should add this to the v1.6 final

frennkie, 2020-07-20:
Why (only) a `curlrc` in `/root` and not (also) in `/home/admin`?

I find it weird that people seem to avoid the dedicated HTTP Proxy in Tor whatever it takes.. ;-)

rootzoll, 2020-07-22:
All major users on RaspiBlitz I think make sense: root, admin, pi & bitcoin

rootzoll, 2020-07-22:
Because changing a CURL default could have many side effetcs on othe install scripts. Too close to a final release. So moving this to next version, so it can have longer testing.

rootzoll, 2021-03-26:
Just tested: To call the ACME script with the `torify` prefix is not working .. getting a error:

```
CURLE_SSL_CONNECT_ERROR (35)

A problem occurred somewhere in the SSL/TLS handshake. You really want the error buffer and read the message there as it pinpoints the problem slightly more. Could be certificates (file formats, paths, permissions), passwords, and others.
```

OK so back to the  `.curlrc` default setting file.

rootzoll, 2021-03-26:
OK with the `.curlrc`on first test I also got the error 35 .. but on retesting it worked. So I will merge the changes and put this issue for final testing.

rootzoll, 2021-03-26:
hhmmm ... tested again seeing the error 35 again with final message: "Giving up sending to CA server after 20 retries."

I think we need to dig deeper here and make changes to the ACME script itself (fork on github - but not include in raspiblitz package).

rootzoll, 2021-03-26:
@frennkie do you have any idea how to fix this CURLE_SSL_CONNECT_ERROR (35)?

openoms, 2021-03-26:
I tried to use letsencrypt through a socat ip2tor túnnel before and couldn't make it work

rootzoll, 2021-03-26:
So I will remove the `.curlrc` setting for now that people can at least use a working letsencrypt for the v1.7

openoms, 2021-03-27:
Artcile about TLS (DV) cerificates for .onions: https://kushaldas.in/posts/get-a-tls-certificate-for-your-onion-service.html
https://www.helpnetsecurity.com/2021/03/26/how-to-get-affordable-dv-certificates-for-onion-sites/

rootzoll, 2021-04-15:
@openoms this Harica service is still 30 USD/year and I dont think its a solution to another problem - for the case we need a valid HTTPS cert for a additional app we run on RaspiBlitz. Normally that is not needed (because tor alread protects against ManInTheMiddle and offers transport security) but some WebApps might need HTTPS to access certain functions (camera, etc) .. but again I think thats a future edge case.

Our problem here is besically that the ACME script is not able to todo the registration with LetsEncrypt over Tor so we have to go over clearnet what is exposing the users IP to letsencrypt mad ties it to the domain the user is using for his IP2Tor service. We have to chack if this is just because something of the script or if its a basic LetsEncrypt thing.

rootzoll, 2021-12-07:
Seems that you can get ´lego´ acme tool working over torsocks.

First lego needs to get installed:
```
./config.scripts/bonus.go.sh on
export GO111MODULE=on
source /etc/profile
go install github.com/go-acme/lego/v4/cmd/lego@v4.5.3
```

_Also check if a commit can be used instead of a version tag number._

Then it can be called for example for dynu with the API key:
```
export DYNU_API_KEY=gbb35xxxxxxxxxxxxxxgdg6XT54f
torsocks lego --accept-tos --email admin@[DYNDNSDOMAIN] --dns dynu --domains [DYNDNSDOMAIN]
```

Something else didnt worked with the dns-challenge dynu - but worked perfect and quick with duckdns ... so basically this tool seems to work and offers a lot of possible dns-providers for the future - even such as `njalla` that allow anon domains payed with Bitcoin.
```
acme-dns, alidns, allinkl, arvancloud, auroradns, autodns, azure, bindman, bluecat, checkdomain, clouddns, cloudflare, cloudns, cloudxns, conoha, constellix, desec, designate, digitalocean, dnsimple, dnsmadeeasy, dnspod, dode, domeneshop, dreamhost, duckdns, dyn, dynu, easydns, edgedns, epik, exec, exoscale, freemyip, gandi, gandiv5, gcloud, gcore, glesys, godaddy, hetzner, hostingde, hosttech, httpreq, hurricane, hyperone, ibmcloud, iij, infoblox, infomaniak, internetbs, inwx, ionos, joker, lightsail, linode, liquidweb, loopia, luadns, manual, mydnsjp, mythicbeasts, namecheap, namedotcom, namesilo, netcup, netlify, nicmanager, nifcloud, njalla, ns1, oraclecloud, otc, ovh, pdns, porkbun, rackspace, regru, rfc2136, rimuhosting, route53, sakuracloud, scaleway, selectel, servercow, simply, sonic, stackpath, transip, vegadns, versio, vinyldns, vscale, vultr, wedos, yandex, zoneee, zonomi
```

But I will push this to the v1.8.0 because it needs quite some refactoring in the background and dialogs that is a bit too much for the v1.7.2. Also it should be part of a fatpack build - because installing it from sourcecode takes a good while.

rootzoll, 2021-12-07:
replacing the ACME script should also fix the problem with the: https://github.com/rootzoll/raspiblitz/issues/2247#issuecomment-878835305

konigra, 2022-05-26:
Hello,

I have the following issue:

I am running a raspiblitz, with an LND node and C-Lightning AND a BTCPAYSERVER lately.

Now, I want my nodes to stay private behind TOR. 
But I want the BTCPAYSERVER to be publicly reachable via an URL.

So I activated the IP2TOR option and I am paying a few SATS.
Now I activated the LETSENCRYPT to get the HTTPS certificate.
I use DUCKDNS, created a URL, I made it pointing to my BTCPAYSERVER IP.

Now the problem: from time to time, the Raspiblitz is populating DUCKDNS with its own external IP. So the IP is changed, and my BTCPAYSERVER stops being reachable. 

How do I stop the raspi from doing that? Or can I somehow, start an "letsencrypt"/acme inside the BTCPAYSERVER? any hints for me?

konigra, 2022-05-29:
somehow solved it by giving raspberry pi IP a different DNS name by a different dns-provider and connecting those two with eachother via

sudo pip3 install domain-connect-dyndns

now btcpayserver IP remains untouched and runs consistently.

case closed!