raspiblitz/raspiblitz #1342: Warn when node is using unsafe cltv_expiry_delta value (<18)

MrManPew, 2020-07-15:
See details here:
lightningnetwork/lightning-rfc#789
and here:
https://twitter.com/realtbast/status/1283298246983659520

I think it would be very efficient to the network if there is some way for raspiblitz to warn when the node has channels with cltv_expiry_delta lower than the recommended value (there are more than 7k such channels on mainnet)

openoms, 2020-07-16:
We are using the LND defaults unless changed by the user.
I think the warning should be on GUIs like for example
ThunderHub does suggest to change low values on the `Details` page and gives a method to change the values individually or for all channels at the same time:
[image]
[image]

MrManPew, 2020-07-16:
Great, I did not know that.
Maybe a warning can also be added on the main screen, it will be a shame if people will lose their money.

[image]

Thank you.

nyxnor, 2021-04-18:
It is not supposed to be a Blitz command, but a service command, cant get displayed in the status screen, as it is already full of info now and not working currently. Also moving for the WebUI. Pls close this.

closed 2021-04-19 by rootzoll