raspiblitz/raspiblitz #1348: LNbits Voucher & TPOS

rootzoll, 2020-07-16:
As part of the idea "RaspiBlitz 4 Local Community" als outlined in the video the v1.6 release should be tested if the Voucher and the TPOS pligins are working with the IP2TOR & LetsEncrypt subscriptions:

https://youtu.be/QXUGg45CWLo?t=3704

rootzoll, 2020-07-21:
@frennkie how does adding a flask database migrate command looks like for LNbits? We need to add it into the script, when on installing an LNbits DB already exists.

frennkie, 2020-07-21:
For install we already have it

https://github.com/rootzoll/raspiblitz/blob/c2386ab3e2c43082d6a390e1acc8bc79c93a7adb/home.admin/config.scripts/bonus.lnbits.sh#L222

Not sure how we upgrade... 'off' = 'on'? Should also be fine.

rootzoll, 2020-07-21:
What I did was I deinstalled LNbits (off) and then reinstalled it (on). Because ben had moved the "raspiblitz" tag I got the new version. But ran into thre error I mentioned on twitter.

rootzoll, 2020-07-23:
OK we are one step further with I am able now to print vouchers and (when behind IP2TOR&LetsEncrypt) I can scan with WalletOfSatoshi and get my satoshies :D But 2 things to take a look at ...

### 1. Exception during LN-Withdraw

Even the direct LNURL-Withdraw worked there maybe is still one Excpetion to take a look:
[image]

### 2. Not working Fallback Web-Tempwallet

I was trying then to scan another fresh QR voucher with my normal Smartphone-Cam APP and it offered me to open the URL `https://bitsblitz.duckdns.org:39254/lnurlwallet?lightning=LNURL1DP68GURN8GHJ7CNFW3EKYMRFW3AZUER4VD4KGMNN9EHHYEE6XVUNYDF59AMKJARGV3EXZAE0V9CXJTMKXYHKCMN4WFKZ7C3NW9RXX62VG4FX6NT5VS65WA2PWGE42SMR9AF55E62VEMYY3N0VF3XWNR52EY8Y4N50YU5S7XFRV7` but I just got a `Internal Server Error` on my mobile Webbrowser and when I tried to open the URL on my desktop in chrome I got `Bad Request`.

Strangely I was not seeing any further error log in this in the `sudo journalctl -u lnbits -b --no-pager -n20` @frennkie maybe already the nginx has problems here? But `sudo cat /var/log/nginx/error_lnbits.log` is empty. On `sudo cat /var/log/nginx/access_lnbits.log` I can see that something get called ... but no error:

```
127.0.0.1 - - [23/Jul/2020:18:38:37 +0100] "GET /withdraw/api/v1/lnurl/b3qFciLERmMtd5GuAr3UCc/SJgJfvBFobbgLtVHrVty9H HTTP/1.1" 200 56 "-" "python-requests/2.23.0"
127.0.0.1 - - [23/Jul/2020:18:38:37 +0100] "GET /lnurlwallet?lightning=LNURL1DP68GURN8GHJ7CNFW3EKYMRFW3AZUER4VD4KGMNN9EHHYEE6XVUNYDF59AMKJARGV3EXZAE0V9CXJTMKXYHKCMN4WFKZ7C3NW9RXX62VG4FX6NT5VS65WA2PWGE42SMR9AF55E62VEMYY3N0VF3XWNR52EY8Y4N50YU5S7XFRV7 HTTP/1.1" 400 175 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.89 Safari/537.36"
127.0.0.1 - - [23/Jul/2020:18:38:38 +0100] "GET /favicon.ico HTTP/1.1" 200 18094 "https://bitsblitz.duckdns.org:39254/lnurlwallet?lightning=LNURL1DP68GURN8GHJ7CNFW3EKYMRFW3AZUER4VD4KGMNN9EHHYEE6XVUNYDF59AMKJARGV3EXZAE0V9CXJTMKXYHKCMN4WFKZ7C3NW9RXX62VG4FX6NT5VS65WA2PWGE42SMR9AF55E62VEMYY3N0VF3XWNR52EY8Y4N50YU5S7XFRV7" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.89 Safari/537.36"
```

arcbtc, 2020-07-23:
This GET request is the problem, as I think it's stripping the port out
https://github.com/lnbits/lnbits/blob/c17ac32c86cf31e924913eb96dfc4db87839d59a/lnbits/core/views/lnurl.py#L36

arcbtc, 2020-07-23:
Nope, it's not. My mistake.
Maybe the screenshot issue is WoS spamming the callback endpoint
Have you tested a few times for the 2nd error?

rootzoll, 2020-07-23:
Will test more tomorrow

rootzoll, 2020-07-23:
Another side note @arcbtc it is possible to make the QR voucher codes a bit bigger?  Wallet of Satohi and my Phones CamApp had a hard time scanning the codes ... worked, but took a bit.

rootzoll, 2020-07-24:
@arcbtc ok I made a new set of voucher codes and just tried to call the URL in them in a normal browser .. to get to the Fallback-LNbits-Webwallet .. here is the expection in full:

```
-- Logs begin at Fri 2020-07-24 13:56:20 BST, end at Fri 2020-07-24 14:02:12 BST. --
Jul 24 14:01:37 raspberrypi sh[3578]:     chunked=chunked,
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 381, in _make_request
Jul 24 14:01:37 raspberrypi sh[3578]:     self._validate_conn(conn)
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 976, in _validate_conn
Jul 24 14:01:37 raspberrypi sh[3578]:     conn.connect()
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connection.py", line 396, in connect
Jul 24 14:01:37 raspberrypi sh[3578]:     _match_hostname(cert, self.assert_hostname or server_hostname)
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connection.py", line 406, in _match_hostname
Jul 24 14:01:37 raspberrypi sh[3578]:     match_hostname(cert, asserted_hostname)
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/usr/lib/python3.7/ssl.py", line 327, in match_hostname
Jul 24 14:01:37 raspberrypi sh[3578]:     % (hostname, dnsnames[0]))
Jul 24 14:01:37 raspberrypi sh[3578]: ssl.SSLCertVerificationError: ("hostname 'bitsblitz.duckdns.org' doesn't match 'd.ip2t.org'",)
Jul 24 14:01:37 raspberrypi sh[3578]: During handling of the above exception, another exception occurred:
Jul 24 14:01:37 raspberrypi sh[3578]: Traceback (most recent call last):
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
Jul 24 14:01:37 raspberrypi sh[3578]:     timeout=timeout
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 725, in urlopen
Jul 24 14:01:37 raspberrypi sh[3578]:     method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/util/retry.py", line 439, in increment
Jul 24 14:01:37 raspberrypi sh[3578]:     raise MaxRetryError(_pool, url, error or ResponseError(cause))
Jul 24 14:01:37 raspberrypi sh[3578]: urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='bitsblitz.duckdns.org', port=443): Max retries exceeded with url: /withdraw/api/v1/lnurl/cb/hGpiT8wxPCraon3CxCWSvL?k1=CQpnhVqH6h69xbxK8RmAAH&pr=lnbc100n1p034ka3pp5udxvm7qsjeqzcnatmwf0rf85d0f4z44a4hjjjxz7tv0nvueksqgsdpqf38xy6t5wvsycnj42fxzqen4dejxjmn8cqzpgsp5yqld75kh5gmuvdre4ckq45awn06v4g5lx43kxsdfj44s9fnatxkq9qy9qsqu7vkajykax4dx5aap6nwn3vnl4t5es57ezj05ru2kvewsxh8d2q9ee9m9fn3gxuvuagmvtfvfgsfsadthgcc3zaxcyekmnz02z3eaaqp4s8l2g (Caused by SSLError(SSLCertVerificationError("hostname 'bitsblitz.duckdns.org' doesn't match 'd.ip2t.org'")))
Jul 24 14:01:37 raspberrypi sh[3578]: During handling of the above exception, another exception occurred:
Jul 24 14:01:37 raspberrypi sh[3578]: Traceback (most recent call last):
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
Jul 24 14:01:37 raspberrypi sh[3578]:     response = self.full_dispatch_request()
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
Jul 24 14:01:37 raspberrypi sh[3578]:     rv = self.handle_user_exception(e)
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask_cors/extension.py", line 161, in wrapped_function
Jul 24 14:01:37 raspberrypi sh[3578]:     return cors_after_request(app.make_response(f(*args, **kwargs)))
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
Jul 24 14:01:37 raspberrypi sh[3578]:     reraise(exc_type, exc_value, tb)
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
Jul 24 14:01:37 raspberrypi sh[3578]:     raise value
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
Jul 24 14:01:37 raspberrypi sh[3578]:     rv = self.dispatch_request()
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
Jul 24 14:01:37 raspberrypi sh[3578]:     return self.view_functions[rule.endpoint](**req.view_args)
Jul 24 14:01:37 raspberrypi sh[3578]:   File "/home/lnbits/lnbits/lnbits/core/views/lnurl.py", line 38, in lnurlwallet
[... cut ...]
Jul 24 14:01:37 raspberrypi sh[3578]: requests.exceptions.SSLError: HTTPSConnectionPool(host='bitsblitz.duckdns.org', port=443): Max retries exceeded with url: /withdraw/api/v1/lnurl/cb/hGpiT8wxPCraon3CxCWSvL?k1=CQpnhVqH6h69xbxK8RmAAH&pr=lnbc100n1p034ka3pp5udxvm7qsjeqzcnatmwf0rf85d0f4z44a4hjjjxz7tv0nvueksqgsdpqf38xy6t5wvsycnj42fxzqen4dejxjmn8cqzpgsp5yqld75kh5gmuvdre4ckq45awn06v4g5lx43kxsdfj44s9fnatxkq9qy9qsqu7vkajykax4dx5aap6nwn3vnl4t5es57ezj05ru2kvewsxh8d2q9ee9m9fn3gxuvuagmvtfvfgsfsadthgcc3zaxcyekmnz02z3eaaqp4s8l2g (Caused by SSLError(SSLCertVerificationError("hostname 'bitsblitz.duckdns.org' doesn't match 'd.ip2t.org'")))
```

rootzoll, 2020-07-24:
Big hint seems to be: `Caused by SSLError(SSLCertVerificationError("hostname 'bitsblitz.duckdns.org' doesn't match 'd.ip2t.org'"))` @frennkie how gets the the `d.ip2t.org` gets in there if that is not even part of the URL? The URL called was: https://bitsblitz.duckdns.org:39254/lnurlwallet?lightning=LNURL1DP68GURN8GHJ7CNFW3EKYMRFW3AZUER4VD4KGMNN9EHHYEE6XVUNYDF59AMKJARGV3EXZAE0V9CXJTMKXYHKCMN4WFKZ76Z8WP54GWRH0PGYXUNPDAHRXSMCGDT4XAJV9ASK2S6ZX4XY6EREGEGYWJJWDEXHSNJ6XF9HG0L5X9S

@arcbtc also maybe interesting: On first call of the URL I get "Internet Server Error" on the second "Bad Request" from the server

frennkie, 2020-07-24:
@rootzoll it's still the same port stripping issue. If you access the ip directly on port 443 there is an nginx running that has this San entries.

rootzoll, 2020-07-24:
> @rootzoll it's still the same port stripping issue. If you access the ip directly on port 443 there is an nginx running that has this San entries.

@arcbtc can we get rid of this port-stripping issue?

rootzoll, 2020-08-02:
We found one problem that the URLs in the Voucher QR codes use "http" not "https".
@arcbtc can you just change this: So a printed voucher should always use https

But that seems just half the problem.
Also just a side note for the future: It would be great if the URL in the Voucher will also be usable as a backup link to the LNbits wallet that is created. At the moment if you try to call this Voucher URL a second time, it just tells you that the URL

frennkie, 2020-08-02:
> QR codes use "http" not "https".

I found that issue.

frennkie, 2020-08-02:
Also found at least one aspect that has to do with the problem:

```
Aug 02 19:59:36 raspberrypi sh[18333]: Exception: withdraw: tag='withdrawRequest' callback=WebUrl('https://XXX.duckdns.org:51234/withdraw/api/v1/lnurl/cb/eUVsQ5hJc3UVt2sMtErQJ7', scheme='https', host='XXX.duckdns.org', tld=
'org', host_type='domain', port='51234', path='/withdraw/api/v1/lnurl/cb/eUVsQ5hJc3UVt2sMtErQJ7') k1='XhdQvpaHsny3NYugoBhFTn' min_withdrawable=10000 max_withdrawable=10000 default_description='#withdraw LNbits LNURL'
https://XXX.duckdns.org:51234/withdraw/api/v1/lnurl/cb/eUVsQ5hJc3UVt2sMtErQJ7 https://rb4.duckdns.org/withdraw/api/v1/lnurl/cb/eUVsQ5hJc3UVt2sMtErQJ7
```

In https://github.com/lnbits/lnbits/blob/master/lnbits/core/views/lnurl.py#L37 `withdraw_res.callback.base` is used instead of `withdraw_res.callback` and `base` seems to exclude the port.

rootzoll, 2020-08-13:
I tested with a fresh installed LNbits und updated `raspiblitz` tag today again the vouchers. If I open a Fallback-URL of the vouchers (running behind IP2TOR & LetsEncrypt HTTPS) I still get a `Internal Server` Error in my mobile Browser and in the LNbits logs I see this message:

@arcbtc Would be really great to get this fixed finally. See the analysis of @frennkie above.

```
Aug 13 22:32:01 raspberrypi sh[13511]: [2020-08-13 22:32:01,932] ERROR in app: Exception on /lnurlwallet [GET]
Aug 13 22:32:01 raspberrypi sh[13511]: Traceback (most recent call last):
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connection.py", line 160, in _new_conn
Aug 13 22:32:01 raspberrypi sh[13511]:     (self._dns_host, self.port), self.timeout, **extra_kw
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/util/connection.py", line 84, in create_connection
Aug 13 22:32:01 raspberrypi sh[13511]:     raise err
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/util/connection.py", line 74, in create_connection
Aug 13 22:32:01 raspberrypi sh[13511]:     sock.connect(sa)
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/gevent/_socket3.py", line 400, in connect
Aug 13 22:32:01 raspberrypi sh[13511]:     raise error(err, strerror(err))
Aug 13 22:32:01 raspberrypi sh[13511]: ConnectionRefusedError: [Errno 111] Connection refused
Aug 13 22:32:01 raspberrypi sh[13511]: During handling of the above exception, another exception occurred:
Aug 13 22:32:01 raspberrypi sh[13511]: Traceback (most recent call last):
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 677, in urlopen
Aug 13 22:32:01 raspberrypi sh[13511]:     chunked=chunked,
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 381, in _make_request
Aug 13 22:32:01 raspberrypi sh[13511]:     self._validate_conn(conn)
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 976, in _validate_conn
Aug 13 22:32:01 raspberrypi sh[13511]:     conn.connect()
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connection.py", line 308, in connect
Aug 13 22:32:01 raspberrypi sh[13511]:     conn = self._new_conn()
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connection.py", line 172, in _new_conn
Aug 13 22:32:01 raspberrypi sh[13511]:     self, "Failed to establish a new connection: %s" % e
Aug 13 22:32:01 raspberrypi sh[13511]: urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0xb42dd730>: Failed to establish a new connection: [Errno 111] Connection refused
Aug 13 22:32:01 raspberrypi sh[13511]: During handling of the above exception, another exception occurred:
Aug 13 22:32:01 raspberrypi sh[13511]: Traceback (most recent call last):
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
Aug 13 22:32:01 raspberrypi sh[13511]:     timeout=timeout
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/connectionpool.py", line 725, in urlopen
Aug 13 22:32:01 raspberrypi sh[13511]:     method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/urllib3/util/retry.py", line 439, in increment
Aug 13 22:32:01 raspberrypi sh[13511]:     raise MaxRetryError(_pool, url, error or ResponseError(cause))
Aug 13 22:32:01 raspberrypi sh[13511]: urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='testbits.duckdns.org', port=443): Max retries exceeded with url: /withdraw/api/v1/lnurl/cb/hgk62AYZouu47ZQoh68Nvh?k1=deuaPgoEfFecwyQ7xYFsRi&pr=lnbc100n1p0ntv23pp5w4s2w3egxujwymlyaqkylugwmdmanqc22094ar7kdq8jq4gf8svqdpqf38xy6t5wvsycnj42fxzqen4dejxjmn8cqzpgsp52j8ulyyju3lmhh6wuydmm5m6452tqq48cfz29k0939cpf55h6ggq9qy9qsqmk9q6spjt5t545dcpx56sc03n7k27s662n0f9wy4309rdj7n35asyahfldsu4rtzxhp8ey7csahwmk2fhj8xrkak3jm7caf68w7d7ecqlmh08s (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb42dd730>: Failed to establish a new connection: [Errno 111] Connection refused'))
Aug 13 22:32:01 raspberrypi sh[13511]: During handling of the above exception, another exception occurred:
Aug 13 22:32:01 raspberrypi sh[13511]: Traceback (most recent call last):
Aug 13 22:32:01 raspberrypi sh[13511]:   File "/home/lnbits/.local/share/virtualenvs/lnbits-xqYMrTZ5/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
Aug 13 22:32:01 raspberrypi sh[13511]:     response = self.full_dispatch_request()
[... cut ...]
Aug 13 22:32:01 raspberrypi sh[13511]: requests.exceptions.ConnectionError: HTTPSConnectionPool(host='testbits.duckdns.org', port=443): Max retries exceeded with url: /withdraw/api/v1/lnurl/cb/hgk62AYZouu47ZQoh68Nvh?k1=deuaPgoEfFecwyQ7xYFsRi&pr=lnbc100n1p0ntv23pp5w4s2w3egxujwymlyaqkylugwmdmanqc22094ar7kdq8jq4gf8svqdpqf38xy6t5wvsycnj42fxzqen4dejxjmn8cqzpgsp52j8ulyyju3lmhh6wuydmm5m6452tqq48cfz29k0939cpf55h6ggq9qy9qsqmk9q6spjt5t545dcpx56sc03n7k27s662n0f9wy4309rdj7n35asyahfldsu4rtzxhp8ey7csahwmk2fhj8xrkak3jm7caf68w7d7ecqlmh08s (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xb42dd730>: Failed to establish a new connection: [Errno 111] Connection refused'))
```

eillarra, 2020-08-14:
@rootzoll I think the Blitz should install using requirements.txt here, instead of pipenv: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/bonus.lnbits.sh#L218
Pipenv is helpful while in development, but the locked requirements are the way to go for you (installation will be faster too, as you don't even need to install Pipenv).

We had to revert some requirement changes for myNode, so we just updated the `lnurl` bugfix manually. But your installation script uses Pipenv so it is not getting the bugfix. I will try to consolidate both in the next days, but in any case I think it is better if your script always installs from requirements.txt.

eillarra, 2020-08-14:
L218-219 can be:

    $ sudo -u lnbits pipenv run pip install python-dotenv
    $ sudo -u lnbits pipenv run pip install -r requirements.txt

And for consistency, L223:

    $ sudo -u lnbits pipenv run flask migrate

rootzoll, 2020-08-14:
> @rootzoll I think the Blitz should install using requirements.txt here, instead of pipenv: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/bonus.lnbits.sh#L218

I changed that and tested - looks good.

> We had to revert some requirement changes for myNode, so we just updated the lnurl bugfix manually. But your installation script uses Pipenv so it is not getting the bugfix. I will try to consolidate both in the next days

Let me know when you are ready - happy to give it another testrun.

rootzoll, 2020-08-14:
@eillarra we have a techdemo with bitcoin magazine next week Tuesday where we want to show the LNbits voucher printing on RaspiBlitz. Would be great if we can sort out the last bugs over the weekend.

frennkie, 2020-08-14:
I still don't understand for what we are using a dedicated RaspiBlitz tag on `LNbits`.  Does it help more than cause extra effort?

rootzoll, 2020-08-14:
> I still don't understand for what we are using a dedicated RaspiBlitz tag on LNbits. Does it help more than cause extra effort?

It gives the benefit that people with "turn it off/on" can update Lnbits if there is a bugfix or important feature for RaspiBlitz .. and LNbits development is happening at the moment between RaspiBlitz releases. So its a in-between solution until LNbits is reaching a version, that we can bundle RaspiBlitz releases.

philbertw4, 2020-08-17:
Just to chime in here. I'm not sure if my issue is the same, but I finally have LNBits running behind IP2TOR + LetsEncrypt and enabled the LNURLw extension. Unfortunately any vouchers it generates seem to give some sort of SSL error when trying to redeem them with an LNURL capable wallet (tried both Phoenix & BLW). No such SSL error exists when visiting LNBits via browser, so I am inclined to think I'm experiencing the same thing that as you all on this thread.

Is this fixed already and I just need to update something? Or are you still scrambling to get it fixed prior to the tech demo tomorrow/Tuesday? Happy to help test on my node if you need, as I am now pretty good at uninstalling/reinstalling LNBits :-)

rootzoll, 2020-08-18:
@philbertw4 looks like we got that SSL error fixed - try to reinstall LNbits again and give it a test.

Background: it seems that an old version of the python url3 lib was not respecting the port when getting the SSL cert (an git the wrong one from the IP2TOR bridge server) - an update on the requirements and using a newer version of url3 seems to do the trick.

Now the fallbackURL (scanning with std cam app and open URL in browser) creates a new LNbits temp wallet on my test runs.

Killed one bug, found some more - @arcbtc would be great to work in this in the next iteration:

1) The temp wallet created has 0 sats .. not the balance of the voucher @eillarra says it might be still LNbits error.

2) The creation of the temp wallet takes quite a long time ... it would be great if a user calls that URL in the voucher on a browser you would get at least a loading page/spinner ... at the moment its just a empty browser sitting there for 10 seconds and it looks like your browser is stuck .. but then in the end it works.

3) Some LNbits functions on that browser wallet are quite slow behind IP2TOR: So when I try to pay an invoice from my browser wallet I can see that the payment is received in the receiving side quite quickly ... but on my browser wallet I dont get a confirmation that the payment worked - stays on the "processing payment" notice. Maybe the IP2TOR background call to check if the payment is payed is so slow that the checking loop always restarts before it finished.

rootzoll, 2020-08-18:
Sidenode: Zap Android Wallet seems not yet to support LNURL-FALLBACK-URLs. I wrote on their GitHub:
https://github.com/btcontract/lnurl-rfc#fallback-scheme

philbertw4, 2020-08-18:
Ok, I just tried a re-install of LNBits via the menu. I chose not to "delete the server data" when prompted, but perhaps I should have? I figured that keeping the server data would keep my IP2TOR and LetsEncrypt stuff working. The reinstall seemed to proceed fine and the lnbits git commit message seemed to match your latest changes to the `raspiblitz` branch of the lnbits repository. However, sadly, when trying to withdraw via the lnurlw extension and using phoenix wallet, I still got the same error I was getting before: "invalid data." I did not have time to test with another wallet but will try again a little later and do the complete re-install (deleting server data too) to see how it goes.

rootzoll, 2020-08-18:
> using phoenix wallet, I still got the same error I was getting before: "invalid data."

This looks similar what I get with the ZAP Android Wallet that says it suppurts LNURL-Withdraw BUT they dont support the "LNURL-FALLBACK-SHEME" that is such an important part of the LUNRL-Spec that its mentioned right at the beginning. See: https://github.com/btcontract/lnurl-rfc#fallback-scheme

@philbertw4 try with WalletOfSatishi (I know, but just to see if it works in general). If it works then it would be great if you can write Phoenix Wallet that it seems that they dont support the "LNURL-FALLBACK-SHEME" that is very easy to implement and an important part of the Spec. It seems we need to pressure some Wallet-Providers to be "LNURL Spec Compliant" here ;)

rootzoll, 2020-08-18:
Also tested LNURL FALLBACK-URL with Zeus - is giving me a "wrong seperator character" error ... but on first look the FALLBACK-URL looks ok. @kaloudis is Zeus supporting the LNURL FALLBACK SPEC? https://github.com/btcontract/lnurl-rfc#fallback-scheme

rootzoll, 2020-08-18:
Also maybe another bug to look into @eillarra or @arcbtc I created two vouchers and print them out. I am using WalletOfSatoshi to scan the first and it works :D Hoooray! And scan the second and it says: "Withdraw spend"?!? I checked they have different LNURL strings in the QR code ... but maybe something else is still off?

philbertw4, 2020-08-18:
Just tried on android Bitcoin Lightning Wallet (BLW) -- fail with some sort of "SSL Hanshake Error"
Tried on Phoenix -- fail "Bad Data"
Tried on BlueWallet -- fail "Bad Data" (didn't catch exact error message, sorry)
Tried on WalletOfSatoshi -- fail "Problem processing the LNURL"

I will try another re-install, this time choosing the option to delete the LNBits server data.

philbertw4, 2020-08-18:
Also tried on the telegram bot `@lntxbot` which failed with `Error: failed to fetch lnurl params: unknown response tag`. Reinstalling now to see if same problem persists after a fresh LNBits+IP2TOR+LetsEncrypt install

philbertw4, 2020-08-18:
Ok, here is some more strange behavior: using the python lnurl library to decode the lnurls that are being generated by the LNBits service on my rapsiblitz, sometimes there is an extra trailing slash (`/`) on the end of the URL, and sometimes there is not such a slash. Unfortunately irrespective of whether there is a slash I am still getting the same errors in the various wallets when trying to perform the withdrawal tests.

This was done on a fresh LNBits "install" on the raspiblitz, choosing "delete server data" when prompted in the menu. Given that one of the wallets is giving me an SSL handshake error (no such error exists in the browser), I will now cancel/re-do my LetsEncrypt service and see if that fixes it.

philbertw4, 2020-08-18:
Yes, there is definitely something strange going on with the extra trailing slash in the url. When the extra trailing slash is there, and using `@lntxbot` as wallet, I get the `Error: failed to fetch lnurl params: unknown response tag`.

However, when the extra slash is not there, then it seems one of two outcomes happens:
1. it works (`@lntxbot` was able to withdraw the sats from LNBits)
2. it fails with a `lnurl error: Withdraw is spent.` (yet, in fact, the withdrawal has NOT been spent).

Not quite sure how to debug from here or whether the issue is officially more on Raspiblitz side or LNBits side or Wallet side.

philbertw4, 2020-08-18:
Sorry for yet another comment. Just trying to continue getting to the bottom of this. Has anyone here with LNBits running on their Raspiblitz been able to successfully transfer sats from one LNBits wallet to another LNBits wallet where both are hosted on the same machine? I have the first wallet generate an invoice and then copy/paste that invoice into the second wallet to pay it. Yet, I keep getting an internal error 502 "backend error" in LNBits when trying to essentially transfer some sats between the wallets in this manner -- seems strange since such a transfer should proceed fine? or is it getting caught by the lightning-send-to-self delay? Wondering if it is related to this issue, or unrelated. Perhaps @arcbtc will be able to chime in?

We are so close to getting this working!

arcbtc, 2020-08-19:
Added a bunch of fixes to LNURLw, one of which is using a splash button to avoid QR scanners hitting the scanned URL (we did already have the splash screen step, I just linked the fallback to the wrong endpoint). The LNURLs + fallbacks should all work fine now. 

@philbertw4 internal payments are possible, although they'll involve some hackyness we are yet to implement

arcbtc, 2020-08-19:
Please test lnurlw on blitz if you get a mo. I have updated the raspiblitz tag

rootzoll, 2020-08-19:
@arcbtc Refreshed LNbits and tested. Opening the fallback-URL of Voucher I now get to a page with button "To Claim Bitcoin" - i press it .. it takes some time and then I got a LNbits account page with 0 sats ... even the voucher should have contained 10 sats.

arcbtc, 2020-08-19:
Whats the url on the claim bitcoin page?have noticed some QR scanners are stripping the LNURL param

arcbtc, 2020-08-19:
lntxbot is doing it for me when I scan the QR, however using a different QRscanner the LNURL is there. Whether I need to make the prints bigger again.
Have you tried using the shared link QR?

rootzoll, 2020-08-19:
@arcbtc thats the URL auf the ClaimBitcoin Page:

https://testbits.duckdns.org:59473/?lightning=LNURL1DP68GURN8GHJ7AR9WD6XY6T5WVHXGATRDDJXUUEWDAEXWW348Y6RWVE0WA5HG6RYWFSHWTMPWP5J7A339AKXUATJDSH5ZD34VEMY2EZ3WC69J52EGAKHZSTEG4N4ZUP0VF65KDJVX3HXU7J989NXY3R429V9XJMF2P6QUR555N

arcbtc, 2020-08-19:
Goodness, that does take a while

philbertw4, 2020-08-19:
I tested the new changes on some wallets. Thanks @arcbtc for the quick turn around!

* `@lntxbot` -- worked
* BlueWallet -- worked
* Bitcoin Lightning Wallet  -- still failed with some sort of "SSL Hanshake Error" as error message in GUI
* Phoenix -- failed, but this time immediately (seemingly prior to even making any network lookup) with "Invalid Data" as error message in GUI

arcbtc, 2020-08-20:
I wish it was faster, poor ol'Rootz has been wrestling this for a while. I'll ask Anton about the BLW one, maybe he has some extra checks.
Prior to the mention earlier, there is a PR for the internal payments hackyness, which seems to work, although I need to do some more testing

frennkie, 2020-08-20:
@arcbtc I also pressed that link... And obviously it's now already taken. Then this error message is displayed 

```
Bad Request
Could not process LNURL-withdraw: LNURL-withdraw not found.
```

Would be better to post a more user friendly message with explanation.

rootzoll, 2020-09-23:
@arcbtc When I try to run the latest master of LNbits with the RaspiBlitz install script its not starting up with the following message - o we need to update the install script?

```
Sep 23 20:15:22 raspberrypi systemd[1]: Started lnbits.
Sep 23 20:15:25 raspberrypi sh[21190]: Loading .env environment variables…
Sep 23 20:15:25 raspberrypi sh[21190]: Error: the command gunicorn could not be found within PATH or Pipfile's [scripts].
Sep 23 20:15:25 raspberrypi systemd[1]: lnbits.service: Main process exited, code=exited, status=1/FAILURE
Sep 23 20:15:25 raspberrypi systemd[1]: lnbits.service: Failed with result 'exit-code'.
```

rootzoll, 2020-10-09:
Implemented a quickfix for the css problem ... pushed better fix to v1.7 with #1646

rootzoll, 2020-10-14:
OK testing for v1.6.1 as a `raspiblitz` tag candidate:
./config.scripts/bonus.lnbits.sh on lnbits 8d135489abd4eaf24a3e53a0594184a363569775

rootzoll, 2020-10-14:
Hmmm .. got lnbits server running now again, but even when I go back to older commits from last week I still get the problem in WalletOfSatoshi that a fresh Voucher is already spend "Withdraw is Spend" and on Fallback I get to a wallet with zero balance showing me a pending invoice in the history.

rootzoll, 2020-10-14:
So there is still something off ... but set LNbits to install with the "raspiblitz" tag and let that fix after going into Release Candidate Testing.

rootzoll, 2020-11-03:
In a first attempt we got LN vouchers with LNbits & IP2TOR working ... so setting this to final testing for v1.6.2

rootzoll, 2020-11-06:
Back in Berlin I retested the LNbits vouchers and Booom now its working and will be part of v1.6.2 release. It should even work with v1.6.1 if people (re-)install LNbits fresh :D

And also now a lot of mobile wallets already or will with the next release support the voucher Codes with the LNURL fallback:
[image]

So LNbits vouchers with RaspiBlitz over IP2Tor will be part of v1.6.2 release. Closing issue :)

closed 2020-11-06 by rootzoll