raspiblitz/raspiblitz #1349: Error when patching to most recent v1.6rc3

adevoss, 2020-07-16:
Sorry I can't give more information.
After patching about 20 minutes ago I rebooted and connected via ssh.

There is an error showing real short, too short for me to read properly.
I think it is 'command not found line 82'.
The file I can't read.

There is a mesaage about an index. Turn off index or do nothing. Choose 'Do nothing'
'raspiblitz' shows the main menu

hermanlim, 2020-07-17:
Encountered the same problem this morning.
I chose to turn off indexing, delete and rebuild the index.
In the interim, I have noticed some problems with nginx and many of the apps which depend on it are obviously not working.

admin@PiOne:~ $ systemctl status nginx
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-07-17 19:48:45 AEST; 3min 10s ago
     Docs: man:nginx(8)
  Process: 3920 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)

admin@PiOne:~ $ sudo journalctl -x | grep nginx
-- Subject: A start job for unit nginx.service has begun execution
-- A start job for unit nginx.service has begun execution.
Jul 17 19:45:15 PiOne nginx[617]: nginx: [emerg] open() "/etc/nginx/snippets/ssl-certificate-app-data-tor.conf" failed (2: No such file or directory) in /etc/nginx/sites-enabled/btcrpcexplorer_tor_ssl.conf:9
Jul 17 19:45:15 PiOne nginx[617]: nginx: configuration file /etc/nginx/nginx.conf test failed
Jul 17 19:45:15 PiOne systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
-- An ExecStartPre= process belonging to unit nginx.service has exited.
Jul 17 19:45:15 PiOne systemd[1]: nginx.service: Failed with result 'exit-code'.
-- The unit nginx.service has entered the 'failed' state with result 'exit-code'.
-- Subject: A start job for unit nginx.service has failed
-- A start job for unit nginx.service has finished with a failure.
-- Subject: A start job for unit nginx.service has begun execution
-- A start job for unit nginx.service has begun execution.
Jul 17 19:48:04 PiOne nginx[3871]: nginx: [emerg] open() "/etc/nginx/snippets/ssl-certificate-app-data-tor.conf" failed (2: No such file or directory) in /etc/nginx/sites-enabled/btcrpcexplorer_tor_ssl.conf:9
Jul 17 19:48:04 PiOne nginx[3871]: nginx: configuration file /etc/nginx/nginx.conf test failed
Jul 17 19:48:04 PiOne systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
-- An ExecStartPre= process belonging to unit nginx.service has exited.
Jul 17 19:48:04 PiOne systemd[1]: nginx.service: Failed with result 'exit-code'.
-- The unit nginx.service has entered the 'failed' state with result 'exit-code'.
-- Subject: A start job for unit nginx.service has failed
-- A start job for unit nginx.service has finished with a failure.
-- Subject: A start job for unit nginx.service has begun execution
-- A start job for unit nginx.service has begun execution.
Jul 17 19:48:43 PiOne nginx[3920]: nginx: [emerg] open() "/etc/nginx/snippets/ssl-certificate-app-data-tor.conf" failed (2: No such file or directory) in /etc/nginx/sites-enabled/btcrpcexplorer_tor_ssl.conf:9
Jul 17 19:48:45 PiOne nginx[3920]: nginx: configuration file /etc/nginx/nginx.conf test failed
Jul 17 19:48:45 PiOne systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
-- An ExecStartPre= process belonging to unit nginx.service has exited.
Jul 17 19:48:45 PiOne systemd[1]: nginx.service: Failed with result 'exit-code'.
-- The unit nginx.service has entered the 'failed' state with result 'exit-code'.
-- Subject: A start job for unit nginx.service has failed
-- A start job for unit nginx.service has finished with a failure.

openoms, 2020-07-17:
can you try running the
 `./XXsyncScripts v1.6 rootzoll`
again?
I do usually run it twice to.make sure everything is updated and copied.

Otherwise test nginx with:
`sudo nginx -t`

frennkie, 2020-07-17:
In this situation turning blitz.web.sh off and on again should also resolve the problem.

adevoss, 2020-07-17:
@openoms Before I read your comment I patched again. Is that the same as XXsyncScripts,sh?

I was only 20 lines of. Think It is line 62 in 80scanLND.sh.
There is a '[' missing. Should it be four separate conditions or should the first two be in parentheses or something else?

openoms, 2020-07-17:
@adevoss yes, XXsyncScripts.sh (corrected)
if you found a bug you can link the code here from the repo or even fork and make PR with the fix.

adevoss, 2020-07-17:
When I reboot I see:

```
*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
0

tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

Use CTRL+c to EXIT to Terminal
bitcoin error: error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.
```

adevoss, 2020-07-17:
@openoms The if statement on these lines can be fixed in many ways. I don't know what should be checked so I can't make a pr. The '[' can be added or a ']' can be deleted.
What should these lines be checking?

The code is in 80scanLND.sh line 52 and 62:
```
if [ ${uptime} -gt 600 ] ||  ${#bitcoinErrorFull} -gt 0 ] || [ "${1}" == "blockchain-error" ]; then
if [ ${uptime} -gt 600 ] ||  ${#bitcoinErrorFull} -gt 0 ] || [ "${bitcoinErrorShort}" == "Error found in Logs" ] || [ "${1}" == "blockchain-error" ]; then
```

frennkie, 2020-07-17:
"Patching" with XXsyncScript.sh" is not a supported way! 

@openoms @rootzoll I suggest to close this.

@adevoss Don't get me wrong: We are always happy to get testers and reports for bugs.. but also we need to focus our time. Please feel free to re-open if you have more details/a fix. Otherwise please wait for RC3 or the final release of 1.6.

adevoss, 2020-07-17:
@frennkie  No problem. I think #1356 fixed the problem. Thanks.
Don't know about what @hermanlim wrote. To me that looks like another issue.

frennkie, 2020-07-17:
@hermanlim running this should fix it - if not please wait for RC3 or the final release

```
./config.scripts/blitz.web.sh off
./config.scripts/blitz.web.sh on
```

closed 2020-07-17 by frennkie

hermanlim, 2020-07-17:
@frennkie 
Your solution worked perfectly, thank you.

./config.scripts/blitz.web.sh off
./config.scripts/blitz.web.sh on

The problem was seen originally patching from RC2 -> RC3 whilst troubleshooting LNBits.