raspiblitz/raspiblitz #1350: Loop In not working

marssystems, 2020-07-17:
I get an error when tryin to do a "loop in".

admin@raspberrypi:~ $ loop in 1000000 6
Max swap fees for 1000000 sat Loop In: 89320 sat
CONTINUE SWAP? (y/n), expand fee detail (x): y
[loop] rpc error: code = Unknown desc = cannot initiate swap: rpc error: code = DeadlineExceeded desc = context deadline exceeded
admin@raspberrypi:~ $ loopd &
[1] 20913
admin@raspberrypi:~ $ 2020-07-17 11:08:32.142 [INF] LOOPD: Version: 0.6.0-beta commit=
2020-07-17 11:08:32.143 [INF] LNDC: Creating lnd connection to localhost:10009
2020-07-17 11:08:32.155 [INF] LNDC: Connected to lnd
unable to obtain macaroons: open /home/admin/.lnd/data/chain/bitcoin/mainnet/invoices.macaroon: no such file or directory

[1]+  Done                    loopd

rootzoll, 2020-07-17:
Seems like some timeframe contrain on the swap - not a problem with the loop/lnd software configuration in general. I would recommend to check out the loop project/support on that.

marssystems, 2020-07-17:
OK - I will raise this issue with Lightninglabs directly.

marssystems, 2020-07-22:
According to Lightninglabs/Loop support, the raspiblitz LND build doesn't create the extra macaroons needed by loop to function.
I only have 3 macaroons - admin.macaroon, invoice.macaroon and readonly.macaroon in the directory /mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/
So when I run loopd, I get this error -
admin@raspberrypi:~ $ loopd
2020-07-21 15:44:16.910 [INF] LOOPD: Version: 0.6.5-beta commit=
2020-07-21 15:44:16.910 [INF] LNDC: Creating lnd connection to localhost:10009
2020-07-21 15:44:16.912 [INF] LNDC: Connected to lnd
2020-07-21 15:44:17.145 [INF] LNDC: lnd version: v0.10.3-beta, build tags 'autopilotrpc,signrpc,walletrpc,chainrpc,invoicesrpc,watchtowerrpc'
loopd exited with an error: unable to obtain macaroons: open /home/admin/.lnd/data/chain/bitcoin/mainnet/invoices.macaroon: no such file or directory
admin@raspberrypi:~ $

See https://github.com/lightninglabs/loop#lnd and https://github.com/lightninglabs/loop#execute-a-swap

rootzoll, 2020-07-22:
Thanks for that out .. will add this to the final fixes for v1.6 release.

marssystems, 2020-07-22:
So is there no temporary fix for v1.5.1 so I can get LOOP working now?

rootzoll, 2020-07-22:
With v1.6 final coming soon, there will be no patch for 1.5.1 ... but you should be able to create that macaroons manually if you cant wait.

marssystems, 2020-07-22:
OK - no worries - I will have to wait for v1.6 as I don't have the skills to create the macaroons manually myself (I wish I did!)

rootzoll, 2020-07-23:
OK I just checked, that when upating to v1.6 it will create the additional macaroons. Next - recheck Loop.

rootzoll, 2020-07-24:
OK tested - Loop for the v1.6 release. Working - Closing issue.

marssystems, 2020-07-24:
Just updated LND to v0.10.4-beta using the RaspiBlitz Menu system and it didn't create the extra macaroons required by LOOP to function correctly.

rootzoll, 2020-07-24:
There is a reason why its called "reckless" update ... bit its still a bit strange. Will test again later with updating from LND 0.9.2.

marssystems, 2020-07-24:
Well LND v0.10.4-beta has been built with the extra tags, its just that the macaroons aren't being saved in /mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/

admin@raspberrypi:~ $ loopd
2020-07-24 16:37:46.476 [INF] LOOPD: Version: 0.7.0-beta commit=
2020-07-24 16:37:46.480 [INF] LNDC: Creating lnd connection to localhost:10009
2020-07-24 16:37:46.481 [INF] LNDC: Connected to lnd
2020-07-24 16:37:46.590 [INF] LNDC: lnd version: v0.10.4-beta, build tags 'autopilotrpc,signrpc,walletrpc,chainrpc,invoicesrpc,watchtowerrpc'
loopd exited with an error: unable to obtain macaroons: open /home/admin/.lnd/data/chain/bitcoin/mainnet/invoices.macaroon: no such file or directory
admin@raspberrypi:~ $

rootzoll, 2020-07-24:
Have you tried MAINMENU > LNDCREDS > RESET Recreate macaroons + TLS

marssystems, 2020-07-24:
Just tried it now and only the 3 basic macaroons were recreated.

rootzoll, 2020-07-24:
Since LND 0.9 there were more then 3 macaroons and when the old ones get deleted the new ones should be created.  Maybe you can head over to the LND Dev Slack and ask why you still just get stuck with 3 macaroons and let us know: https://dev.lightning.community

marssystems, 2020-07-24:
OK - I'll try that and get back to you.

marssystems, 2020-07-27:
I just looked at the RaspiBlitz script. It looks like it copies these three macaroons after installation. So in the $HOME/.lnd/data/chain/bitcoin/mainnet/ directory, only the three default ones are ever going to exist.

Do you find any macaroons with this command?

find /mnt/hdd/lnd -name "*.macaroon"
I checked and lnd 0.10.4 does create all required macaroons. It's just RaspiBlitz that sets up the paths strangely. There's /mnt/hdd/lnd where the actual lnd runs, then there's /mnt/hdd/app-data/lnd where some of the files are copied to and then there's /home/admin/.lnd/ where only the macaroons are copied to.

I think this line needs to change to include all macaroons:
https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/lnd.credentials.sh#L86

marssystems, 2020-07-27:
I have edited the lnd.credentials.sh file and now all the macaroons are copied to the /home/admin/.lnd/.......... correctly.

The problem I have is that the 3 original macaroons have an owner as "admin" and the extra macaroons have an owner as "root".
Where and how can I change the owner of the extra macaroons from root to admin?

marssystems, 2020-07-29:
Edit lnd.credentials.sh and add this -

###########################
# RESET Macaroons and TLS
###########################
if [ "$1" = "reset" ]; then
  clear
  echo "###### RESET MACAROONS AND TLS.cert ######"
  echo ""
  echo "All your macaroons and the tls.cert get deleted and recreated."
  echo "Use this to invalidate former EXPORTS for example if you loose a device."
  echo ""
  cd || exit
  echo "- deleting old macaroons"
  sudo find /mnt/hdd/app-data/lnd/data/chain/"${network}"/"${chain}"net/ -iname '*.macaroon' -delete
  sudo find /home/bitcoin/.lnd/data/chain/"${network}"/"${chain}"net/ -iname '*.macaroon' -delete
  sudo rm /home/bitcoin/.lnd/data/chain/"${network}"/"${chain}"net/macaroons.db
  echo "- resetting TLS cert"
  sudo /home/admin/config.scripts/lnd.newtlscert.sh
  echo "- restarting LND ... wait 10 secs"
  sudo systemctl start lnd
  sleep 10
  sudo -u bitcoin lncli --chain="${network}" --network="${chain}"net unlock
  echo "- creating new macaroons ... wait 10 secs"
  sleep 10
  echo "- copy new macaroons to central app-data directory and ensure unix ownerships and permissions"
  copy_mac_set_perms admin.macaroon lndadmin "${network}" "${chain}"
  copy_mac_set_perms invoice.macaroon lndinvoice "${network}" "${chain}"
  copy_mac_set_perms readonly.macaroon lndreadonly "${network}" "${chain}"
  copy_mac_set_perms chainnotifier.macaroon lndchainnotifier "${network}" "${chain}"
  copy_mac_set_perms walletkit.macaroon lndwalletkit "${network}" "${chain}"
  copy_mac_set_perms invoices.macaroon lndinvoices "${network}" "${chain}"
  copy_mac_set_perms signer.macaroon lndsigner "${network}" "${chain}"
  copy_mac_set_perms router.macaroon lndrouter "${network}" "${chain}"
  echo "OK DONE"

###########################
# SYNC
###########################
elif [ "$1" = "sync" ]; then
  echo "###### SYNCING MACAROONS, RPC Password AND TLS Certificate ######"

  echo "# make sure LND app-data directories exist"
  sudo /bin/mkdir --mode 0755 --parents /mnt/hdd/app-data/lnd/data/chain/"${network}"/"${chain}"net/

  echo "# copy macaroons to central app-data directory and ensure unix ownerships and permissions"
  copy_mac_set_perms admin.macaroon lndadmin "${network}" "${chain}"
  copy_mac_set_perms invoice.macaroon lndinvoice "${network}" "${chain}"
  copy_mac_set_perms readonly.macaroon lndreadonly "${network}" "${chain}"
  copy_mac_set_perms chainnotifier.macaroon lndchainnotifier "${network}" "${chain}"
  copy_mac_set_perms walletkit.macaroon lndwalletkit "${network}" "${chain}"
  copy_mac_set_perms invoices.macaroon lndinvoices "${network}" "${chain}"
  copy_mac_set_perms signer.macaroon lndsigner "${network}" "${chain}"
  copy_mac_set_perms router.macaroon lndrouter "${network}" "${chain}"
  
Then do a groupadd for each of the extra macaroons - lndrouter, lndsigner etc. etc.

rootzoll, 2020-07-29:
Much thanks ... I will give the `lnd.credentials.sh` a freshup before the final v1.6 release.

marssystems, 2020-07-29:
OK - that's great!

The only thing I couldn't find is where the groups for lndadmin, lndinvoice and lndreadonly are created in the build process so I could add the extra groups there. Maybe you know and you can add them as well, with the correct permissions and users.

frennkie, 2020-08-02:
@marssystems **do not run** this command: `loopd &` !

We are running loopd as a systemd service (as user `bitcoin`) and it has no problem to access the macaroons in `/home/bitcoin/.lnd/data/chain/bitcoin/mainnet`.

marssystems, 2020-08-02:
Ok - no worries.

marssystems, 2020-08-03:
You will need to add the extra macaroons to the lnd.export.sh file -

########################
# HEXSTRING
########################
elif [ "${exportType}" = "hexstring" ]; then

  clear
  echo "###### HEXSTRING EXPORT ######"
  echo ""
  echo "admin.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/admin.macaroon
  echo ""
  echo "chainnotifier.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/chainnotifier.macaroon
  echo ""
  echo "invoice.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/invoice.macaroon
  echo ""
  echo "invoices.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/invoices.macaroon
  echo ""
  echo "readonly.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/readonly.macaroon
  echo ""
  echo "router.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/router.macaroon
  echo ""
  echo "signer.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/signer.macaroon
  echo ""
  echo "walletkit.macaroon:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/${network}/${chain}net/walletkit.macaroon
  echo ""
  echo "tls.cert:"
  sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/tls.cert
  echo ""

###########################
# SHH / SCP File Download
###########################
elif [ "${exportType}" = "scp" ]; then

  local_ip=$(ip addr | grep 'state UP' -A2 | tail -n1 | awk '{print $2}' | cut -f1 -d'/')
  clear
  echo "###### DOWNLOAD BY SCP ######"
  echo "Copy, paste and execute these commands in your client terminal to download the files."
  echo "The password needed during download is your Password A."
  echo ""
  echo "admin.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/admin.macaroon ./"
  echo ""
  echo "chainnotifier.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/chainnotifier.macaroon ./"
  echo ""
  echo "invoice.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/invoice.macaroon ./"
  echo ""
  echo "invoices.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/invoices.macaroon ./"
  echo ""
  echo "readonly.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/readonly.macaroon ./"
  echo ""
  echo "router.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/router.macaroon ./"
  echo ""
  echo "signer.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/signer.macaroon ./"
  echo ""
  echo "walletkit.macaroon:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/data/chain/${network}/${chain}net/walletkit.macaroon ./"
  echo ""
  echo "tls.cert:"
  echo "scp bitcoin@${local_ip}:/home/bitcoin/.lnd/tls.cert ./"
  echo ""

###########################
# HTTP File Download
###########################
elif [ "${exportType}" = "http" ]; then

  local_ip=$(ip addr | grep 'state UP' -A2 | tail -n1 | awk '{print $2}' | cut -f1 -d'/')
  randomPortNumber=$(shuf -i 20000-39999 -n 1)
  sudo ufw allow from 192.168.0.0/16 to any port ${randomPortNumber} comment 'temp http server'
  clear
  echo "###### DOWNLOAD BY HTTP ######"
  echo ""
  echo "Open in your browser --> http://${local_ip}:${randomPortNumber}"
  echo ""
  echo "You need to be on the same local network - not reachable from outside."
  echo "In browser click on files or use 'save as' from context menu to download."
  echo ""
  echo "Temp HTTP Server is running - use CTRL+C to stop when you are done"
  echo ""
  cd 
  randomFolderName=$(shuf -i 100000000-900000000 -n 1)
  mkdir ${randomFolderName}
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/admin.macaroon ./${randomFolderName}/admin.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/chainnotifier.macaroon ./${randomFolderName}/chainnotifier.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/invoice.macaroon ./${randomFolderName}/invoice.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/invoices.macaroon ./${randomFolderName}/invoices.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/readonly.macaroon ./${randomFolderName}/readonly.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/router.macaroon ./${randomFolderName}/router.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/signer.macaroon ./${randomFolderName}/signer.macaroon
  sudo cp /home/bitcoin/.lnd/data/chain/${network}/${chain}net/walletkit.macaroon ./${randomFolderName}/walletkit.macaroon
  sudo cp /home/bitcoin/.lnd/tls.cert ./${randomFolderName}/tls.cert
  
Then the 'Manage LND Credentials' menu item will work correctly.

frennkie, 2020-08-03:
> You will need to add the extra macaroons to the lnd.export.sh file

Why? 

What are you trying to accomplish? What problems do you encounter?

Also: Please don't paste long lines of code into the issues here. If you want to create a pull request please do so.. :-D

frennkie, 2020-08-03:
Just to be clear: loop in is working just fine without any modifications.

Let us know whether we can close this ticket.

marssystems, 2020-08-03:
Well LND now creates 8 macaroons but RaspiBlitz is only dealing with the 3 basic ones. That's why LOOP is having problems working as it can't find the extra macaroons which it needs. 
Since I amended my code (as above), I have had no problems with my setup.
I wanted to share the fix so other people wouldn't have issues in the future and for completeness.

P.S. sorry for the long lines of code - just wanted to make it easy for you to 'cut and paste' it into the relevant files - a 2 minute job.

And no - LOOP wasn't working 'just fine'  without these modifications.

marssystems, 2020-08-03:
I have LOOP working fine on my system with these modifications, so if you feel that you don't need to implement these changes foe everyone then you can close this ticket.

frennkie, 2020-08-03:
> Well LND now creates 8 macaroons but RaspiBlitz is only dealing with the 3 basic ones. 

We are aware of this. I have not seen a case were a different macaroon that the 3 basic ones are needed by apps - YET.

> That's why LOOP is having problems working as it can't find the extra macaroons which it needs.

As loop is running as `bitcoin` on RaspiBlitz it can easily find all 8 macaroons in their location: `/home/bitcoin/.lnd/data/chain/${network}/${chain}net/` 

> just wanted to make it easy for you to 'cut and paste' it into the relevant files - a 2 minute job.

What you did is the exact opposite. Using a pull request to make clear which lines you suggest to be changed in which is the easy way (both for you and us .. :-)

 
> And no - LOOP wasn't working 'just fine' without these modifications.

I  saw that you ran `loopd &` as `admin`  - this is not the supported setup on RaspiBlitz. Then I also understand why there is an issue with the macaroons.

marssystems, 2020-08-03:
Yes - but the lnd.credentials.sh script (as is) doesn't copy all the macaroons needed by LOOP to that location.

frennkie, 2020-08-03:
> Yes - but the lnd.credentials.sh script (as is) doesn't copy all the macaroons needed by LOOP to that location.

Correct: `lnd.credentials.sh` doesn't copy them.

Reason: They don't need to be copied.

I'll try a last time:

On RaspiBlitz `loopd` is running as a systemd service as the user `bitcoin`. Because it is running as `bitcoin` it will look for the macaroons in this directory: `/home/bitcoin/.lnd/data/chain/${network}/${chain}net/` - and that is where all macaroons are. So nothing needs to be copied.

Don't get me wrong.. we are always happy for people testing, finding issues and reporting them (even with suggested solutions). But in this case you are using it in the wrong way. Do it correctly and it will not try to look for macaroon in the wrong location.

closed 2020-08-03 by frennkie

marssystems, 2020-08-03:
Obviously you missed this comment by the LOOP support team -

I just looked at the RaspiBlitz script. It looks like it copies these three macaroons after installation. So in the $HOME/.lnd/data/chain/bitcoin/mainnet/ directory, only the three default ones are ever going to exist.

Do you find any macaroons with this command?

find /mnt/hdd/lnd -name "*.macaroon"
I checked and lnd 0.10.4 does create all required macaroons. It's just RaspiBlitz that sets up the paths strangely. There's /mnt/hdd/lnd where the actual lnd runs, then there's /mnt/hdd/app-data/lnd where some of the files are copied to and then there's /home/admin/.lnd/ where only the macaroons are copied to.

I think this line needs to change to include all macaroons:
https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/lnd.credentials.sh#L86

I am running RaspiBlitz v1.5.1 so my system is running correctly.

frennkie, 2020-08-03:
> It's just RaspiBlitz that sets up the paths strangely. 

We decided to change the way how access to the macaroons is provided to "Apps" for better security. Please have a look at the issue and PR if you are interested:  https://github.com/rootzoll/raspiblitz/issues/1143 https://github.com/rootzoll/raspiblitz/pull/1147

frennkie, 2020-08-03:
Btw - the link to the same *non-issue* is this: https://github.com/lightninglabs/loop/issues/253

marssystems, 2020-08-03:
Yes - good - very interesting. But from the proposals it looks like the extra 5 macaroons have not been implemented fully.

GID | Needed? | Group Name | Macaroon File Name
-- | -- | -- | --
9700 | YES | lndadmin | admin.macaroon
9701 | YES | lndinvoice | invoice.macaroon
9702 | YES | lndreadonly | readonly.macaroon
9703 | ??? | lndinvoices | invoices.macaroon
9704 | ??? | lndchainnotifier | chainnotifier.macaroon
9705 | ??? | lndrouter | router.macaroon
9706 | ??? | lndsigner | signer.macaroon
9707 | ??? | lndwalletkit | walletkit.macaroon

lightninglabs/loop#253 - Yes - I raised this issue with the LOOP support team first. All sorted now.

frennkie, 2020-08-03:
As I indicated above with "YET" we have only done this for the three file that are need.

Should we find any apps that are not running as `bitcoin` but need an additional macaroon we will add it.

But so far this has not been the case. Definitely not for loopd.

marssystems, 2020-08-03:
OK - no worries - I will leave that with you then.