raspiblitz/raspiblitz #1351: Kindle Display server doesn't refresh after restarting RaspiBlitz

grnqrtr, 2020-07-17:
I've restarted my RaspiBlitz a couple of times since activating the kindle-display, and both times the kindle-display server doesn't continue to refresh the image (checking at RASPIBLITZ_IP:3030/display.png) after restart.

When I `sudo su - kindledisplay` and do `systemctl status kindle-display` it shows that it is running, but display doesn't update.

Both times I have fixed it by disabling and re-enabling the service.

rootzoll, 2020-07-17:
@dennisreimann any ideas?

dennisreimann, 2020-07-18:
That sounds like the cronjob isn't running. Are you using the latest version of the script which installs kindle-display v0.4.0?

grnqrtr, 2020-07-18:
It looks like I was on an older version, but even after downloading and enabling the new v0.4.0 script, and restarting, same thing.
```
admin@raspberrypi:~ $ sudo su - kindledisplay 
kindledisplay@raspberrypi:~ $ systemctl status kindle-display
● kindle-display.service - kindle-display
   Loaded: loaded (/etc/systemd/system/kindle-display.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-07-18 10:15:52 BST; 1h 29min ago
 Main PID: 7767 (npm)
    Tasks: 23 (limit: 4915)
   Memory: 15.6M
   CGroup: /system.slice/kindle-display.service
           ├─7767 npm
           ├─8026 sh -c node -r dotenv/config index.js
           └─8032 node -r dotenv/config index.js
```
It's running, but not updating the image at `http://RASPIBLITZ_IP:3030/display.png`. Disabling and re-enabling the script makes it work again. Is this not reproducible for anyone else?

dennisreimann, 2020-07-18:
By "disabling/enabling the script" you mean running the kindle-display setup with off/on or the systemctl?

grnqrtr, 2020-07-19:
I mean running the script with off/on.
```
/home/admin/config.scripts/bonus.kindle-display.sh off
/home/admin/config.scripts/bonus.kindle-display.sh on
```

dennisreimann, 2020-07-19:
Just tried and my `systemctl status kindle-display` after reboot matches yours.

My display gets updated – btw you can also check with `http://RASPIBLITZ_IP:3030/`, which is the "live" version that gets screenshot. (if you see a ongoing discrepancy between this and the screnshot it'd mean that the data is being fetched, but the screeenshot does not get updated)

I'm running Raspiblitz v1.6rc2

grnqrtr, 2020-07-19:
This seems to be what is happening. The "live" version at `http://RASPIBLITZ_IP:3030/` is updating, but the `display.png` is not getting updated.

dennisreimann, 2020-07-19:
That's weird, because the cron refreshes the data and updates the screenshot. Can you see if there are errors generating the screnshot? It doesn't really make any sense, because it works after switching it off and on, but what's the output when you run it manually?

```bash
sudo su - kindledisplay
/home/kindledisplay/kindle-display/server/cron.sh
```

Mine looks like this:

```
*** You are running in headless mode.
  Recompressing IDAT chunks in screenshot.png to display.png
   Total length of data found in critical chunks            =     68401
   Best pngcrush method        =   7 (ws 15 fm 0 zl 9 zs 0) =     31750
CPU time decode 0.088628, encode 0.612554, other 0.012251, total 0.745791 sec
```

And just to make sure: My `crontab -l` gives me this …

```
SHELL=/bin/bash
PATH=/bin:/usr/bin:/usr/local/bin
*/5 * * * * /home/kindledisplay/kindle-display/server/cron.sh > /dev/null 2>&1
```

grnqrtr, 2020-07-19:
Running it manually, the display.png still doesn't update, but there is some complaining about Firefox unlike yours:

```
kindledisplay@raspberrypi:~ $ /home/kindledisplay/kindle-display/server/cron.sh
*** You are running in headless mode.
Firefox is already running, but is not responding. To open a new window, you must first close the existing Firefox process, or restart your system.
  Recompressing IDAT chunks in screenshot.png to display.png
   Total length of data found in critical chunks            =     67061
   Best pngcrush method        =   7 (ws 15 fm 0 zl 9 zs 0) =     31274
CPU time decode 0.091959, encode 0.601472, other 0.012952, total 0.740437 sec
```

Also my `crontab -l`
```
SHELL=/bin/bash
PATH=/bin:/usr/bin:/usr/local/bin
*/5 * * * * /home/kindledisplay/kindle-display/server/cron.sh 2>&1 > /dev/null
```

dennisreimann, 2020-07-19:
Ok, looks like either a dangling instance of Firefox running or a previous run of Firefox didn't shut down cleanly and there is a lock file sitting around. The error is described [here](https://support.mozilla.org/en-US/kb/firefox-already-running-not-responding) and you can see if there's a profile lock file sitting around using `ls -la ~/.mozilla/firefox/`. If so, please delete it and run the script again to see the output.

grnqrtr, 2020-07-19:
Yup, that seems to have fixed it. I must have just had a bad shutdown? I did a reboot of the Pi just to make sure, and have no problems now. Thanks!
```
kindledisplay@raspberrypi:~/.mozilla/firefox/949sn751.default-esr $ rm lock 
kindledisplay@raspberrypi:~/.mozilla/firefox/949sn751.default-esr $ rm .parentlock 
kindledisplay@raspberrypi:~/.mozilla/firefox/949sn751.default-esr $ cd
kindledisplay@raspberrypi:~ $ /home/kindledisplay/kindle-display/server/cron.sh
*** You are running in headless mode.
  Recompressing IDAT chunks in screenshot.png to display.png
   Total length of data found in critical chunks            =     50672
   Best pngcrush method        =   7 (ws 15 fm 0 zl 9 zs 0) =     23867
CPU time decode 0.084749, encode 0.532384, other 0.012333, total 0.662174 sec
```

dennisreimann, 2020-07-19:
Alright, we can close this then @rootzoll :)

closed 2020-07-20 by rootzoll