raspiblitz/raspiblitz #1357: Instructions on how to access the Fully Noded quick connect QR?

Fonta1n3, 2020-07-18:
Hi, I often get your users asking me how to connect their raspiblitz to Fully Noded, unfortunately I do not yet run a raspiblitz so have no clue..

Please can you supply me with basic instructions and ideally screen shots so I can add them to my docs and share it with your users when they ask.

Thank you!

rootzoll, 2020-07-18:
Tor needs to be switched on on that node.
SSH-MAINMENU > FULLY_NODED ... and then follow the instructions.
- install from testflight
- run app & scan the pairing QR code displays on LCD or in terminal (ASCII-QR)

Fonta1n3, 2020-07-18:
> Tor needs to be switched on on that node.
> 
> SSH-MAINMENU > FULLY_NODED ... and then follow the instructions.
> 
> - install from testflight
> 
> - run app & scan the pairing QR code displays on LCD or in terminal (ASCII-QR)

Thanks for the info, will add it to our docs.

Fwiw the App Store version is definitely where you should be pointing people, Testflight is not updated/utilized for FN. https://apps.apple.com/us/app/fully-noded/id1436425586

rootzoll, 2020-07-18:
@Fonta1n3 I updated the AppStore link. Thx. Will be part of the v1.6 release.

nyxnor, 2021-04-18:
Please close this issue.

closed 2021-04-18 by Fonta1n3