raspiblitz/raspiblitz #1360: ip2tor: limit retries when PO (order) is rejected

frennkie, 2020-07-18:
When a RaspiBlitz IP2TOR Subscription is rejected the client will currently try to renew it anyway continuously:

[image]

rootzoll, 2020-07-18:
Ah OK ... will add this to client.

rootzoll, 2020-07-18:
@frennkie ok this should be fixed .. please check if you see this again with RC3

frennkie, 2020-07-18:
@rootzoll Please be aware that I added some more status options the the Purchase Orders and the Bridges:

* https://github.com/frennkie/django-ip2tor/blob/master/charged/lnpurchase/models.py#L26
* https://github.com/frennkie/django-ip2tor/blob/master/shop/models.py#L378

As far as I see it you have not been using them so far and I didn't do any *breaking* changes when adding the additional codes... so just FYI.

closed 2020-07-25 by rootzoll