raspiblitz/raspiblitz #1366: LND sync scanning progress

frennkie, 2020-07-20:
On LND 0.10.4 the log format seem to have changed. The [00infoBlitz.sh (INFO)](https://github.com/rootzoll/raspiblitz/blob/c6a98161468de1faa717d45f77c9470af181858c/home.admin/00infoBlitz.sh#L271) uses `(height` but the line (with **loglevel DEBUG**)  is:

```
2020-07-20 14:56:39.287 [DBG] LNWL: Filtering block 605826 (00000000000000000000f383c11667be88942faced981f45e42b4a366b5c31a3) with 3159 transactions
2020-07-20 14:56:42.683 [DBG] LNWL: Filtering block 605827 (00000000000000000004ee6591455c105127659213990e6aaa1436a022d5d0de) with 2905 transactions
```

frennkie, 2020-07-20:
I used this to enble debug logging: `lncli debuglevel --level debug`

frennkie, 2020-07-20:
The LCD (`00infoLCD.sh`) calls `80scanLND.sh` which uses `/home/admin/config.scripts/blitz.statusscan.sh`. There `getinfo` is called and the `best_header_timestamp` value is used to determine the progress. 

I guess it makes sense to switch the `00infoBlitz.sh` file to use the same way.

rootzoll, 2020-07-20:
- [ ] change to lnd `getinfo`

openoms, 2020-07-20:
looking good on the latest master
[image]

rootzoll, 2020-07-22:
Working with the quick fix for v1.6 ... will go with better implementation for v1.7.

bitsam, 2020-07-30:
@frennkie & @rootzoll  is there a possibility to apply a quick fix to 1.5 release? Like @openoms suggested with an updated script. I would really like to fix this issue on my node running 1.5 release (and manually updated LND to 10.4 & bitcoin to v0.20.0).

rootzoll, 2020-07-31:
@bitsam you can try to apply the changes frennkie made for the v1.6 release manually to your script and reboot:
https://github.com/rootzoll/raspiblitz/pull/1370/files

frennkie, 2020-07-31:
@bitsam note that (as stated) #1370 requires LND to be run with debug log level.

bitsam, 2020-08-11:
@frennkie & @rootzoll , i am running v1.6-2020-08-04.img.gz and notice that the LND sync drops out several times a day, similar to the itial reported error by frennkie. Is the fix a work in progress due to changes with bitcoin v0.20.0 & LND v0.11 ? Or could this be my Rpi3 and SATA HDD are not quick enough to keep up with the LND chain ? What info should I be looking at to see why my node is losing LND sync so often ?

rootzoll, 2020-08-13:
I removed the "scanning progress" completly on the ´00infoBlitz.sh´ .. it will just show "Waiting for Chain Sync" now when LND signals that `synced_to_chain` is false from `getinfo`. Also I changed the color from red to amber ... because its not a super critical event ... its even normal that LND might take a second to scan a block and if the raspiblitz measures at exactly that point in time the info might stay longer on the screen (until next refresh) than it even takes fro LND to catch up. Also I dont think your LND is "offline" during that time ... to forward or receive it should not be crucial to have the latest block in that second. You just need to know if channel force closed ... and if that channel is making in that second a payment you would have a newer closing transaction to punish that channel partner.

rootzoll, 2020-10-12:
All tests look good ... this wil be part of v1.6.1 release. Closing issue.

closed 2020-10-12 by rootzoll