raspiblitz/raspiblitz #1369: Telegraf as bonus for collecting system metrics

frennkie, 2020-07-20:
Telegraf is a nice solution to gather system metrics that can be collected locally or remotely in a `InfluxDB` and this can then be display in `Grafana` dashboard. https://medium.com/@dorian599/iot-raspberry-pi-container-and-system-monitoring-with-influxdb-telegraf-and-grafana-a1767c38c109

```
sudo -i 

curl -sL https://repos.influxdata.com/influxdb.key | apt-key add -
DISTRIB_ID=$(lsb_release -c -s)
echo "deb https://repos.influxdata.com/debian ${DISTRIB_ID} stable" | tee /etc/apt/sources.list.d/influxdb.list
apt-get update
apt-get install -y telegraf
sudo usermod -aG video telegraf
setcap 'cap_net_admin,cap_net_raw+ep' $(which ping)
```

Get a RaspiBlitz specific `telegraf.conf` from somewhere and put in `/etc/telegraf/telegraf.conf`

```
systemctl restart telegraf
```

PatrickScheich, 2020-08-05:
Hi, I would like to contribute to this topic. How can I do so?

I had created a telegraf.conf for my own raspiblitz and a matching grafana dashboard
[image]

I'm quite keen with programming, etc... but haven't yet worked on a github project
so I might need some pointers there

kindest regards, patrick

rootzoll, 2020-08-09:
In the next week we will produce some introduction tutorials on how to start contributing. But basically you can start small with the following: 

- get it working on your own raspiblitz
- edit the `/mnt/hdd/app-data/custom-installs.sh` in a way that it contains all needed setup shell commands so that it will reinstall the service sucessfully if you start your RaspiBlitz with a fresh sd card.
- then you can can post that shell script code in this issue here .. we can point you to the next steps from there

PatrickScheich, 2020-08-11:
Great, I will take a look into this. 

Unfortunately **telegraf** needs a bunch of individual decisions on config settings like
- ip and port of the backend influxdb
- name of the target database
- credentials for the database

I will assume that proper entries will exists in raspiblitz.conf

rootzoll, 2020-08-13:
Check if you can set those parameters on the install commands you can add to `custom-installs.sh` so that they dont need interaction ... in the next step when turning `custom-installs.sh` into an install script those parameters then get written into the `raspiblitz.conf` and the install script gets called using those config values during the recover process.

PatrickScheich, 2020-08-29:
thanks a lot
I will open a PR

PatrickScheich, 2020-08-30:
@frennkie I opened the PR 
I also connected my local VS Code with my github account and cloned the raspiblitz repository
then I made all the changes for the telegraf bonus option to the source tree.
How do I send my commits to the PR?

rootzoll, 2021-03-22:
OK I merged the PR into dev for upcomming v1.7 release :) @PatrickScheich & @frennkie I am a complete noob to telegraf .. so if one of you can test this on RC1 release (targeting this weekend) or even demo as part of the lightning hacksprint this weekend please feel invited - all info on the event: https://wiki.fulmo.org/wiki/Lightning_HackSprint_March_2021

rootzoll, 2021-04-18:
OK .. so the install looks good. Closing issue.

closed 2021-04-18 by rootzoll

PatrickScheich, 2021-04-19:
Hi,

I had been a bit busy lately so I had tested the v1.7.0RC2 yesterday.
It the "telegraf" part looks fine for me.

After updating in made sure that /mnt/hdd/raspiblitz.conf contains the
required telegraf entries.
Then the installation went smoothly with sudo /home/admin/config.scripts/
bonus.telegraf.sh on