raspiblitz/raspiblitz #1373: BTCPay Reset

hermanlim, 2020-07-23:
1.6RC3

There does not appear to be an easy way to reset BTCPay for users who have lost their admin password. The option to email password does not appear to be working either leaving users stuck.

Should consideration be given to including the option to reset BTCPay back to default settings in the Repair menu or adding the option to bonus.btcpayserver.sh?

rootzoll, 2020-07-23:
I think it makes sense to put this reset option into the "off" section. So the user can decide on deinstall if data should be kept or deleted. So if you want to reset your BTCPayServer you would uninstall it with delete data and then reinstall it thru the SERVICES menu.

rootzoll, 2020-07-23:
OK the optional delete/reset of the data during uninstall is working so far.

@openoms I am not sure if the BTC pay server picks up the data from the settings file - can you check this again? Also now that we can run BTCPayServer behind IP2TOR+LetsEncrypt ... do we need to set the external Domain?

rootzoll, 2020-07-25:
@openoms I will close this issue for the v1.6 release. Please reopen if there is anything left to do .. and just in case we can put out a 1.6.1 with smaller improvements if we can improve on BTCPayServer configuration.

closed 2020-07-25 by rootzoll