raspiblitz/raspiblitz #1375: 1.6rc3 Hang on setup and now Electrs won't install. error: failed to extract package (perhaps you ran out of disk space?)

grnqrtr, 2020-07-23:
I'm not sure if this is related to 1.6rc3 or not, but thought I should report, and maybe someone can help me either way!

When RC3 first came out, I flashed SD and updated, but the update hung up for a long time (~2.5 hours). Seems like it got hung up right after installing the kindle-display server. I took a screenshot when it was hung up:

[image]

Because it was hung up, I did `sudo shutdown -r now` and when it came back up it said something about it was able to "recover the update" (or something like that). Things seemed normal at first, but then I realized several things weren't working (joinmarket, elecrs, rpc-btc-explorer). It seemed like several of the problems were permission related (and after the fact, I think that can be fixed with something in the menu), but in particular I got errors about the "index" for rpc-btc-explorer/electrs, saying that I might need to resync the blockchain because it couldn't build the index.

I ended up reflashing RC3 onto a different SD card and going through the update process again thinking it might solve my problems. But before updating, I disabled the kindle-service (thought it might be related to the hang up).

When it came back up, I still had problems (particularly the index.) So I deleted the blockchain, copied it over from another disk, and let it finish syncing. I enabled rpc-btc-explorer and it seems to have successfully built the index, BUT I can't seem to get Electrs to install now.

This is the output of `/home/admin/config.scripts/bonus.electrs.sh on` (https://pastebin.com/35fERjGq)

In particular I see the errors:
```
info: installing component 'rust-std'
182.3 MiB / 182.3 MiB (100 %)   3.5 MiB/s in  1m  5s ETA:  0s
  6 IO-ops /   6 IO-ops (100 %)   0 IOPS in 25s ETA: Unknown
info: rolling back changes
error: failed to extract package (perhaps you ran out of disk space?)
error: caused by: No space left on device (os error 28)
```
And then later on:
```
HEAD is now at 840da89 Bump version
error: no override and no default toolchain set
```

I googled `caused by: No space left on device (os error 28)` and see that it may be related to "inodes" but I don't quite understand why I have this problem or how to fix it. The disk certainly isn't actually full. (~40%, 1TB SSD)

I hope I didn't include too much non-relevant information!

rootzoll, 2020-07-23:
@grnqrtr thanks for the report. Yes its a problem with the RC3 and the kindle-server - good that you catched that. The problem is that the kindle systemd service expectes that bitcoind service is started before it is starting. But during update/recovery the bitcoind is not started and so the kindle script on calling `sudo systemctl start kindle-display` waits forever. I had similar problem with BTCPay before.

I will edit the kindle-server so that its not blocking during update .. for the final v1.6 release.

grnqrtr, 2020-07-23:
Yay, glad to be of some help :)

I still can't figure out this `No space left on device (os error 28)`. I've just tried enabling other various service and am seeing similar errors. This is from installing BOS: (https://pastebin.com/gjcFVxvS)

I see things like this over an over:
```
npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc There appears to be insufficient space on your system to finish.
npm ERR! nospc Clear up some disk space and try again.
```

rootzoll, 2020-07-23:
BOS install is running on my testmachine without such errors.

@grnqrtr can you check your LCD on HDD usage? Also run `debug` on terminal and check the section on "*** MOUNTED DRIVES ***"

grnqrtr, 2020-07-24:
The LDC only shows 43% usage:

[image]

But the `debug` info for `*** MOUNTED DRIVES ***` I think probably shows some problems:

```
*** MOUNTED DRIVES ***
Filesystem     Type     1K-blocks      Used Available Use% Mounted on
/dev/root      ext4      30450024  28624176    538792  99% /
devtmpfs       devtmpfs   1966332         0   1966332   0% /dev
tmpfs          tmpfs      2000124       284   1999840   1% /dev/shm
tmpfs          tmpfs      2000124    205324   1794800  11% /run
tmpfs          tmpfs         5120         4      5116   1% /run/lock
tmpfs          tmpfs      2000124         0   2000124   0% /sys/fs/cgroup
tmpfs          tmpfs        32768      4072     28696  13% /var/cache/raspiblitz
/dev/mmcblk0p1 vfat        258095     52243    205853  21% /boot
/dev/sda1      ext4     960330776 385676596 525802320  43% /mnt/hdd
tmpfs          tmpfs       400024         0    400024   0% /run/user/1000
tmpfs          tmpfs       400024         0    400024   0% /run/user/1001
```
I'm guessing all the `tmpfs` are some kind of temporary files?

rootzoll, 2020-07-24:
@grnqrtr yes the tmpfs should not be the problem. The HDD looks ok .. but the sd card seems full. Not sure why ... can you run the following two commands on terminal and post the results:

`sudo du -h -s /home/*`

and

`sudo du -h -s /*`

A solution would be to download the LND backup (MENU > REPAIR > BACKUP-LND), then shutdown, write a fresh raspiblitz image to the sd card and start it up again. But it would be good to know why the sd card got full in the first place first.

grnqrtr, 2020-07-24:
```
admin@raspberrypi:~ $ sudo du -h -s /home/* 2>/dev/null
488M    /home/admin
72K     /home/bitcoin
4.9M    /home/bos
101M    /home/btcrpcexplorer
81M     /home/joinmarket
40M     /home/kindledisplay
25M     /home/pi
admin@raspberrypi:~ $ sudo du -h -s /* 2>/dev/null
13M     /bin
52M     /boot
0       /dev
5.3M    /etc
738M    /home
343M    /lib
16K     /lost+found
4.0K    /media
368G    /mnt
41M     /opt
0       /proc
23G     /root
209M    /run
9.3M    /sbin
4.0K    /srv
0       /sys
28K     /tmp
3.0G    /usr
304M    /var
```

grnqrtr, 2020-07-24:
I just did this too, `sudo du -h -s /root/.bitcoin/* 2>/dev/null`. Is that 23G of blocks on the SD card?
```
root@raspberrypi:/home/admin# sudo du -h -s /root/.bitcoin/* 2>/dev/null
4.0K    /root/.bitcoin/banlist.dat
23G     /root/.bitcoin/blocks
454M    /root/.bitcoin/chainstate
72M     /root/.bitcoin/debug.log
244K    /root/.bitcoin/fee_estimates.dat
4.0K    /root/.bitcoin/mempool.dat
4.0K    /root/.bitcoin/onion_private_key
624K    /root/.bitcoin/peers.dat
1.5M    /root/.bitcoin/wallets
```

frennkie, 2020-07-24:
@rootzoll instead of `du -h - s` I usually run this:

```
du -BM - -max-depth=1 / 2>/dev/null | sort -nr
```

@grnqrtr that's really strange... There should not be blockchain/Bitcoin data on the SDcard. I would recommend you start over with a new SDcard flash.

grnqrtr, 2020-07-24:
Something strange is going on. I flashed 1.6rc3 onto different SD like you both suggested and put it in the pi. After it went through the first setup and rebooted, the first thing I got was this:

[image]

I select `Do Nothing` and get this:

```

*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
error code: -28 error message: Loading block index...

-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

Use CTRL+c to EXIT to Terminal
bitcoin error: error code: -28
error message:
Verifying wallet(s)...
```

This is where I was a couple days ago and why I had reset the chain and resynced.

grnqrtr, 2020-07-24:
Okay, maybe everything is fine? I waited a little bit, then `menu` and had to unlock LND wallet. Then checked `INFO` and everything is up and running. It seems like it was just giving those warnings too early before bitcoind was up and running?

grnqrtr, 2020-07-25:
Not sure why blockchain/Bitcoin data was on the SD card, but after flashing new SD things are working fine so am going to close the issue.

closed 2020-07-25 by grnqrtr