raspiblitz/raspiblitz #1384: Port 11010

marssystems, 2020-07-27:
Is port 11010 assigned to anything by default?

I am getting this error when starting loopd - 

loopd exited with an error: RPC server unable to listen on localhost:11010: listen tcp 127.0.0.1:11010: bind: address already in use

frennkie, 2020-07-27:
A useful command to know in such a situation is `sudo netstat -tulpen`.

marssystems, 2020-07-27:
OK - I did that and I have this -

tcp        0      0 127.0.0.1:11010         0.0.0.0:*               LISTEN      1002       167689     2062/loopd          

Do I need to stop this somehow so loopd can set up a new connection for itself?

rootzoll, 2020-07-29:
So it looks that loopd is already running (or it crashed and was not freeing that port). How to you start lood? Normally the loop-deamon should already be running (when activated in raspiblitz) started by systemd `loopd.service`. And you can interact with it by just using the `loop` command on the terminal.

marssystems, 2020-07-29:
From LOOP support - The Loop client needs its own short-lived daemon which will deal with the swaps in progress.

I understand from that, that when I execute a loop (in/out) it starts the loopd service, executes the in/out and then the loopd service terminates.
But I may be wrong.
If the loopd service is already ruuning when loop in/out is called then it fails with the above error, as it can't start another loopd service.

rootzoll, 2020-07-29:
What du you get back if you run on the terminal `loop terms`?

marssystems, 2020-07-29:
I get this -

admin@raspberrypi:~ $ loop terms
Loop Out
--------
Amount: 250000 - 10000000
Cltv delta: 100 - 400

Loop In
------
Amount: 250000 - 10000000
admin@raspberrypi:~ $

rootzoll, 2020-07-29:
So this looks good so far ... and when you try a swap now with the loop
command its failing? See `loop --help` for options.

marssystems, 2020-07-29:
I don't have enough funds to try another swap at the moment.
Loop isn't working in RTL - completely different error - which I could do with sorting out first.
I will top up my funds shortly and try a swap.

frennkie, 2020-08-02:
As mentioned in #1350 - loopd is already set up as a system service.

Please reopen if there still is a problem.

closed 2020-08-02 by frennkie