raspiblitz/raspiblitz #1386: Question: Install Raspiblitz on normal amd64 PC?

maltokyo, 2020-07-28:
I have an old PC, can I use it instead of a Pi?  I am looking for a tutorial on this, but cannot find, thank you!
(sorry if this is the wrong place to ask, if it is, just redirect me please!)

openoms, 2020-07-28:
Not tested or supported, but worth a try. Will likely need some custom work. Be ready to debug and modify some scripts.
Here are the rough principles:
https://github.com/rootzoll/raspiblitz/tree/master/alternative.platformsare 
Flash a compatible image to the USB drive (not a disk - need to use an equivalent to the SDcard on the RPi).
So set up Debian or Ubuntu to a USB. Boot and run the build_sdcard.sh.
The scripts will format the internal SSD for the blockchain and data.
From there you will need to figure out.
It helps if you run an RPI4 node so you know how it should work.

maltokyo, 2020-07-28:
Thank you. The reason I asked is because I heard you on the Stephan Livera podcast, and there you mentioned that it could be installed on an old PC.  At least that's how I heard it ;)

openoms, 2020-07-28:
Yes it certainly can, most of the install scripts are written to support 32 and 64 bit ARM and 64bit x86.
Testing is not fully done on other boards than the RPi4 (and 3) because of lack of interest.
Any custom setup can be standardised once someone wants to work on it and share the results.

maltokyo, 2020-07-28:
Great. Will try!

adevoss, 2020-07-28:
See https://github.com/rootzoll/raspiblitz/pull/1297.
Drive names (/dev/sdXn) can be different. Check if /etc/fstab is valid after executing build_sdcard.sh. It is modified by the script.
If it is not valid the PC won't boot (happened to me, make backup first)

I tested my pr on a amd64 vm (Debian)
- install Debian (other Linux distro will do I think)
- install and configure sudo if not already installed
- execute: `sudo ./build_sdcard.sh` 
- not used it as a node after the first installation

closed 2022-04-04 by openoms