raspiblitz/raspiblitz #1387: LND was starting but missing /mnt/hdd/lnd/tls.cert

Freeland90, 2020-07-28:
I've downloaded the whole chain but it keeps giving me this error: LND was starting but missing /mnt/hdd/lnd/tls.cert
relevant log:  
unable to load RPC credentials for bitcoind: please set all or none of bitcoind.rpcuser, bitcoind.rpcpass, bitcoind.zmqpubrawblock, bitcoind.zmqpubrawtx

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.5.1
chainnetwork: bitcoin / main
 14:29:29 up  3:02,  2 users,  load average: 1.37, 1.90, 2.01

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-07-28 11:29:27 BST; 3h 0min ago
 Main PID: 3498 (bitcoind)
    Tasks: 18 (limit: 4915)
   Memory: 3.3G
   CGroup: /system.slice/bitcoind.service
           └─3498 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoi…

Jul 28 11:29:27 dnode systemd[1]: Starting Bitcoin daemon...
Jul 28 11:29:27 dnode systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2020-07-28 11:26:28 BST, end at Tue 2020-07-28 14:29:29 BST. --
Jul 28 11:29:27 dnode systemd[1]: Starting Bitcoin daemon...
Jul 28 11:29:27 dnode systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-07-28T13:29:18Z UpdateTip: new best=000000000000000023b7c04f2d086825f5f5b720a3413ccda2113865d561e290 height=298142 version=0x00000002 log2_work=78.269971 tx=37712388 date='2014-04-28T13:47:49Z' progress=0.067476 cache=126.6MiB(1115902txo)
2020-07-28T13:29:20Z Pre-allocating up to position 0x2000000 in blk00137.dat
2020-07-28T13:29:28Z Pre-allocating up to position 0x3000000 in blk00137.dat
2020-07-28T13:29:28Z UpdateTip: new best=0000000000000000565d9d24d06ddb2e60aacca96885489f5e4c94edc4c0b44d height=298143 version=0x00000002 log2_work=78.270089 tx=37713707 date='2014-04-28T14:03:33Z' progress=0.067479 cache=126.7MiB(1116613txo)
2020-07-28T13:29:28Z UpdateTip: new best=000000000000000007552a7d5459098bab7794791a3f8ea0c8e085987525db91 height=298144 version=0x00000002 log2_work=78.270208 tx=37713884 date='2014-04-28T14:00:46Z' progress=0.067479 cache=126.7MiB(1116770txo)
2020-07-28T13:29:28Z UpdateTip: new best=000000000000000011a9fdae4f107b23e10411c5ed46878d888ada36c2836b42 height=298145 version=0x00000002 log2_work=78.270327 tx=37714182 date='2014-04-28T14:09:40Z' progress=0.067480 cache=126.7MiB(1116795txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000003e87ca7f6502812321d910e5fc72bf5435cc2428dd031e68 height=298146 version=0x00000002 log2_work=78.270445 tx=37714203 date='2014-04-28T14:05:29Z' progress=0.067480 cache=126.7MiB(1116831txo)
2020-07-28T13:29:29Z UpdateTip: new best=0000000000000000034cecca9ba69e0c6ea97f3e1e77fa4879cd21121fa9e574 height=298147 version=0x00000002 log2_work=78.270564 tx=37714780 date='2014-04-28T14:20:14Z' progress=0.067481 cache=126.7MiB(1116787txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000009d2519a2fada34521237268604f0b4e17e1652640f98ed3b height=298148 version=0x00000002 log2_work=78.270682 tx=37714964 date='2014-04-28T14:25:32Z' progress=0.067481 cache=126.9MiB(1118242txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000004de4a59fa973aba04ab212ffd8cc4be8f18484a4747f5776 height=298149 version=0x00000002 log2_work=78.270801 tx=37715602 date='2014-04-28T14:35:34Z' progress=0.067482 cache=126.9MiB(1118695txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000001fba9d3abfbc00da0d4ecbb8d377f0390d1815664112554a height=298150 version=0x00000002 log2_work=78.27092 tx=37716525 date='2014-04-28T14:50:36Z' progress=0.067484 cache=127.0MiB(1119239txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000005be42843a69d6ad5b46f6d4af2093d58772f77198817a5ea height=298151 version=0x00000002 log2_work=78.271038 tx=37716589 date='2014-04-28T15:01:10Z' progress=0.067484 cache=127.0MiB(1119365txo)
2020-07-28T13:29:29Z UpdateTip: new best=000000000000000042c03a0d3aad688ab1c6062a5a391605ff854c7a7ff47033 height=298152 version=0x00000002 log2_work=78.271157 tx=37717773 date='2014-04-28T15:14:59Z' progress=0.067486 cache=127.1MiB(1120362txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000006461b7705f56d81bbdfaf5522002285d7cc336ce5e8807d1 height=298153 version=0x00000002 log2_work=78.271275 tx=37718708 date='2014-04-28T15:37:16Z' progress=0.067488 cache=127.2MiB(1120957txo)
2020-07-28T13:29:29Z UpdateTip: new best=000000000000000009a44d45dec0a6f6f7850ce59fe114c9238537f5ec2d6e41 height=298154 version=0x00000002 log2_work=78.271394 tx=37718971 date='2014-04-28T15:37:24Z' progress=0.067488 cache=127.3MiB(1121881txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000006b5f4fe992cbb2e580cc24c16bca2d5a75a82a1e7b874155 height=298155 version=0x00000002 log2_work=78.271512 tx=37719561 date='2014-04-28T15:46:33Z' progress=0.067489 cache=127.2MiB(1121739txo)
2020-07-28T13:29:29Z UpdateTip: new best=00000000000000008a749bcba09a28ca281b7b6dfa635df7817e4ed562a9bda8 height=298156 version=0x00000002 log2_work=78.271631 tx=37720112 date='2014-04-28T15:53:20Z' progress=0.067490 cache=127.3MiB(1122268txo)
2020-07-28T13:29:29Z Pre-allocating up to position 0xe00000 in rev00135.dat
2020-07-28T13:29:30Z UpdateTip: new best=00000000000000002aa5bec4918c88bf9e0556e95e2869206103f55dbde2b02b height=298157 version=0x00000002 log2_work=78.271749 tx=37720357 date='2014-04-28T15:57:42Z' progress=0.067491 cache=127.3MiB(1122346txo)
2020-07-28T13:29:30Z UpdateTip: new best=0000000000000000344de6b3110c56bd6771a974508926802f1f3f4a6f1849b5 height=298158 version=0x00000002 log2_work=78.271868 tx=37720419 date='2014-04-28T15:58:48Z' progress=0.067491 cache=127.3MiB(1122307txo)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-07-28 14:29:26 BST; 3s ago
  Process: 24325 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 24331 ExecStart=/usr/local/bin/lnd --externalip=${publicIP}:${lndPort} ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 24331 (code=exited, status=1/FAILURE)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Tue 2020-07-28 11:26:28 BST, end at Tue 2020-07-28 14:29:30 BST. --
Jul 28 14:28:25 dnode systemd[1]: Started LND Lightning Daemon.
Jul 28 14:28:26 dnode lnd[22987]: unable to load RPC credentials for bitcoind: please set all or none of bitcoind.rpcuser, bitcoind.rpcpass, bitcoind.zmqpubrawblock, bitcoind.zmqpubrawtx
Jul 28 14:28:26 dnode systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Jul 28 14:28:26 dnode systemd[1]: lnd.service: Failed with result 'exit-code'.
Jul 28 14:29:26 dnode systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Jul 28 14:29:26 dnode systemd[1]: lnd.service: Scheduled restart job, restart counter is at 5.
Jul 28 14:29:26 dnode systemd[1]: Stopped LND Lightning Daemon.
Jul 28 14:29:26 dnode systemd[1]: Starting LND Lightning Daemon...
Jul 28 14:29:26 dnode systemd[1]: Started LND Lightning Daemon.
Jul 28 14:29:26 dnode lnd[24331]: unable to load RPC credentials for bitcoind: please set all or none of bitcoind.rpcuser, bitcoind.rpcpass, bitcoind.zmqpubrawblock, bitcoind.zmqpubrawtx
Jul 28 14:29:26 dnode systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Jul 28 14:29:26 dnode systemd[1]: lnd.service: Failed with result 'exit-code'.

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

- TOUCHSCREEN is OFF by config

- Loop is OFF by config

- RTL is OFF by config

- Electrum Rust Server is OFF by config

- BTCPayServer is OFF by config

- LNbits is OFF by config

- SPECTER is OFF by config

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='837500 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='5800 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.8.102'
tempCelsius='55.9'
uptime=10973
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=298244
initialSync=1
syncProgress=6.76 
startcountLightning=185
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=6

openoms, 2020-07-28:
as wrote on TG:
reflash the SDcard with a clean RaspiBlitz image and close this issue if that helped.

The settings and LND wallet will be restored from the disk.

Freeland90, 2020-07-28:
FIX: 
openoms, [28 Jul 2020 at 5:58:47 PM]:
type:
repair
reset the LND wallet
(nothing to lose if there are no funds)

from the logs it seems your passwords are not in sync either so can type:
menu
change password
change password B

closed 2020-07-28 by Freeland90