raspiblitz/raspiblitz #1388: File permissions/ownership for /mnt/hdd/* wrong in 1.6rc3?

freerko, 2020-07-28:
Hi,

when setting up a fresh 1.6rc3 with a clean hdd (formated as ext4) on a Raspberry Pi 3 I got a permission denied error during setup for /mnt/hdd/raspiblitz.conf with the request to report the issue in https://github.com/rootzoll/raspiblitz/issues/293 (which I did, please see that issue for details). I just realized that issue 293 is already closed so I create this new issue. With 1.5.1 and the same hdd (but now formated as FAT as mentioned in the documentation) I didn't have any issues.

I did have the same permission denied issue for /mnt/hdd/raspiblitz.conf when updating my Raspberry Pi 4 from 1.3 to 1.6rc3 where I copied the files from the old hdd to a new ssd and I thought I just messed up the ownership of the files/directories, but apparently it is an issue with 1.6rc3 since it happened with a fresh install as well.

rootzoll, 2020-07-29:
Thanks I will take a look again at this and make a testrun before v1.6 final ... so to reproduce this I would use:
- 1TB HDD freshly formatted as ext4 (formatted from a linux laptop)
- A RaspberryPi 3
Anything else I should take into account to get as closer to your setup?

freerko, 2020-07-29:
Did a new test:

Raspberry Pi 4 with 500GB USB HDD freshly formated as **FAT32** with 1.6rc3:
Got an error after formating happened on the Raspiblitz:

>  \# Wiping all partitions (sfdisk/wipefs)
>  \# Creating the one big partition
>  \# waiting until the partion gets available
> \# Formatting
> mke2fs 1.44.5 (15-Dec-2018)
> \# waiting until formatted drives gets available
> \# OK EXT 4 format done
> \/dev/fd/63: line 2: The: command not found
> \/dev/fd/63: line 3: syntax error near unexpected token `('

> \/dev/fd/63: line 3: `Calling ioctl() to re-read partition table.'
> -# OK

and generating the dhparam.pem took about an hour(!) on the RPI4 but the permissions/ownership seems to be fine. After the setup the Raspberry Pi script complains about a non running bitcoind although the service is actually running (but in the state "Loading block index"):

> \*****************************************
> \* The bitcoind service is not running.
> \*****************************************
>If you just started some config/setup, this might be OK.
>
>More Error Detail:
>error code: -28 error message: Loading block index...
>
>-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
>-> Use command 'debug' for more log output you can use for getting support.
>-> Use command 'menu' to open main menu.
>-> Have you tried to turn it off and on again? Use command 'restart'
>
>Use CTRL+c to EXIT to Terminal
>bitcoin error: error code: -28
>error message:
>Loading block index...
>

The permissions/ownership look like this:

>admin@raspberrypi:~ $ ls -alh /mnt/hdd/
>total 2.1G
>drwxr-xr-x 10 root    root    4.0K Jul 29 19:09 .
>drwxr-xr-x  4 root    root    4.0K Jul 29 17:41 ..
>drwxr-xr-x  5 bitcoin bitcoin 4.0K Jul 29 19:10 app-data
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 17:42 app-storage
>drwxr-xr-x  4 bitcoin bitcoin 4.0K Jul 29 19:11 bitcoin
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 17:41 litecoin
>drwxr-xr-x  4 bitcoin bitcoin 4.0K Jul 29 19:09 lnd
>drwx------  2 root    root     16K Jul 29 17:40 lost+found
>-rwxrwxrwx  1 root    root     272 Jul 29 19:09 raspiblitz.conf
>drwxr-xr-x  2 root    root    4.0K Jul 29 17:41 ssh
>-rw-------  1 root    root    2.0G Jul 29 19:07 swapfile
>drwxrwxrwx  2 bitcoin bitcoin 4.0K Jul 29 19:11 temp

Next test again with Raspberry 4, a fresh SD card with 1.6rc3 but now the 500GB HDD cleanly formated as **ext4**:
After setting the passwords in the setup it fails with this message:

> bitcoin still not running
> ls: cannot access '/mnt/hdd/raspiblitz.conf': Permission denied

> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> FAIL: /mnt/hdd/raspiblitz.conf should exists at this point, but not found!
> Please report to: https://github.com/rootzoll/raspiblitz/issues/293
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> Press ENTER to EXIT.

And the ownership of /mnt/hdd belongs to pi:pi instead of root:root:

> admin@raspberrypi:~ $ ls -alh /mnt/hdd/
>ls: cannot open directory '/mnt/hdd/': Permission denied
>admin@raspberrypi:~ $ sudo ls -alh /mnt/hdd/
>total 52K
>drwx------ 10 pi      pi      4.0K Jul 29 19:41 .
>drwxr-xr-x  4 root    root    4.0K Jul 20 01:15 ..
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 19:41 app-data
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 19:41 app-storage
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 19:38 bitcoin
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 19:38 litecoin
>drwxr-xr-x  2 bitcoin bitcoin 4.0K Jul 29 19:38 lnd
>drwx------  2 root    root     16K Jul 29 19:21 lost+found
>-rwxrwxrwx  1 root    root       0 Jul 29 19:41 raspiblitz.conf
>drwxr-xr-x  2 root    root    4.0K Jul 29 19:41 ssh
>drwxrw-rw-  2 bitcoin bitcoin 4.0K Jul 29 19:41 temp
>admin@raspberrypi:~ $ 

So it is fails with a Raspberry Pi 4 as well, no need to try it with a Raspberry 3 :)

rootzoll, 2020-07-29:
> dhparam.pem took about an hour
@frennkie I havent seen taking this long on my test runs, but it always delays the setup quite a bit ... at least keep in mind for a v1.6.1 update?

rootzoll, 2020-08-02:
OK the v1.6 will also format ext4 fresh of they dont contain a blockchain and I improved the initial writing of the `raspiblitz.conf`.

closed 2020-08-02 by rootzoll