raspiblitz/raspiblitz #1389: IP2Tor Down?

normandmickey, 2020-07-29:
Is the IP2Tor Service down? Had IP2Tor working yesterday but cannot get to my DuckDNS domain today.

rootzoll, 2020-07-29:
In preperation to the final v1.6 release we are closing down the bridge we had running on AWS (which was for testing the relase candidate). Changing the bridge should work like: Go to SUBSCRIBE and List all Subscription - cancel the old IP2TOR Bridge and the setup a new one that directs to the same tor address. Check if your DuckDNS is updating the new IP (if not please report back). The LetsEncrypt should normally stay valid during that process.

frennkie, 2020-07-29:
Additionally to that there really was a bug in the activation script: https://github.com/frennkie/django-ip2tor/commit/662d7b9ec1133f509ae7c44f1d95c1d9670da137#diff-a06799e4473b486e72da6187a734011b

So thanks for the info.. 😉

normandmickey, 2020-07-29:
> frennkie/django-ip2tor@662d7b9#diff-a06799e4473b486e72da6187a734011b

Thank you.  As far as I can tell it's working now.
BTW, I tried to follow the instructions to self host and things got a
little fuzzy.

Here's where I gave up...
"Thirdly you need to register the Hosts in at least one Shop first to get
the credentials that are needed to retrieve the specific information for
this Host from the Shop database and to make updates (e.g. mark a Bridge as
active).

Don't forget to allow the TCP port ranges on your firewall (both local and
network/cloud based)."

frennkie, 2020-07-29:
Did you want to setup a `Shop` where people can pruchase such Bridges?

Or do you have a VPS and want to use it to host one or two bridges just for your self?

normandmickey, 2020-07-29:
To start out I'm just trying to host my own, but i you need more publicly available hosts I would be willing to provide one.

frennkie, 2020-07-29:
The idea is to get enough people using the Fulmo IP2Tor Shop and paying a few sats so that the virtual machines hosting these are paid for. So if like 50 people pay 10 cents a month that would cover the ~ 5€ for a cheap VPS.

If you have a host with public IP all you need is this script: https://github.com/frennkie/django-ip2tor/blob/master/scripts/ip2torc.sh

closed 2020-07-29 by frennkie