raspiblitz/raspiblitz #1390: Changing to LTC Chain and back

hermanlim, 2020-07-29:
Is there a script/menu option to change to LTC (and back to BTC) after one has set up the Raspiblitz (like how one can change from mainnet BTCto testnet BTC and back again)? 

From what I can see, the only way is during setup time.

rootzoll, 2020-07-29:
You choose BTC or LTC at the beginning ... there is no option to switch while running.

hermanlim, 2020-07-29:
Is this deliberately designed this way to avoid fund loss? Is it inherently risky to change post setup?

rootzoll, 2020-07-31:
Its because we cannot test any bonus application how they would work with such a (unsusual) change under the hood. It would be a lot of testing on every release for a edge use case.

closed 2020-08-01 by hermanlim