raspiblitz/raspiblitz #1391: Display doesn't work with new kernel 5.4

freerko, 2020-07-29:
When updating the kernel of a Rasperry Pi 4 to version 5.4 via `rpi-update` on 1.6rc3, the display just stays white. According to https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769 the new kernel comes with a new display driver.

With `sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` it is possible to revert to an old 4.19 kernel where the display works again.

The post https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&sid=e92e16e9faaec20c35efd4ec0a5d693f&start=375#p1700051 might indicate that this is an upstream bug and we just have to wait to get it fixed. 
Just want to document the bug here in order to track it, maybe warn people to update the kernel and show a fix.

rootzoll, 2020-07-29:
Thanks for letting us know ... shipping with that new kernel might is on the list for the v1.7. Will add this to the this release to check on it when it gets closer.

frennkie, 2020-09-05:
duplicate #1436

closed 2020-09-05 by frennkie