raspiblitz/raspiblitz #1395: macron's directory not found

LupusPluvia, 2020-07-31:
hello have problems the macrons directory is gone how can I fix it.  Measure I had set up the SD card.  macrons reset and tls.  Message the directory could not be found
[image]

rootzoll, 2020-07-31:
thats v1.6rc3?

LupusPluvia, 2020-07-31:
yes it is version 16.3rc I forgot to state

rootzoll, 2020-08-02:
The LND seems to have another problem and so is not able to generate macaroons. Can you send the LND logs section from the output of: `MAINMENU > REPAIR > SOFTWARE` or on terminal type `debug`.

LupusPluvia, 2020-08-02:
hello or good 
evening here is the Software Report

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.0rc3
chainnetwork: bitcoin / main
 18:40:31 up 2 days,  4:25,  3 users,  load average: 0.41, 0.36, 0.36

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-07-31 14:17:48 BST; 2 days ago
  Process: 3417 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3422 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 3487 (bitcoind)
    Tasks: 18 (limit: 4915)
   Memory: 3.0G
   CGroup: /system.slice/bitcoind.service
           └─3487 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2020-08-02 06:53:46 BST, end at Sun 2020-08-02 18:40:31 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-08-02T16:42:27Z Pre-allocating up to position 0x7000000 in blk02178.dat
2020-08-02T16:42:27Z UpdateTip: new best=00000000000000000009f5ba1c192f06dca49ad56f18f3bcd8753ad1f136e490 height=641908 version=0x3fff0000 log2_work=92.168014 tx=554614499 date='2020-08-02T16:42:14Z' progress=1.000000 cache=25.9MiB(177073txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T16:43:24Z socket send error Broken pipe (32)
2020-08-02T16:45:48Z socket send error Broken pipe (32)
2020-08-02T16:51:57Z UpdateTip: new best=0000000000000000000c3f41530672109f9594148b5df0ed55969ab88a04741c height=641909 version=0x2000e000 log2_work=92.168032 tx=554616710 date='2020-08-02T16:51:28Z' progress=1.000000 cache=26.7MiB(184500txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T16:52:24Z UpdateTip: new best=00000000000000000001e2c3cb8e46d2378e75ed4df2855ab7f42ac9eef44aae height=641910 version=0x20002000 log2_work=92.168051 tx=554619164 date='2020-08-02T16:52:12Z' progress=1.000000 cache=27.1MiB(188251txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T16:55:28Z socket send error Broken pipe (32)
2020-08-02T16:56:04Z UpdateTip: new best=000000000000000000039e103a37a003a45ccd80ffff391b29e8436c3cfb51f0 height=641911 version=0x20400000 log2_work=92.16807 tx=554620085 date='2020-08-02T16:55:36Z' progress=1.000000 cache=28.2MiB(198008txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T16:56:51Z Pre-allocating up to position 0xf00000 in rev02178.dat
2020-08-02T16:56:51Z UpdateTip: new best=0000000000000000000bbcd1231da3cc6c3ae025866b679fab52a69735241a1a height=641912 version=0x27ffe000 log2_work=92.168089 tx=554621118 date='2020-08-02T16:56:03Z' progress=1.000000 cache=29.0MiB(205879txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T17:05:05Z socket send error Broken pipe (32)
2020-08-02T17:09:24Z Socks5() connect to laeacs7zj2tntqrc.onion:8333 failed: connection refused
2020-08-02T17:14:43Z socket send error Broken pipe (32)
2020-08-02T17:17:08Z New outbound peer connected: version: 70015, blocks=641912, peer=3894 (full-relay)
2020-08-02T17:17:11Z UpdateTip: new best=0000000000000000000c4975f5f6e3d1ed2ca2e42166f3b8d91e9843faa698f9 height=641913 version=0x37ffe000 log2_work=92.168108 tx=554622497 date='2020-08-02T17:16:42Z' progress=1.000000 cache=30.7MiB(221592txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T17:20:33Z UpdateTip: new best=0000000000000000000d6b9740bb7ea68413fa2d407ac47d522d1932fb90385e height=641914 version=0x20400000 log2_work=92.168126 tx=554624879 date='2020-08-02T17:20:13Z' progress=1.000000 cache=31.2MiB(226489txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T17:23:10Z UpdateTip: new best=0000000000000000000a1bec229cbca66e275a04c22526ee7cba93f6d5325261 height=641915 version=0x3fffe000 log2_work=92.168145 tx=554627316 date='2020-08-02T17:23:12Z' progress=1.000000 cache=31.7MiB(230869txo) warning='69 of last 100 blocks have unexpected version'
2020-08-02T17:24:53Z UpdateTip: new best=000000000000000000083e142e7bea5d1b7990e282881326ad387057f6df07ec height=641916 version=0x3fff0000 log2_work=92.168164 tx=554628697 date='2020-08-02T17:24:02Z' progress=1.000000 cache=32.6MiB(239472txo) warning='70 of last 100 blocks have unexpected version'
2020-08-02T17:28:05Z Socks5() connect to iptp5xtxxv5ojfsh.onion:8333 failed: connection refused
2020-08-02T17:33:37Z UpdateTip: new best=000000000000000000009e4abe6784180142575a16debeb343bf5f8976d96f30 height=641917 version=0x20c00000 log2_work=92.168183 tx=554630769 date='2020-08-02T17:33:20Z' progress=1.000000 cache=33.4MiB(246578txo) warning='71 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-07-31 14:42:35 BST; 2 days ago
[... cut ...]
# OK: log 'ERROR: [lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory' written to /home/admin/systemd.lightning.log
lndRPCReady=0
blitzTUIActive=1
blitzTUIRestarts=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

Press ENTER to return to main menu.

LupusPluvia, 2020-08-02:
[image]

LupusPluvia, 2020-08-02:
[image]
[image]

rootzoll, 2020-08-02:
We found the uniary-operator bug and fixed those for the v1.6 final release.

I will retest the regeneration/reset of TLS/Macaroons again.

frennkie, 2020-08-02:
> We found the uniary-operator bug and fixed those for the v1.6 final release.

Did we? I found the same coding mistake in a completely different place.. so this one would still need to be fixed.

LupusPluvia, 2020-08-02:
that's good if everything works out.

rootzoll, 2020-08-02:
> Did we? I found the same coding mistake in a completely different place.. so this one would still need to be fixed.

Ah ok .. I improved now also the line in the raspiblitz.sh file. Will run a retest of LND setup.

I also found the bug why the "TLS/MACAROON RESET" was not working. The new lndunlock was not working correctly when the macaroons are missing ... for v1.6 final it should work. @LupusPluvia thanks for reporting

LupusPluvia, 2020-08-02:
always listening as long as a man can help @rootzoll

LupusPluvia, 2020-08-03:
I fixed that with the older version 15.1.  should I carry out the reset at 1.6 again or leave it as it is now.

if so I would do that tonight

closed 2020-08-09 by rootzoll