raspiblitz/raspiblitz #1396: dial tcp 127.0.0.1:9050: connect: connection refused

pocketleg, 2020-08-01:
I can run RaspiBlitz fine without using TOR but when i activate the "Run Behind TOR", LND fails to start-up after reboot with the following errors:

7 RESTARTS DETECTED - LND might be in a error loop
Aug 01 03:10:27 raspberrypi lnd[24704]: dial tcp 127.0.0.1:9050: connect: connection refused
Aug 01 03:11:28 raspberrypi lnd[27197]: dial tcp 127.0.0.1:9050: connect: connection refused

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

Can anyone help?

pocketleg, 2020-08-01:
In fact, it seems I cannot even start TOR.  This is what i get when I try to start it.

Aug 01 03:30:05.158 [warn] Skipping obsolete configuration option "PortForwarding".
Aug 01 03:30:05.159 [warn] Directory /mnt/hdd/tor/web80/ cannot be read: Permission denied

frennkie, 2020-08-01:
Which version of Raspiblitz are you running? V1.5?

pocketleg, 2020-08-01:
RaspiBlitz v1.5.1 
LND 0.11.0-beta.rc1
bitcoin v0.19.1 mainnet Sync OK 100.00%

When I first set it up 2 months ago, it ran fine (albeit with some TOR errors).  Then I had a power outage, and had to restart the node.  Thats when I started to have issues and turned TOR Services off temporarily.

frennkie, 2020-08-01:
Did you try to start tor as `admin`? I needs to be the `bitcoin` user.

`sudo ls -l /mnt/hdd/tor`?

pocketleg, 2020-08-01:
ah yes u were right, tor seems to be already running after i changed user to 'bitcoin'.

Any idea why i get the error i mentioned in my original message, after i turn on TOR, and restart the node?

7 RESTARTS DETECTED - LND might be in a error loop
Aug 01 03:10:27 raspberrypi lnd[24704]: dial tcp 127.0.0.1:9050: connect: connection refused
Aug 01 03:11:28 raspberrypi lnd[27197]: dial tcp 127.0.0.1:9050: connect: connection refused

admin@raspberrypi:~ $ sudo ls -l /mnt/hdd/tor
total 8271632
drwx------ 3 bitcoin bitcoin       4096 Aug  1 03:56 bitcoin8332
-rwx------ 1 bitcoin bitcoin 8435875930 Aug  1 07:34 info.log
drwx------ 3 bitcoin bitcoin       4096 Aug  1 03:56 lnd9735
drwx------ 3 bitcoin bitcoin       4096 Aug  1 03:56 lndrest8080
drwx------ 2 bitcoin bitcoin       4096 Aug  1 03:56 lndrest8080fallback
drwx------ 3 bitcoin bitcoin       4096 Aug  1 03:56 lndrpc10009
drwx------ 2 bitcoin bitcoin       4096 Aug  1 03:56 lndrpc10009fallback
drwx------ 2 bitcoin bitcoin       4096 Jun  5 20:50 lndrpc9735
drwx------ 2 bitcoin bitcoin       4096 Jun  5 20:50 lndrpc9735fallback
-rwx------ 1 bitcoin bitcoin   34211518 Aug  1 07:33 notice.log
drwx------ 3 bitcoin bitcoin       4096 Aug  1 03:56 RTL
drwx------ 3 bitcoin bitcoin       4096 Aug  1 07:34 sys
drwx------ 3 bitcoin bitcoin       4096 Aug  1 03:56 web80

nyxnor, 2021-05-17:
@pocketleg Kinda hard to debug after months, please close this issue or maintain open if still persists.

pocketleg, 2021-05-17:
Downgraded raspiblitz to older version. Resolved.

closed 2021-05-17 by pocketleg