raspiblitz/raspiblitz #1399: Update Bitcoin Core 0.20.1

rootzoll, 2020-08-01:
https://twitter.com/bitcoincoreorg/status/1289550335598424064?s=20

marssystems, 2020-08-01:
Just updated Bitcoin to v0.20.1 on RaspiBlitz v1.5.1 - no issues, all working as it should,

Here is the link to the update script should anyone want to try it -

sudo wget https://raw.githubusercontent.com/marssystems/bitcoin-tutorials/master/raspiblitz.updates/bitcoincore.update.v0.20.1.sh && bash bitcoincore.update.v0.20.1.sh

Enjoy!

hermanlim, 2020-08-02:
0.20.1 running good on 1.6rc3 as well.

rootzoll, 2020-08-02:
Even those testings look good. I moved this to the v1.7 release, because the v1.6 basic sd card image is now tested well and such changes in the basic sd card image should always be with a new version (not a sub release).

NilByte, 2020-12-13:
I used this update script two times on 1.6.1. and one time on 1.6.2. Found no problems.

rootzoll, 2021-04-08:
As a result of v1.7RC1 there where no problems - will be part of final release ... so closing this issue.

closed 2021-04-08 by rootzoll