raspiblitz/raspiblitz #1400: File Permissions (automate repair)

hermanlim, 2020-08-02:
@rootzoll What are your thoughts on implementing a system to repair file permissions based off acl (ACL utilities which comprise of getfacl and setfacl)? [ reference https://www.linux.com/training-tutorials/how-easily-back-and-restore-linux-file-permissions/ ]

This package is not installed by default on the raspiblitz images but can be done by "apt install acl"

The 'correct' permissions file for the relevant folders in each version could be included with each release in case users get into permissions trouble (or wishes to ensure their system is in a 'safe' or working configuration).

rootzoll, 2020-08-02:
@frennkie any thoughts on this?

hermanlim, 2020-08-04:
Was thinking about this a little more.

In order to trust minimize for users, the intended permissions file could be generated at setup time through a shell script with the output file timestamped. Each time an application is installed through the raspiblitz menu, the script to generate the current permissions would be triggered. The 'new'  output file generated would be compared to previous file generated. If identical, it should be deleted and a hard/symbolic link used in order to link to the previous file to prevent wastage of storage space.

In terms of the overhead this adds, the getfacl process is pretty fast. When done on entire file system, this process took <2min and the resultant file is 63MB large. In the final version, directories housing the blockchain themselves should be excluded, since this would cause the output of getfacl to differ as the blockchain grows. These special directories should have their permissions set automatically by the script to the correct values.