raspiblitz/raspiblitz #1401: support for amd64 cpu?

xpsx, 2020-08-02:
Hello,
can i create an image or iso for installing Raspiblitz in VMware server?
thanks

rootzoll, 2020-08-02:
@openoms any thoughts on this?

openoms, 2020-08-02:
The most tricky part is to get the simulated disks right but certainly possible.
No permission just go ahead and try @xpsx.
You can browse this issue about some clues when we last worked on this:
https://github.com/openoms/raspiblitz/issues/46

Personally I have enough SBC nodes and no servers with superfluous RAM so don't really feel the need.

adevoss, 2020-08-08:
Looks similar to #1386 to me. Did a quick internet search. Disk names are /dev/sdXn. Same as in my linux kvm/qemu vm.
I dont have VMWare so @xpsx can test #1297

openoms, 2021-09-12:
Running in an Ubuntu VM on TrueNAS. See the notes here: https://github.com/rootzoll/raspiblitz/issues/1500#issuecomment-917444439

openoms, 2021-11-08:
closing this for inactivity, see the newer VM related issues