raspiblitz/raspiblitz #1405: EXPERIMENTAL FEATURE: non-existent USB detected

Lit3block, 2020-08-02:
Hi all,
i've a problem with the Raspiblitz configuration whit an Odroid HC1 board.
I've build. the SD card with the SD card build script and when it's time to set up the hard drive, the program detects a second Hard Drive connected, with consequent activation of the EXPERIMENTAL FEATURE. This is what I see from the terminal app:

# *** 30initHDD.sh ***

# --> Checking HDD/SSD status...
isMounted=0
hddRaspiData=0
hddCandidate=sda
OK

# --> Check HDD/SSD for Size ...
 OK

# --> Check HDD/SSD for Blockchain ...
# hddGotBlockchain=0
# raidCandidates=1
# raidSizeGB=0

# !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# EXPERIMENTAL FEATURE: BTRFS + RAID
# !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
# You connected an extra USB thumb drive to your RaspiBlitz.
# This activates the exterimental feature of running BTRFS
# instead of EXT4 and is still unstable but needs testing.
# PRESS ENTER to continue with BTRFS+RAID setup or press
# CTRL+C, remove device & call ./30initHDD.sh again.

No second terminal is connected. Have you experienced with similar errors?

rootzoll, 2020-08-02:
I added a quick fix for this for the v1.6 release ... but @openoms for hardware platforms like the Odroid HC1 we should take a better look at it for the v1.7 release.

Lit3block, 2020-08-02:
> I added a quick fix for this for the v1.6 release ... but @openoms for hardware platforms like the Odroid HC1 we should take a better look at it for the v1.7 release.

oh great! that's amazing. First of all, thanks for reply me.
In this moment there're some things that i can do for go on and don't get stuck?

rootzoll, 2020-08-02:
You can try again with the upcoming v1.6 final release and see if the quick fix works for you.

Lit3block, 2020-08-02:
> You can try again with the upcoming v1.6 final release and see if the quick fix works for you.

Thanks You again.

closed 2021-04-05 by rootzoll